espressif/openocd-esp32 #309: Error: can't add breakpoint: unknown reason

diplfranzhoepfinger, 2023-12-27:
### Development Kit

ESP32-S3-DEVKITC

### Module or chip used

ESP32-S3

### Debug Adapter

none. Want to use USB-JTAG

### OpenOCD version

GNU gdb (esp-gdb) 12.1_20221002

### Operating System

Linux

### Using an IDE ?

Eclipse with Espressif Plugin

### OpenOCD command line

eclipse

### JTAG Clock Speed

default

### ESP-IDF version

esp-idf-v5.1.2

### Problem Description

see log

### Debug Logs

```shell
https://gist.github.com/diplfranzhoepfinger/0586f0724534a997b6d77d7bca200615

Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : address 0x42008a80 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Warn : address 0x42008b3d not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Warn : address 0x42008b46 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Cannot access memory at address 0x42008a80

Info : dropped 'gdb' connection
shutdown command invoked
```

### Expected behavior

Breakpoint to be set

franz-ms-muc, 2023-12-27:
same in Windows: 

[https://gist.github.com/franz-ms-muc/26b6434bc92a5beb30901e3055da7541](https://gist.github.com/franz-ms-muc/26b6434bc92a5beb30901e3055da7541)

erhankur, 2023-12-28:
I see the application resets periodically. Would you happen to know what the reason is?  Can you send me your partition table? And serial console output without OpenOCD running. Also, could you add OpenOCD verbose logs enabled with -d3?

erhankur, 2024-01-15:
Closed due to inactivity.

closed 2024-01-15 by erhankur