espressif/openocd-esp32 #310: Could not find target in FreeRTOS compatibility list error

gamename, 2024-01-08:
### Development Kit

esp32-DevKitC-v4

### Module or chip used

esp32-wroom-32d

### Debug Adapter

esp-prog

### OpenOCD version

v0.12.0

### Operating System

MacOS 14.2.1 (23C71)

### Using an IDE ?

vscode

### OpenOCD command line

openocd -l openocd_log.txt -d3 -f board/esp32-wrover-kit-3.3v.cfg 

### JTAG Clock Speed

20000

### ESP-IDF version

ESP-IDF v5.3-dev-796-g30870c819f

### Problem Description

1. When I try to start `openocd` with the above options, I get an error that says:
```
Error: Could not find target in FreeRTOS compatibility list
/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
```

### Debug Logs

```shell
User : 3 7 options.c:52 configuration_output_handler(): debug_level: 3
User : 4 7 options.c:52 configuration_output_handler(): 
Debug: 5 7 options.c:233 add_default_dirs(): bindir=/usr/local/Cellar/open-ocd/0.12.0_1/bin
Debug: 6 7 options.c:234 add_default_dirs(): pkgdatadir=/usr/local/Cellar/open-ocd/0.12.0_1/share/openocd
Debug: 7 7 options.c:235 add_default_dirs(): exepath=/usr/local/Cellar/open-ocd/0.12.0_1/bin
Debug: 8 7 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 7 configuration.c:33 add_script_search_dir(): adding /Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
Debug: 10 7 configuration.c:33 add_script_search_dir(): adding /Users/tennis/Library/Preferences/org.openocd
Debug: 11 7 configuration.c:33 add_script_search_dir(): adding /Users/tennis/.config/openocd
Debug: 12 7 configuration.c:33 add_script_search_dir(): adding /Users/tennis/.openocd
Debug: 13 7 configuration.c:33 add_script_search_dir(): adding /usr/local/Cellar/open-ocd/0.12.0_1/bin/../share/openocd/site
Debug: 14 7 configuration.c:33 add_script_search_dir(): adding /usr/local/Cellar/open-ocd/0.12.0_1/bin/../share/openocd/scripts
Debug: 15 7 command.c:155 script_debug(): command - ocd_find board/esp32-wrover-kit-3.3v.cfg
Debug: 16 8 configuration.c:88 find_file(): found /Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/board/esp32-wrover-kit-3.3v.cfg
Debug: 17 8 command.c:155 script_debug(): command - ocd_find interface/ftdi/esp32_devkitj_v1.cfg
Debug: 18 8 configuration.c:88 find_file(): found /Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg
Debug: 19 9 command.c:155 script_debug(): command - adapter driver ftdi
Debug: 20 9 command.c:155 script_debug(): command - ftdi vid_pid 0x0403 0x6010 0x0403 0x6014
Debug: 21 9 command.c:155 script_debug(): command - ftdi channel 0
Debug: 22 9 command.c:155 script_debug(): command - ftdi layout_init 0x0008 0xf00b
Debug: 23 9 command.c:155 script_debug(): command - ftdi layout_signal LED -data 0x1000
Debug: 24 9 command.c:155 script_debug(): command - ftdi layout_signal LED2 -data 0x2000
[... 139 lines cut ...]
Debug: 164 13 command.c:155 script_debug(): command - transport select
Info : 165 13 transport.c:266 jim_transport_select(): auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Debug: 166 13 command.c:155 script_debug(): command - transport select
Debug: 167 13 command.c:155 script_debug(): command - expr  [ string first "jtag" $_TRANSPORT ] != -1 
Debug: 168 13 command.c:155 script_debug(): command - jtag newtap esp32 cpu0 -irlen 5 -expected-id 0x120034e5
Debug: 169 13 tcl.c:558 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu0, Dotted: esp32.cpu0, 4 params
Debug: 170 13 tcl.c:582 jim_newtap_cmd(): Processing option: -irlen
Debug: 171 13 tcl.c:582 jim_newtap_cmd(): Processing option: -expected-id
Debug: 172 13 core.c:1478 jtag_tap_init(): Created Tap: esp32.cpu0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 173 13 command.c:155 script_debug(): command - jtag newtap esp32 cpu1 -irlen 5 -expected-id 0x120034e5
Debug: 174 13 tcl.c:558 jim_newtap_cmd(): Creating New Tap, Chip: esp32, Tap: cpu1, Dotted: esp32.cpu1, 4 params
Debug: 175 13 tcl.c:582 jim_newtap_cmd(): Processing option: -irlen
Debug: 176 13 tcl.c:582 jim_newtap_cmd(): Processing option: -expected-id
Debug: 177 13 core.c:1478 jtag_tap_init(): Created Tap: esp32.cpu1 @ abs position 1, irlen 5, capture: 0x1 mask: 0x3
Debug: 178 13 command.c:155 script_debug(): command - target create esp32.cpu0 esp32 -chain-position esp32.cpu0 -coreid 0 -rtos FreeRTOS
Debug: 179 14 target.c:2199 target_free_all_working_areas_restore(): freeing all working areas
Error: 180 14 FreeRTOS.c:556 freertos_create(): Could not find target in FreeRTOS compatibility list

User : 181 14 command.c:608 command_run_line(): /Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
```

### Expected behavior

Not sure what to expect, really. :) Its the first time I've used JTAG.  I assume `Error:` isn't good, so I'm opening this ticket.

erhankur, 2024-01-09:
You are using a different OpenOCD version with Espressif config files. Just check your PATH settings and be sure the OpenOCD binary is running from `/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/`

gamename, 2024-01-09:
@erhankur Thanks! I removed the original one and am using the toolchain version. Got another issue though. :(

Now with this executable:
```/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/openocd```

...I get this error:
```
openocd -f board/esp32-wrover-kit-3.3v.cfg                                                                                                                                                     1 ↵
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/tennis/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9```

erhankur, 2024-01-09:
Which debug adapter do you have?

gamename, 2024-01-09:
@erhankur esp-prog

erhankur, 2024-01-09:
Please check here for the driver issue.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html#macos

gamename, 2024-01-09:
@erhankur Thanks! Here are the results

```sudo kextunload -b com.apple.DriverKit-AppleUSBFTDI                                                                                                                                          
Password:
Executing: /usr/bin/kmutil unload -b com.apple.DriverKit-AppleUSBFTDI
```

... Now I get:
```openocd -f board/esp32-wrover-kit-3.3v.cfg         
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : [esp32.cpu0] Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu0 examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = ffffffff
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
```

erhankur, 2024-01-09:
Ok,  First, I suggest you to load hello-world application to eliminate application-related issues.  Then, make sure your JTAG pin connection is correct.
```
IO12 - TDI
IO13 - TCK
IO14 - TMS
IO15 - TDO
```

erhankur, 2024-01-09:
And, this page can be helpful to solve initial step issues.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ

gamename, 2024-01-09:
Looks good now. :)
```
openocd -f board/esp32-wrover-kit-3.3v.cfg                                                                                                                                  130 ↵
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : target esp32.cpu1 examination failed
Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40009A41, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (16) - (RTC WDT core and rtc reset)
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40079AE0, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (16) - (RTC WDT core and rtc reset)
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
```
@erhankur Thanks for your help. Closing.

closed 2024-01-09 by gamename