espressif/openocd-esp32 #311: Launch problem with "Error: esp_usb_jtag: could not find or open device!"  ESP32-C3 via USB/IP (Windows -> VM)

lure23, 2024-01-09:
### Development Kit

ESP32-C3-DevKitC-02

### Module or chip used

ESP32-C3-WROOM-02

### Debug Adapter

built-in

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Windows + Ubuntu Linux

### Using an IDE ?

no

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

don't know - please advice

### ESP-IDF version

5.3.0

### Problem Description

My situation is a bit ... different. 

I'm running ESP32 development on a Multipass VM (Ubuntu Linux), under macOS, and primarily (=only) interested in setting up a nice, embedded Rust development environment.

Access to the development board is via USB/IP, because ... Multipass doesn't allow normal USB pass-throughs. 

This works. :)

What doesn't work is OpenOCD. I would like eventually to be able to develop using Rust Rover (from JetBrains), but I know they currently don't support OpenOCD in that product. Yet, I have hopes. So wanted to see how far I can get.

Where am I now?

"Everything" works; I can do Rust development on the VM, and flash my code (over USB/IP) to the device. 

I installed OpenOCD via the Linux guidance - all the steps are visible in the repo below - and it seems healthy. When reaching for the device, it gives:

```
$ openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:41)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
/home/ubuntu/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/home/ubuntu/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
```

[... 156 lines cut ...]
Debug: 97 7 command.c:151 script_debug(): command - esp32c3 riscv expose_csrs 2053=cpu_gpio_out
Info : 98 7 server.c:298 add_service(): Listening on port 6666 for tcl connections
Info : 99 7 server.c:298 add_service(): Listening on port 4444 for telnet connections
Debug: 100 7 command.c:151 script_debug(): command - init
Debug: 101 7 command.c:151 script_debug(): command - target init
Debug: 102 7 command.c:151 script_debug(): command - target names
Debug: 103 7 command.c:151 script_debug(): command - esp32c3 cget -event gdb-flash-erase-start
Debug: 104 7 command.c:151 script_debug(): command - esp32c3 configure -event gdb-flash-erase-start reset init
Debug: 105 7 command.c:151 script_debug(): command - esp32c3 cget -event gdb-flash-write-end
Debug: 106 7 command.c:151 script_debug(): command - esp32c3 configure -event gdb-flash-write-end reset halt
Debug: 107 7 command.c:151 script_debug(): command - esp32c3 cget -event gdb-attach
Debug: 108 7 target.c:1624 handle_target_init_command(): Initializing targets...
Debug: 109 7 riscv.c:413 riscv_init_target(): [esp32c3] riscv_init_target()
Debug: 110 7 semihosting_common.c:105 semihosting_common_init():  
Error: 111 9 esp_usb_jtag.c:646 esp_usb_jtag_init(): esp_usb_jtag: could not find or open device!
Debug: 112 9 command.c:541 run_command(): Command 'init' failed with error code -4
User : 113 9 command.c:612 command_run_line(): /home/ubuntu/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/home/ubuntu/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
Debug: 114 9 riscv.c:463 riscv_deinit_target(): [esp32c3] riscv_deinit_target()
Debug: 115 9 target.c:2166 target_free_all_working_areas_restore(): freeing all working areas
```

### Expected behavior

OpenOCD will be able to connect with the device, via built-in JTAG/USB.

lure23, 2024-01-09:
How OpenOCD is installed: https://github.com/lure23/EmbeddedRover/blob/main/mp-prime/manual/openocd.sh

erhankur, 2024-01-13:
@lure23 What do you see on the client side when you try to attach a USB device? I gave it a try and saw the below error on my macOS host. 
```
[2024-01-13T14:20:14Z WARN  usbip::device] unknown desc type: SetupPacket { request_type: 80, request: 6, value: 2000, index: 0, length: ff }
```
I think this definitely can be solved on the rust side. 

Another workaround solution is to skip requesting device descriptors from the esp32c3 and fill the parameters manually on the OpenOCD side. This will require building from the source.

https://github.com/espressif/openocd-esp32/blob/master/src/jtag/drivers/esp_usb_jtag.c#L677

This is an interesting setup. Let me know if you need help while building the OpenOCD.

lure23, 2024-01-14:
>This is an interesting setup. Let me know if you need help while building the OpenOCD.

Thanks!

I'll try to get the connection working on macOS, natively. If that succeeds, I have added confidence and knowledge of the state of JTAG-over-USB.   In reality though, it looks like `probe-rs` might be a better alternative for my needs since it's Rust native and doesn't need the gdb layer.

Steps today:

1. Tried OpenOCD-ESP release 0.12, but it doesn't have `boards/esp32-jtag-builtin.cfg`.

2. Cloned the `master`. 

Prepared dependencies (with HomeBrew).

! Do I need the optional `hidapi` and/or `libftdi` for the C3 JTAG-over-USB?

```
$ ./bootstrap
[...]

$ ./configure
[...]
OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes (auto)
ST-Link Programmer                      yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes (auto)
ANGIE Adapter                           yes (auto)
Altera USB-Blaster II Compatible        yes (auto)
Bitbang mode of FT232R based devices    yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
USBProg JTAG Programmer                 yes (auto)
Espressif JTAG Programmer               yes (auto)
CMSIS-DAP Compliant Debugger            no
Nu-Link Programmer                      no
Cypress KitProg Programmer              no
Altera USB-Blaster Compatible           no
ASIX Presto Adapter                     no
OpenJTAG Adapter                        no
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                no
Bus Pirate                              yes (auto)
Use Capstone disassembly framework      no

````

Observation: Expected to see "JTAG-over-USB"-something, but (looking at the source) "Espressif JTAG Programmer" seems to be it?

(installed locally)

With the ESP32-C3-DevKitC-02 connected:

```
$ openocd -f /usr/local/share/openocd/scripts/board/esp32c3-builtin.cfg 
Open On-Chip Debugger v0.12.0-36eab91e (2024-01-14-12:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
/usr/local/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/usr/local/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
Error: [esp32c3] Unsupported DTM version: -1
Error: [esp32c3] Could not identify target type.
```

This is expected. 

Looking at System Profiler, the device is (only) seen as:

```
$ system_profiler SPUSBDataType
2024-01-14 11:11:06.791 system_profiler[13252:392083] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
2024-01-14 11:11:06.792 system_profiler[13252:392083] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:
[...]
        CP2102N USB to UART Bridge Controller:

          Product ID: 0xea60
          Vendor ID: 0x10c4  (Silicon Laboratories, Inc.)
          Version: 1.00
          Serial Number: bc2f214f809ded11aafa5f84e259fb3e
          Speed: Up to 12 Mb/s
          Manufacturer: Silicon Labs
          Location ID: 0x14600000 / 15
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0
[...]
```

! Should I set it to bootloader mode, somehow?  Would that change the VID:PID?

Reading [README.macOS](https://github.com/espressif/openocd-esp32/blob/master/README.macOS):

- `brew install libtool automake libusb [hidapi] [libftdi]`

   ! Do I need the optional `hidapi` and/or `libftdi` for the C3 JTAG-over-USB?

erhankur, 2024-01-14:
It should be fine as long as you see this in the config summary. 
```
Espressif JTAG Programmer               yes (auto)
```
`esp_usb_jtag` is dependent on `libusb` and looks like it is installed.

The board config file is `board/esp32c3-builtin.cfg`

Unfortunately, I have no idea why your macOS couldn't enumerate the Jtag device. I don't remember if I installed something special on my mac. You should see something like this; 

```
system_profiler SPUSBDataType

USB:
...
        USB JTAG/serial debug unit:

          Product ID: 0x1001
          Vendor ID: 0x303a
          Version: 1.01
          Serial Number: 7C:DF:A1:A2:8F:38
          Speed: Up to 12 Mb/s
          Manufacturer: Espressif
          Location ID: 0x14400000 / 1
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0
 ```

erhankur, 2024-02-24:
@lure23 I don't see an issue related to OpenOCD here. Feel free to re-open if you need something from our side.

closed 2024-02-24 by erhankur