espressif/openocd-esp32 #312: Profiling: xtensa-esp-elf-gprof: can't find .text section in X.elf

bwhitman, 2024-01-13:
### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20230921 

### Operating System

macOS

### Using an IDE ?

no

### OpenOCD command line

 openocd -f board/esp32s3-builtin.cfg

### ESP-IDF version

5.2-beta2

### Problem Description

```
% openocd -f board/esp32s3-builtin.cfg
```

In separate terminal...

```
% telnet localhost 4444
profile 10 gmon.out
```

In third terminal
```
xtensa-esp32s3-elf-gprof build/X.elf gmon.out
```
I always get
```
xtensa-esp-elf-gprof: can't find .text section in X.elf
```

### Expected behavior

Expected: that the xtensa gprof would understand xtensa elf .text sections.

erhankur, 2024-01-13:
Yes because in the elf file, we have `.flash.text` section instead of `.text` You can rename the section names. e.g;
```
cp build/hello_world.elf build/gprof_hello_world.elf
xtensa-esp32s3-elf-objcopy -I elf32-xtensa-le --rename-section .flash.text=.text build/gprof_hello_world.elf
```

bwhitman, 2024-01-13:
That fixed it. Thank you !

closed 2024-01-13 by erhankur

bwhitman, 2024-01-13:
I wrote a HOWTO that brings all this together as I couldn't find it anywhere else, if someone comes looking for help here: 

https://github.com/bwhitman/tulipcc/blob/main/docs/tulip_debug.md

erhankur, 2024-01-13:
@bwhitman Thanks for your support. I'll update our troubleshooting page as well.