espressif/openocd-esp32 #315: ESP32-C6 Debugging with USB-Serial-Jtag

Vineetha1705, 2024-02-05:
### Development Kit

ESP32-C6-DevKitM

### Module or chip used

ESP32-C6

### Debug Adapter

Builtin USB-JTAG

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Windows

### Using an IDE ?

Espressif ide

### OpenOCD command line

-s ${openocd_path}/share/openocd/scripts -f board/esp32c6-builtin.cfg

### JTAG Clock Speed

Default

### ESP-IDF version

 v5.1.2

### Problem Description

1. Go to debug configuration.
2. Choose the application file and click on debug.
3. See the error in console.

### Expected behavior

Esp32c6 should enter into debug mode without any errors.

### Screenshots

[image]

Vineetha1705, 2024-02-08:
> Hi
>
>   1.  Have you flash app binary to flash?
>   2.  There is a warning message with the possible solution following the error. If your app binary or partition table were written to flash at the non-default offsets you need to specify it using appimage_offset command. See https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ#openocd-reports-a-flasher-stub-running-problem-because-of-the-application-binary-location

Thanks for the response,

  1.  Yes, I have flashed the app binary to flash.
  2.
After setting an application image offset explicitly with the OpenOCD command(-s openocd -f board/esp32c6-builtin.cfg -c "init; halt; esp appimage_offset 0x210000") it is not entering into debug mode and showing some errors which are mentioned below.

[cid:b0376b06-48ad-40c7-ade4-bcf9dc77e716]

  1.

Being new to this I didn't understand why I am getting this type or errors. Please help me to solve this errors.

Thanks & Regards,
D. Vineetha.

Vineetha1705, 2024-02-19:
Sorry for that . Is it clear ?.

erhankur, 2024-02-19:
@Vineetha1705 We couldn't see the picture.  Can you attach it again? 

Also, can you send OpenOCD logs? You will add `-d3 -l log.txt` to the end of the config options.  https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/OpenOCD%20Debugging.md#debugger-tab

And it would be good to see application start log screen and partition_table.csv.

Vineetha1705, 2024-02-20:
This is the output log before adding -d3 -1 log.txt to the end of the config options.

***************************************************************************************************
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : esp_usb_jtag: serial (40:4C:CA:46:C4:8C)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Error: esp_usb_jtag: IN buffer overflow! (0, size 48)
Error: missing data from bitq interface
Error: Trying to use configured scan chain anyway...
Error: esp32c6.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: [esp32c6] Unsupported DTM version: 15
Warn : target esp32c6 examination failed
Info : starting gdb server for esp32c6 on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error: 
at file "C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Error: [esp32c6] Unsupported DTM version: 15
*********************************************************************************************************

This is the output after adding -d3 -1 log.txt to the end of the config options.

*********************************************************************************************************

Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3User : 4 2 options.c:52 configuration_output_handler(): 
Debug: 5 2 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 2 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 2 options.c:235 add_default_dirs(): exepath=C:/Users/Vineetha/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
Debug: 8 3 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 3 configuration.c:33 add_script_search_dir(): adding C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
Debug: 10 3 configuration.c:33 add_script_search_dir(): adding C:/Users/Vineetha/AppData/Roaming/OpenOCD
Debug: 11 3 configuration.c:33 add_script_search_dir(): adding C:/Users/Vineetha/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/site
Debug: 12 3 configuration.c:33 add_script_search_dir(): adding C:/Users/Vineetha/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/scripts
Debug: 13 3 command.c:151 script_debug(): command - gdb_port 3333
Debug: 14 3 command.c:151 script_debug(): command - telnet_port 4444
Debug: 15 3 command.c:151 script_debug(): command - tcl_port 6666
Debug: 16 4 command.c:151 script_debug(): command - ocd_find board/esp32c6-builtin.cfg
Debug: 17 4 configuration.c:88 find_file(): found C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/board/esp32c6-builtin.cfg
Debug: 18 5 command.c:151 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg
Debug: 19 5 configuration.c:88 find_file(): found C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/interface/esp_usb_jtag.cfg
Debug: 20 5 command.c:151 script_debug(): command - adapter driver esp_usb_jtag
Info : 21 5 transport.c:106 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 22 5 command.c:151 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 23 5 esp_usb_jtag.c:901 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 24 5 command.c:151 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 25 5 esp_usb_jtag.c:912 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 26 6 command.c:151 script_debug(): command - adapter speed 40000
Debug: 27 6 adapter.c:249 adapter_config_khz(): handle adapter khz
Debug: 28 6 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 29 6 adapter.c:213 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 30 6 command.c:151 script_debug(): command - ocd_find target/esp32c6.cfg
Debug: 31 6 configuration.c:88 find_file(): found C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp32c6.cfg
Debug: 32 6 command.c:151 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 33 6 configuration.c:88 find_file(): found C:\Users\Vineetha\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg
Debug: 34 7 command.c:151 script_debug(): command - ocd_find bitsbytes.tcl
[... cut ...]
Debug: 224 42159 riscv.c:463 riscv_deinit_target(): [esp32c6] riscv_deinit_target()
Error: 225 42159 riscv.c:393 get_target_type(): [esp32c6] Unsupported DTM version: 8
Debug: 226 42159 target.c:2166 target_free_all_working_areas_restore(): freeing all working areas

***************************************************************************************************************************************

And most of the times I am getting error like "openocd timed out failed".

Thanks & Regards,
D. Vineetha.

erhankur, 2024-02-20:
In summary, now you have 2 kinds of issues. 
1 - Failed to get flash maps error. 
2 - After changing the app_image offset you started to get `openocd timed out failed`

Looks strange. They are not related to me.

_Error: esp32c6.CPU: IR capture error; saw 0x1f not 0x01_  usually points to the jtag pins connection or cable problem. Might be also a driver issue. This is windows and drivers can be mixed up. 
_Failed to get flash maps error. _ can be seen after a successful connection when OpenOCD can not read app image details from the specified address. 

Can you try the hello-world example without setting appimage_offset just to be sure your setup is OK?

Vineetha1705, 2024-02-20:
Thanks for the quick response,

When I tried using this simple code 

#include <stdio.h>

void app_main(void)
{
   printf("heloo");
}
I am getting errors like which are mentioned below
***********************************************************************
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32c6.c
pu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
Info : Detected FreeRTOS version: (10.4.3)
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
Info : Detected FreeRTOS version: (10.4.3)

Thread 
1 "IDLE" received signal SIGINT, Interrupt.
Warn : [esp32c6] Could not set a trigger that will match a whole address range. As a fallback, this trigger (and maybe others) will only match against the first address of the range.
Error: [esp32c6] not halted (gdb fileio)
Warn : Target [esp32c6] not HALTED!
1 "IDLE" received signal SIGINT, Interrupt.
Info : Halt timed out, wake up GDB.
Error: [esp32c6] not halted (add breakpoint)
Error: can't add breakpoint: target running
Warning:
Cannot insert breakpoint 1.

Error: [esp32c6] not halted (add breakpoint)
Error: can't add breakpoint: target running
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x420065c6
Error: [esp32c6] not halted (add breakpoint)
Error: can't add breakpoint: target running
******************************************************************
Drivers are installed with Espressif ide installation only.

erhankur, 2024-02-21:
@Vineetha1705 Can you set your flash size to 4MB and try again? From SDK configuration - Serial flasher config → Flash size

Vineetha1705, 2024-02-21:
Yes, Now esp32c6 is entering into the debug mode but sometimes it is not entering into the debug mode showing some errors which are mentioned below.

***************************************************************************************
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show configuration" for configuration details.
<https://www.gnu.org/software/gdb/bugs/>.
shutdown command invoked
*****************************************************************

erhankur, 2024-02-21:
I can reproduce the issue now. Will come up with a solution soon.

erhankur, 2024-02-21:
I have added additional rom symbol file to the Debugger/Commands like below. You need to change `esp-rom-elfs` path according to your setup. Also, increasing remotetimeout can be helpful. 
```
set mem inaccessible-by-default off
set remotetimeout 100
add-symbol-file C:\\Espressif\\tools\\esp-rom-elfs\\20230320\\esp32c6_rev0_rom.elf
```

Can you try and let us know the result?

cc @alirana01

Vineetha1705, 2024-02-22:
If I add this command "add-symbol-file C:\\Espressif\\tools\\esp-rom-elfs\\20230320\\esp32c6_rev0_rom.elf" getting errors otherwise working fine. And why we need to set flash size 4MB ?

erhankur, 2024-02-22:
_And why we need to set flash size 4MB ?_

OpenOCD stub flasher access to the flash through cache. It is related to cache configuration.  Some settings like mmu_page_size change according to the flash size.  Stub flasher configures the cache for the 4MB or bigger flash. It is not adaptive to the smaller flash yet. In my todo list.

So as I understand we can close this issue. Right?

Vineetha1705, 2024-02-23:
Yes, Thank you so much.

Thanks & Regards,
D. Vineetha.

closed 2024-02-23 by erhankur

NirajSiroya, 2024-03-14:
Hello ,
@Vineetha1705 @erhankur can you please look on my issue too.

I'm using esp-idf plugin with Eclipse Ide & ESP32-C6 vboard. 

After updating master version to 5.3 debugging is going to failed.
while using debug mode using ESP_IDF GDB OpenOCD Debugging i'm getting some issues that is listed below,

Info : accepting 'gdb' connection on tcp/3333
Error: no watchpoint at address 0x40825740 found
Error: timed out while waiting for target halted
Error executing event gdb-attach on target esp32c6:
C:\Users\user-id.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error:
at file "C:\Users\user-id.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Error: Target not halted
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
shutdown command invoked

And same thing it was working with previous version 5.2.

And i tried the same thing after incresing that launch timeout.
you can find that snapshot below.

[image]

Vineetha1705, 2024-03-14:
Hello,

Can you set your flash size to 4MB and try again? From SDK configuration - Serial flasher config → Flash size.

image.png (view on web)<https://github.com/espressif/openocd-esp32/assets/159756945/12e458f5-eb22-4c79-96a5-ede96f786e09>

NirajSiroya, 2024-03-14:
Thanks for the information .
As im using same only as 4MB.

this issue arrising after updating of version to 5.3

Vineetha1705, 2024-03-14:
Can you share debug configuration settings?

NirajSiroya, 2024-03-14:
[image]

Vineetha1705, 2024-03-14:
You should give executable path for openocd in openocd setup->Executable path.

NirajSiroya, 2024-03-14:
No , im having 5.2 version in another system there it is working without path .

Vineetha1705, 2024-03-14:
once try it.

NirajSiroya, 2024-03-14:
I tried with Executable path.
after that im getting another error.
Error in services launch sequence

Launching command [F:\esp_work\dout_interrupt\build\Firmware.elf -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\Users\niraj.siroya\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin/share/openocd/scripts -c set ESP32_FLASH_VOLTAGE 1.8 -f board/esp32c6-bridge.cfg -c program_esp_bins F:/esp_work/dout_interrupt/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"] failed.
Launching command [F:\esp_work\dout_interrupt\build\Firmware.elf -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -s C:\Users\niraj.siroya\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin/share/openocd/scripts -c set ESP32_FLASH_VOLTAGE 1.8 -f board/esp32c6-bridge.cfg -c program_esp_bins F:/esp_work/dout_interrupt/build flasher_args.json verify reset -c echo "Started by GNU MCU Eclipse"] failed.
Cannot run program "F:\esp_work\dout_interrupt\build\Firmware.elf": Launching failed

Vineetha1705, 2024-03-14:
Can you send me the screenshots of which procedure are you following?.

NirajSiroya, 2024-03-14:
this is my debug configuration.
[image]

**after changing to this one im getting below errors.**
Error in final launch sequence
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
localhost:3333: Connection timed out.

Vineetha1705, 2024-03-14:
Are you using debugger as Esp USB bridge?.

image.png (view on web)<https://github.com/espressif/openocd-esp32/assets/159756945/622a8252-da83-443d-a192-0e87b73b9f53>

NirajSiroya, 2024-03-14:
yes , i'm.

Vineetha1705, 2024-03-14:
Sorry , I don't have idea about that. I worked using builtin jtag.

erhankur, 2024-03-14:
@NirajSiroya Please make sure that you are using onboard usb port to start debugging or bridge board as a debug probe from here. https://github.com/espressif/esp-usb-bridge

For the on board usb_serial_jtag you must select `C6 chip (via builtin USB-JTAG)`  For the latter, `C6 chip (via ESP USB Bridge)` 

And you don't need to set flash voltage as 1.8v. This parameter is only for the esp32 chips to set flash voltage via TDI pin during boot.

Please select `enable verbose output` from debugger config in the startup section and share the openocd logs.

NirajSiroya, 2024-03-15:
This is the errors , what i'm getting after enabling verbose output in debug configuration.

**Error in final launch sequence**
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
Failed to execute MI command:
-target-select remote localhost:3333
Error message from debugger back end:
localhost:3333: Connection timed out.
localhost:3333: Connection timed out.

erhankur, 2024-03-16:
I suggest you to run debugging from the command line first. "idf.py openocd" and "idf.py gdb" commands are pretty helpful and for us it will be easier to understand what is going on.

Then, you can be sure your env is working and switch to the IDE again.

NirajSiroya, 2024-07-12:
@erhankur Yes , Its working with command line . But as i'm not using command line tool So i wanted it works with eclipse ide also

cmsd2, 2024-08-12:
was this error specific to the c6? i'm getting something similar on the h2:
```
Info : accepting 'gdb' connection on tcp/50000
Info : [esp32h2] Target halted, PC=0x40800400, debug_reason=00000008
Info : [esp32h2] Halt cause (5) - (PMP Load access fault)
Error: [esp32h2] Algorithm timed out after 3001 ms.
Info : [esp32h2] Target halted, PC=0x40800400, debug_reason=00000000
Info : [esp32h2] Halt cause (7) - (PMP Store access fault)
```
using openocd v0.12.0-esp32-20240726

erhankur, 2024-08-12:
@cmsd2 can you give a detail about the steps to reproduce the issue?

mnigbor, 2025-02-16:
I was having a similar issue getting gdb to connect to my ESP32-C3 using the ESP-IDF command-line. It's about a year later, so I'm using IDF 5.4 and OpenOCD 120.0.  When I ran gdb, the OpenOCD server would reject the connection after reporting it could not find FreeRTOS symbols (and other error messages similar to the ones above). 

Changing the flash memory size from 2M to 4M in menuconfig fixed the issue.

erhankur, 2025-02-16:
@mnigbor Flash size issue was related to ESP32-C6 and H2 and it is fixed at  this [release](https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20240726)

> loaders/esp: Fixed dynamic mmu page size issue. With the 2MB flash config, the stub flasher could not properly configure the Cache MMU, so mapping and flash breakpoints were not working.

I am not aware that we have the same issue on ESP32-C3 and ESP32-S3. Please create a new issue with the OpenOCD logs and details of your hardware and application config.