espressif/openocd-esp32 #316: usb-jtag OpenOCD error on esp32c6 (devkit) restart with on-board button (RST)

vandy, 2024-02-16:
### Development Kit

ESP32-C6-DEVKIT

### Module or chip used

ESP32-C6-WROOM-1

### Debug Adapter

Built-in usb-jtag

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Windows 10

### Using an IDE ?

vsCode

### OpenOCD command line

openocd.exe -f board/esp32c6-builtin.cfg

### JTAG Clock Speed

24000 kHz

### ESP-IDF version

v5.2

### Problem Description

I'm working inside vsCode with Espressif IDF Extension v1.7.0, running all commands inside ESP-IDF Terminal (provided by the extension).
1. Press Open ESP-IDF Terminal button in vscode
2. Run command openocd.exe -f board/esp32c6-builtin.cfg
3. Press Monitor Device button in vscode. (Firmware information appears with some logging inside the app - which increments a counter every 2 seconds and logs it)
4. In Monitor I can restart the board with `Ctrl+T, Ctrl+R` shortcut, OpenOCD feels ok `Info : [esp32c6] Reset cause (21) - (USB (UART) core reset)`. Monitor reconnects with the full information as in step 3.
5. Press RST button on the devboard, OpenOCD starts reporting errors (but Monitor is able to reconnect, though its output is in the middle of running app, not from the beginning as in step 3).

Errors are similar to [issue 176](https://github.com/espressif/openocd-esp32/issues/176#issuecomment-937796625). I've updated esp-idf and tools several days ago, I believe there were no problems on the previous versions, but I'm not sure.

I've tried different options with usb drivers described in different posts (using Zadig, Espressif, default Windows) - there is no difference in the behavior. 

I can flash, monitor, debug through usb-jtag in vsCode, there is only problem when I press reset button. When I connect my board through usb-uart (on-chip bridge) everything is fine after reset (apparently no OpenOCD is running in this case).

[oocd.log](https://github.com/espressif/openocd-esp32/files/14309018/oocd.log)

### Debug Logs

```shell
ESP-IDF Terminal output:

openocd.exe -f board/esp32c6-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (40:4C:CA:48:CF:74)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32c6.cpu tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] datacount=2 progbufsize=16
Info : [esp32c6] Examined RISC-V core; found 2 harts
Info : [esp32c6]  XLEN=32, misa=0x40903105
Info : starting gdb server for esp32c6 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32c6] Hart unexpectedly reset!
Info : [esp32c6] Reset cause (21) - (USB (UART) core reset)
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32c6] Failed DMI read at 0x11; status=2
Error: [esp32c6] polling failed!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32c6] Failed DMI read at 0x11; status=2
Error: [esp32c6] polling failed!
...
```

### Expected behavior

I expect OpenOCD doesn't throw errors after resetting the board with on-board button (as it's when resetting in Device Monitor).

erhankur, 2024-02-21:
It is because, when you hard reset the board, the usb-serial-jtag peripheral also will be under reset. This is a hardware design for now. However, we added a recovery mechanism into the OpenOCD to cover such cases. Communication should be established again. As I see recovery is not working for you, right?

vandy, 2024-02-22:
Yes, the recovery mechanism doesn't work for me. As I do a hard reset, OpenOCD starts spitting errors out and never gets reconnected. The only method to make it work is to stop it (`Ctrl+C`) and restart `openocd.exe -f board/esp32c6-builtin.cfg`.

erhankur, 2024-02-22:
OK. I will have a look at it.

erhankur, 2024-05-17:
@vandy We have a fix for the Windows. Can you give it a try with the master branch built? https://github.com/espressif/openocd-esp32/actions/runs/9130378811

erhankur, 2024-05-23:
I am closing the issue. Feel free to re-open if you have a similar problem with the latest master.

closed 2024-05-23 by erhankur

vandy, 2024-07-31:
Hi, I've installed ESP-IDF v5.3, vscode extension v1.8.0, openocd Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26) and working with H2 chip (dev board) right now. It seems the problem stays the same.
When I restart the board from the Monitor with `Ctrl+T, Ctrl+R` shortcut, OpenOCD feels ok `Info : [esp32c6] Reset cause (21) - (USB (UART) core reset)`. Monitor reconnects with all information.
When pushing the RST button on the dev board OpenOCD reports errors:
```cmd
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32h2] Failed DMI read at 0x11; status=2
Error: [esp32h2] polling failed!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32h2] Failed DMI read at 0x11; status=2
Error: [esp32h2] polling failed!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32h2] Failed DMI read at 0x11; status=2
Error: [esp32h2] polling failed!
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: dmi_scan failed jtag scan
Error: libusb_bulk_write error: LIBUSB_ERROR_IO
Error: missing data from bitq interface
Error: failed jtag scan: -104
Error: [esp32h2] Failed DMI read at 0x11; status=2
Error: [esp32h2] polling failed!
```

erhankur, 2024-08-01:
@vandy We are about to release the new OpenOCD version. You can wait for it or try with the master branch artifact. https://github.com/espressif/openocd-esp32/actions/runs/10113579666/artifacts/1744402679