espressif/openocd-esp32 #317: Debugging ESP32-S3 2nd Core

ProfFan, 2024-02-16:
### Development Kit

ESP32-S3-MINI

### Module or chip used

ESP32-S3-MINI

### Debug Adapter

BUILTIN USB

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Linux

### Using an IDE ?

VSCode

### OpenOCD command line

openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /home/fan/magic-loc-rs -c "set ESP_RTOS none" -c "adapter serial DC:54:75:EF:A2:2C" -f /home/fan/.vscode-server/extensions/marus25.cortex-debug-1.12.1/support/openocd-helpers.tcl -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

5

### Problem Description

How can I attach to core 1 when using GDB?

### Expected behavior

There is no docs at all

erhankur, 2024-02-17:
You can set `ESP_RTOS` as  `hwthread`.  Then when gdb connects, you can execute `info threads` to see individual cores. And you can switch to the core by `thread 1`  or `thread 2`

ProfFan, 2024-02-18:
> You can set `ESP_RTOS` as `hwthread`. Then when gdb connects, you can execute `info threads` to see individual cores. And you can switch to the core by `thread 1`  or `thread 2`

Thank you! Could this be put in the official ReadTheDocs as well?

erhankur, 2024-02-21:
Since this is not a common use case for our users, I am not sure where to put this information. 

It is mentioned in the official document, section 20.7. https://openocd.org/doc/html/GDB-and-OpenOCD.html 

Thanks for the verification. I am closing this.

closed 2024-02-21 by erhankur

5ami, 2024-08-01:
@erhankur erhankur How can we do this via the extension debugging user interface   ?