espressif/openocd-esp32 #319: Change partition table offset, OpenOCD fails (IDFGH-12364)

szekelyisz, 2024-03-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Changing `PARTITION_TABLE_OFFSET` in `sdkconfig` makes debugger startup fail, even with `esp appimage_offset <offset>`.

Using ESP32 Ethernet Kit.

<details>
<summary>OpenOCD log</summary>
GNU gdb (esp-gdb) 12.1_20231023
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : New GDB Connection: 1, Target esp32.cpu1, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu1] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : Detected FreeRTOS version: (10.5.1)
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] requesting target halt and executing a soft reset
Info : [esp32.cpu0] Debug controller was reset.
Info : [esp32.cpu0] Core was reset.
Info : [esp32.cpu0] Target halted, PC=0x500000CF, debug_reason=00000000
Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.
Info : [esp32.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32.cpu1] Target halted, PC=0x500000CF, debug_reason=00000000
[... 109 lines cut ...]
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /home/szabi/esp/esp-idf
GDK_CORE_DEVICE_EVENTS: 1
DESKTOP_SESSION: plasmawayland
USER: szabi
GTK2_RC_FILES: /etc/gtk-2.0/gtkrc:/home/szabi/.gtkrc-2.0:/home/szabi/.config/gtkrc-2.0
QT_ACCESSIBILITY: 1
SSH_AUTH_SOCK: /run/user/1000/openssh_agent
XDG_SEAT: seat0
STM32CubeMX_PATH: /home/szabi/apps/STM32CubeMX
IDF_MAINTAINER: 1
XDG_RUNTIME_DIR: /run/user/1000
XDG_VTNR: 1
IDF_PYTHON_ENV_PATH: /home/szabi/.espressif/python_env/idf5.2_py3.11_env
HOME: /home/szabi
ESP_ROM_ELF_DIR: /home/szabi/.espressif/tools/esp-rom-elfs/20230320/

Operating System: linux
Java Runtime Version: 17.0.10+7-Debian-1deb12u1
Eclipse Version: 4.31.100.v20240229-0520
Eclipse CDT Version: 11.5.0.202312062348
IDF Eclipse Plugin Version: 2.12.1.202402231119
ESP-IDF v5.2.1-182-g08299f5f9b
Python set for IDF_PYTHON_ENV: Python 3.11.2
</details>

erhankur, 2024-03-16:
@szekelyisz how did you run appimage_offset command? Can you send verbose(-d3) OpenOCD logs from the beginning?

Also please send your partition table or application console output to check the new offsets.

I see from the env variables, openocd script location doesn't show the current OpenOCD version you run. I don't say related to this issue but always better to use correct script set.

szekelyisz, 2024-03-19:
@erhankur Thanks for looking into this issue.

> @szekelyisz how did you run appimage_offset command?

I add `-c "init; halt; esp appimage_offset 0xa000"` to OpenOCD command line as described [here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/jtag-debugging/tips-and-quirks.html#flash-mappings-vs-sw-flash-breakpoints)

> Can you send verbose(-d3) OpenOCD logs from the beginning?

<details>
<summary>openocd -d3</summary>
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Type "show copying" and "show warranty" for details.
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
Info : 7680 9504 server.c:90 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 7681 9504 breakpoints.c:363 breakpoint_clear_target_internal(): Delete all breakpoints for target: esp32.cpu0
Debug: 7682 9504 breakpoints.c:363 breakpoint_clear_target_internal(): Delete all breakpoints for target: esp32.cpu1
Debug: 7683 9504 breakpoints.c:551 watchpoint_clear_target(): Delete all watchpoints for target: esp32.cpu1
Debug: 7684 9504 target.c:1812 target_call_event_callbacks(): target event 22 (gdb-attach) for core esp32.cpu1
Debug: 7685 9504 target.c:5047 target_handle_event(): target(1): esp32.cpu1 (esp32) event: 22 (gdb-attach) action: 
			if { $_ESP_SMP_BREAK != 0 } {
				$_TARGETNAME_1 xtensa smpbreak BreakIn BreakOut
			}
			# necessary to auto-probe flash bank when GDB is connected
			halt 1000
			if { [$_ESP_MEMPROT_IS_ENABLED] } {
				# 'reset halt' to disable memory protection and allow flasher to work correctly
				echo "Memory protection is enabled. Reset target to disable it..."
				reset halt
			}
		
Debug: 7686 9504 command.c:151 script_debug(): command - esp32.cpu1 xtensa smpbreak BreakIn BreakOut
Debug: 7687 9504 xtensa.c:871 xtensa_smpbreak_write(): [esp32.cpu0] write smpbreak set=0x30000 clear=0x600000
Debug: 7688 9504 xtensa.c:887 xtensa_smpbreak_set(): [esp32.cpu0] set smpbreak=30000, state=2
Debug: 7689 9504 xtensa.c:871 xtensa_smpbreak_write(): [esp32.cpu1] write smpbreak set=0x30000 clear=0x600000
Debug: 7690 9504 xtensa.c:887 xtensa_smpbreak_set(): [esp32.cpu1] set smpbreak=30000, state=2
Debug: 7691 9504 command.c:151 script_debug(): command - halt 1000
Debug: 7692 9504 target.c:3273 handle_halt_command(): -
Debug: 7693 9504 xtensa.c:1483 xtensa_halt(): [esp32.cpu1] start
Debug: 7694 9504 xtensa.c:1485 xtensa_halt(): [esp32.cpu1] target was already halted
Debug: 7695 9505 FreeRTOS.c:1387 freertos_clean(): freertos_clean
Debug: 7696 9505 FreeRTOS.c:882 freertos_update_threads(): freertos_update_threads
Warn : 7697 9505 FreeRTOS.c:890 freertos_update_threads(): No symbols for FreeRTOS!
Info : 7698 9505 gdb_server.c:1056 gdb_new_connection(): New GDB Connection: 1, Target esp32.cpu1, state: halted
Debug: 7699 9505 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+
Debug: 7700 9505 xtensa.c:1421 xtensa_get_gdb_reg_list(): reg_class=0, num_regs=173
Debug: 7701 9505 gdb_server.c:392 gdb_log_outgoing_packet(): [esp32.cpu1] sending packet: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02
Debug: 7702 9505 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: +
Debug: 7703 9505 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: vMustReplyEmpty
Debug: 7704 9505 gdb_server.c:392 gdb_log_outgoing_packet(): [esp32.cpu1] sending packet: $#00
Debug: 7705 9505 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: +
Debug: 7706 9505 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: QStartNoAckMode
Debug: 7707 9505 gdb_server.c:392 gdb_log_outgoing_packet(): [esp32.cpu1] sending packet: $OK#9a
Debug: 7708 9506 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: +
Debug: 7709 9506 gdb_server.c:678 gdb_get_packet_inner(): Received first acknowledgment after entering noack mode. Ignoring it.
Debug: 7710 9506 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: Hg0
Debug: 7711 9506 rtos.c:484 rtos_thread_packet(): RTOS: GDB requested to set current thread to 0x0
Debug: 7712 9506 gdb_server.c:392 gdb_log_outgoing_packet(): [esp32.cpu1] sending packet: $OK#9a
Debug: 7713 9506 gdb_server.c:375 gdb_log_incoming_packet(): [esp32.cpu1] received packet: qXfer:features:read:target.xml:0,1000
Debug: 7714 9506 xtensa.c:1421 xtensa_get_gdb_reg_list(): reg_class=0, num_regs=173
Debug: 7715 9506 gdb_server.c:2340 smp_reg_list_noread(): [esp32.cpu1] pc not found in combined list
Debug: 7716 9506 gdb_server.c:2340 smp_reg_list_noread(): [esp32.cpu1] ar0 not found in combined list
Debug: 7717 9506 gdb_server.c:2340 smp_reg_list_noread(): [esp32.cpu1] ar1 not found in combined list
[... 1822 lines cut ...]
Debug: 9456 10169 xtensa.c:959 xtensa_core_status_check(): [esp32.cpu0] DSR (8000CC11)
[... cut ...]
Debug: 9475 10186 esp_xtensa.c:266 esp_xtensa_reset_reason_read(): [esp32.cpu1] Resume after reset cause read
Debug: 9476 10186 xtensa.c:1592 xtensa_resume(): [esp32.cpu1] start
Debug: 9477 10186 xtensa.c:1520 xtensa_prepare_resume(): [esp32.cpu1] current=1 address=0x00000000, handle_breakpoints=1, debug_execution=0)
Debug: 9478 10186 xtensa.c:1533 xtensa_prepare_resume(): [esp32.cpu1] DEBUGCAUSE 0x20 (watchpoint 0) (break 0)
Debug: 9479 10186 xtensa.c:613 xtensa_write_dirty_registers(): [esp32.cpu1] start
Debug: 9480 10186 xten

erhankur, 2024-03-19:
@szekelyisz When you don't set the appimage offset manually, OpenOCD will try to get that offset from the default partition table. (0x8000) 

If you change the partition table offset you have to set your new appimage location. In this case, it is 0x20000 not 0xA000.

Because if the appimage offset is set, OpenOCD will not check the partition table anymore. It will try to read the specified address directly. 

Please try; 
```
-c "init; halt; esp appimage_offset 0x20000"
```

erhankur, 2024-03-22:
@szekelyisz I have shared the issue with the IDE team. We will get back when we have a clear understanding.

erhankur, 2024-05-02:
@szekelyisz I made a fix that can solve your issue. Can you test again with the master branch? You can find the Linux built [here](https://github.com/espressif/openocd-esp32/actions/runs/8929760151)

erhankur, 2024-05-06:
@szekelyisz can you confirm if the fix in the master branch works for you?

szekelyisz, 2024-05-08:
@erhankur Your fix seems to work. At least debugging starts up properly and breakpoints are hit with the partition table offset set to `0xa000` and `-c "init; halt; esp appimage_offset 0x20000"` added to OpenOCD command line.

closed 2024-05-08 by szekelyisz