espressif/openocd-esp32 #320: OpenOCD target for ESP32-C6 is missing RISC-V CSRs (IDFGH-12420)

robotman2412, 2024-03-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For low-level debugging purposes, I would like to be able to read RISC-V CSRs like `mie` and `mip`. Unfortunately, the `esp32c6-builtin.cfg` target does not expose these CSRs.

I need to be able to access the following CSRs:
- medeleg
- mideleg
- mie
- mip
- mtinst
- mvendorid
- marchid
- mimpid
- mhartid

These CSRs exist in any RISC-V core and I need to see them but they're not available either through GDB or directly with monitor commands.

igrr, 2024-03-21:
Some of these CSRs don't seem to exist in ESP32-C6 (referring to the [TRM](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf)), but we'll add those which do!

erhankur, 2024-03-22:
@robotman2412 Thanks for reporting this. I will be looking at the missing registers. Some of them are read only. We will need to add a flag to disable store/restore option during the algorithm run.

erhankur, 2024-04-03:
@robotman2412 I am adding the missing regs which are mentioned in the TRM. 
```
mvendorid
marchid
mimpid
mhartid
mideleg
mie
mip
```
The others (medeleg, mtinst) are not documented so I am leaving them out.

erhankur, 2024-04-04:
Completed via 15867d670d2a3d02125cf1a3f0ba92bd8053e175

closed 2024-04-04 by erhankur

robotman2412, 2024-04-10:
> @robotman2412 I am adding the missing regs which are mentioned in the TRM.
> 
> ```
> mvendorid
> marchid
> mimpid
> mhartid
> mideleg
> mie
> mip
> ```
> 
> The others (medeleg, mtinst) are not documented so I am leaving them out.

Actually, according to the RISC-V privileged specification, `medeleg` is necessarily present if a lower privilege mode can handle traps, and the ESP32-C6 has a CPU that can delegate interrupts and exceptions to U-mode.

If `mideleg` exists, then `medeleg` does too and I have verified this by writing assembly that accesses the `medeleg` CSR.

erhankur, 2024-04-12:
@robotman2412 you are right. I have added the `medeleg`