espressif/openocd-esp32 #323: ESP32H2 new project template does not work with JTAG/USB (for flashing and debugging). Something missing. BLINK demo code do work.

riscy00, 2024-04-26:
### Development Kit

ESP32H2 Dev Board with dual USB 

### Module or chip used

ESP32H2

### Debug Adapter

JTAG/USB

### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)

### Operating System

Window 11

### Using an IDE ?

VSC with ESP-IDF 5.2.1 extension. 

### OpenOCD command line

Using Flash feature (bottom symbol) via JTAG and ESP32H2 Traget

### JTAG Clock Speed

Default

### ESP-IDF version

V5.2.1

### Problem Description

Using ESP32H2 board devkit with dual USB port (USB and UART), both are connected to 4:1 USB adaptor. I use blink demo code under ESP32H2 target with JTAG/USB enabled, it flash successfully after running twice and it also debug successfully after running twice. 
Now I have new project TESTH2_GPIO_4B with esp template . I also added FMT component and correctly sorted out C/C++ lang linkage. It complies/links without error. I use the UART (COM7) which flash the board successfully and it did work. I then switched to JTAG/USB so I can flash and debug this way. it failed. I ruled out the OPENOCD driver issue because it is working fine in the BLINK demo code (I did update the USB driver earlier; I got two). I then rebooted the laptop and cleaned the build, and it complies with no error. I check the JSON setting, but nothing comes out significant, there is a difference between Blink and this project. I could not flash or debug ESP32H2, just like BLINK. I struggled to find a solution for this issue; something was missing in the config/setup. 
[ESPH2_JTAGUSB_ISSUE_TESTH2_GPIO_4B_26Apr24.pdf](https://github.com/espressif/openocd-esp32/files/15133538/ESPH2_JTAGUSB_ISSUE_TESTH2_GPIO_4B_26Apr24.pdf)
[openocdH2_blink_log.txt](https://github.com/espressif/openocd-esp32/files/15133545/openocdH2_blink_log.txt)
[openocdH2_TestH2GPIO4B_log.txt](https://github.com/espressif/openocd-esp32/files/15133557/openocdH2_TestH2GPIO4B_log.txt)

### Debug Logs

```shell
See attached file between BLINK and TESTH2GPIO4B based on this command

openocd -l openocdH2_log.txt  -d3 -f board/esp32h2-builtin.cfg
```

### Expected behavior

Successfully flash and run debug on TESTH2_GPIO_4B just like BLINK do on the same ESP32H2 devkit board device. 

### Screenshots

esw

erhankur, 2024-05-02:
@riscy00 Thank you for preparing such a detailed issue report page. 

Can you connect OpenOCD from telnet and send the below commands?

```
halt
flash probe 0
```

Please share the OpenOCD -d3 logs, and sdkconfig file.

Or, If you can send us a reproducible app, it would be the fastest way to help you.

erhankur, 2024-05-06:
@riscy00 Did you have a chance to have a look?

riscy00, 2024-05-06:
Hi, sorry for delay in reply, I have been busy on C++ studies with ESP32. Not sure what you mean by reproducible app, can you explain how to do that, thanks.

I could not get halt/flash working from the terminal, see below. The logs parameter is not recognised.   

PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> OpenOCD -d3 logs
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3User : 4 5 options.c:52 configuration_output_handler(): 
Unexpected command line argument: logs
PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> openOCD -version
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin\openocd.exe: unknown option -- e
PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> openOCD -d3
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 2 options.c:52 configuration_output_handler(): debug_level: 3User : 4 5 options.c:52 configuration_output_handler(): 
Debug: 5 7 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 6 10 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 7 16 options.c:235 add_default_dirs(): exepath=C:/Users/riscy/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin
Debug: 8 22 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 9 24 configuration.c:33 add_script_search_dir(): adding C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
Debug: 10 30 configuration.c:33 add_script_search_dir(): adding C:/Users/riscy/AppData/Roaming/OpenOCD
Debug: 11 34 configuration.c:33 add_script_search_dir(): adding C:\Users\riscy/.config/openocd
Debug: 12 38 configuration.c:33 add_script_search_dir(): adding C:\Users\riscy/.openocd
Debug: 13 40 configuration.c:33 add_script_search_dir(): adding C:/Users/riscy/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/site
Debug: 14 46 configuration.c:33 add_script_search_dir(): adding C:/Users/riscy/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin/../share/openocd/scripts
Debug: 15 51 command.c:151 script_debug(): command - ocd_find openocd.cfg
Debug: 16 54 command.c:541 run_command(): Command 'ocd_find' failed with error code -603
User : 17 60 command.c:612 command_run_line(): embedded:startup.tcl:28: Error: Can't find openocd.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 28
Info : 18 67 server.c:298 add_service(): Listening on port 6666 for tcl connections
Info : 19 70 server.c:298 add_service(): Listening on port 4444 for telnet connections
Debug: 20 71 command.c:151 script_debug(): command - init
Debug: 21 75 command.c:151 script_debug(): command - target init
Debug: 22 77 command.c:151 script_debug(): command - target names
Debug: 23 79 target.c:1624 handle_target_init_command(): Initializing targets...
Error: 24 83 adapter.c:127 adapter_init(): Debug Adapter has to be specified, see "adapter driver" command
Debug: 25 85 command.c:541 run_command(): Command 'init' failed with error code -101
User : 26 88 command.c:612 command_run_line(): embedded:startup.tcl:28: Error: 
in procedure 'script'
at file "embedded:startup.tcl", line 28
PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> 

PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> halt
halt: The term 'halt' is not recognized as a name of a cmdlet, function, script file, or executable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B> C:\Users\riscy\esp\openocd-esp32\bin\openocd.exe
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:28: Error: Can't find openocd.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 28
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Debug Adapter has to be specified, see "adapter driver" command
embedded:startup.tcl:28: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 28
PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B>

riscy00, 2024-05-06:
I have previously setup paths in envrionment as attached. 
[image]

erhankur, 2024-05-06:
I meant from one terminal you will run OpenOCD same as before 
```
openocd -l openocdH2_log.txt  -d3 -f board/esp32h2-builtin.cfg
```

And from the other terminal
```
telnet localhost 4444
halt
flash probe 0
```

Reproducible app = Minimal application that once I loaded it, I will be able to reproduce the same issue. You can share the main.c file along with the additional configs.

riscy00, 2024-05-06:
I just installed telnet into Window 11 based on <dism /online /enable-feature /featurename:TelnetClient> (based on advise from LLM Gemini) via admin cmd terminal, so I can use telnet command above.
----------------------------------------------------------------------
In VSC terminal

PS E:\016_Work_ESP32\010_Example\BlinkH2_ESPIDF\WorkspaceH2\TestH2_GPIO_4B>
 *  History restored 

dH2_log.txt  -d3 -f board/esp32h2-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
----------------------------------------------------------------------
Then on cmd.exe, I typed 
telnet localhost 4444        // connected
halt
flash probe 0

and I get this under telnet host
s8 = 0x0
s9 = 0x0
s10 = 0x0
s11 = 0x0
t3 = 0x0
t4 = 0x0
t5 = 0x0
t6 = 0x0
pc = 0x40800000
mstatus = 0x1800
mepc = 0x40800000
mcause = 0x5
Failed to wait algorithm (-302)!
Algorithm run failed (-302)!
Failed to run flasher stub (-302)!
Failed to get flash mappings (-302)!
Target is already running an algorithm
Failed to start algorithm (-4)!
Failed to run flasher stub (-4)!
Target is already running an algorithm
Failed to start algorithm (-4)!
Failed to run flasher stub (-4)!
Failed to probe flash, size 0 KB
C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
>
>
>

I will send you file kit in a moment.

riscy00, 2024-05-06:
[TestH2_GPIO_4B.zip](https://github.com/espressif/openocd-esp32/files/15223751/TestH2_GPIO_4B.zip)
I have removed the fmt library since I think you can install fmt component from your end. 
Anything else you need?

erhankur, 2024-05-06:
I will try and turn back to you. 
In the meantime can you set flash size as 4MB and try again? 

[image]

riscy00, 2024-05-06:
I use VSC menuconfig (5.2.1) and use search flash and set to 4MB (see attached), restarted VSC and checked
[image]
 it 4MB there.
---------------------------------------------------------
telnet localhost 4444
halt
flash probe 0
---------------------------------------------------------
<snipped by terminal>
tp = 0x40801488
t0 = 0x4000af8c
t1 = 0xffffffe0
t2 = 0x0
s0 = 0x8
s1 = 0x1
a0 = 0x5
a1 = 0xffffffff
a2 = 0x40804794
a3 = 0x1
a4 = 0x1c
a5 = 0x8
a6 = 0x0
a7 = 0x4080110c
s2 = 0x0
s3 = 0x0
s4 = 0x0
s5 = 0x0
s6 = 0x0
s7 = 0x0
s8 = 0x0
s9 = 0x0
s10 = 0x0
s11 = 0x0
t3 = 0x0
t4 = 0x0
t5 = 0x0
t6 = 0x0
pc = 0x40800000
mstatus = 0x1800
mepc = 0x40800000
mcause = 0x5
Failed to wait algorithm (-302)!
Algorithm run failed (-302)!
Failed to run flasher stub (-302)!
Failed to get flash mappings (-302)!
Target is already running an algorithm
Failed to start algorithm (-4)!
Failed to run flasher stub (-4)!
Target is already running an algorithm
Failed to start algorithm (-4)!
Failed to run flasher stub (-4)!
Failed to probe flash, size 0 KB
C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:\Users\riscy\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
>

erhankur, 2024-05-06:
On Windows, after the target reset, OpenOCD may lose communication and you may need to hard reset and re-connect.  Related to https://github.com/espressif/openocd-esp32/issues/316

Please add the below configs to your `sdkconfig. defaults` file, rebuild and load the application again. And try debugging. In each try, make sure the OpenOCD connection is alive and no error message is printed.

```
CONFIG_IDF_TARGET="esp32h2"
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
CONFIG_ESP_SYSTEM_PMP_IDRAM_SPLIT=n
```

riscy00, 2024-05-07:
[ESP32HP_Result.pdf](https://github.com/espressif/openocd-esp32/files/15228348/ESP32HP_Result.pdf)
Please find attached as the result of sdkconfig adjustment, only change is the CONFIG_ESP_SYSTEM_PMP_IDRAM_SPLIT=n while other are correct. This is 4B file. 
I'm familiar with step to disconnect and restart VSC app which has to be performed for each flash and debug session since it lacks way to reset the OPENOCD in a similar fashion as eclipse IDE. It is hard to see from the documentation to use command .
I have read the #316 but I think this is different, it did say they have flash and debug working under JTAG/USB where in my case it does not.

erhankur, 2024-05-07:
I spent some time experimenting with VScode IDE debugging. Not bad as the first time.  Here are my notes; 

- When you see the below error, you can re-try. In the end, it works. 
```
ERROR: Unable to start debugging. Failed to find thread 1 for break event
```
- Sometimes I noticed that, OpenOCD process kept open in the background. You need to kill it manually to restart new debug session.
- If you want to stop at the `app_main`, you need to set the breakpoint from the IDE manually. 
- If you enable OpenOCD verbose logs to the file, it will be created under the gdb.exe folder.
 
 I am sharing my JSON files just for reference. Windows support is not ready yet.

If you still have the same problem, please enable OpenOCD logs as I did, and send me the log file.

settings.json
```
{
  "esp_idf_path"              : "${env:IDF_PATH}",
  "esp_idf_openocd_version"   : "v0.12.0-esp32-20240318",
  "esp_idf_gdb_version"       : "14.2_20240403",
  "esp_idf_python_version"    : "idf5.4_py3.11_env",
  "esp_idf_riscv_gdb"         : "riscv32-esp-elf-gdb/${config:esp_idf_gdb_version}/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb",
  "esp_idf_xtensa_gdb"        : "xtensa-esp-elf-gdb/${config:esp_idf_gdb_version}/xtensa-esp-elf-gdb/bin/xtensa-esp32-elf-gdb",
  "esp_idf_toolchain_root"    : "${env:HOME}/.espressif",
  "esp_idf_openocd_path"      : "${config:esp_idf_toolchain_root}/tools/openocd-esp32/${config:esp_idf_openocd_version}/openocd-esp32/bin",

  //app specific variables based on the target architecture
  "esp_idf_app_elf_file"      : "TestH2_Code_5A.elf",
  "esp_idf_app_gdb"           : "${config:esp_idf_toolchain_root}/tools/${config:esp_idf_riscv_gdb}",
  "esp_idf_app_openocd"       : "${config:esp_idf_openocd_path}",
  "esp_idf_openocd_config"    : "-f board/esp32h2-builtin.cfg -d3 -locd-log.txt",
}
```
launch.json
```
{
  "version": "0.2.0",
  "configurations": [
    {
        "name": "ESP-IDF Debug",
        "type": "cppdbg",
        "request": "launch",
        "program": "${workspaceFolder}/build/${config:esp_idf_app_elf_file}",
        "args": [],
        "cwd": "${workspaceFolder}",
        "externalConsole": false,
        "MIMode": "gdb",
        "miDebuggerPath": "${config:esp_idf_app_gdb}",
        "miDebuggerServerAddress": ":3333",
        "useExtendedRemote": true,
        "debugServerPath": "${config:esp_idf_app_openocd}/openocd",
        "debugServerArgs": "${config:esp_idf_openocd_config}",
        "filterStderr": true,
        "filterStdout": false,
        "launchCompleteCommand": "None",
        "setupCommands": [
          { "text": "set remotetimeout 10" },
        ],
        "postRemoteConnectCommands": [
          { "text": "mon reset halt" },
          { "text": "maintenance flush register-cache"},
        ],
        "stopAtConnect": false,
        "logging": {
          "exceptions": false,
          "engineLogging": true,
          "moduleLoad": false,
          "programOutput": true,
          "trace": true,
          "traceResponse": true
        },
    }
  ]
}
```

[image]

riscy00, 2024-05-08:
Thank you for looking into this case, the 5B file kit is working from my end which is why I send it to compare between 5B and 4B in case it may help to determine the root cause. But I do have issue to restart debug or flash session. Can you advise how to kill it manually and what command to use (terminal or Command Palette?), I am keen to try this. 

I wish to note the 4B does not flash and debug, which is the whole point of this issue.

The screenshot comes from an Apple PC. "Windows support is not ready yet." The documentation seems to have supported Windows VSC for some time, and ESP_IDF works with Windows. so I'm quite confused by this statement. out of curiosity, when support be ready?

I do more test with above later on this week.

erhankur, 2024-05-08:
Sorry for the confusion. I meant my example JSON files do not support Windows yet. Of course, esp-idf extension works on Windows.

I upload the application via `idf.py flash` and then start debugging. I haven't seen an OpenOCD error so far. Also with your 4B test. app. Stop and re-start debugging may require 2-3 tries. But as I said, in the end a new debug session starts.

[image]

So I don't see a problem related to OpenOCD. I am closing this issue. 

Recheck your JSON files and compare them with the ones I shared above. 

If you see a problem related to ESP-IDF VScode extension please create a new issue here. https://github.com/espressif/vscode-esp-idf-extension

Later if you have an OpenOCD error, please share the logs so that we can have a look in detail.

closed 2024-05-08 by erhankur

riscy00, 2024-05-08:
Can you advise how to kill it manually and what command to use (terminal or Command Palette?), I am keen to try this.

erhankur, 2024-05-08:
On Windows I think you can check task manager to see if there is a stale OpenOCD process remaining from the previous sessions.

Or from the command line; 

https://www.comparitech.com/net-admin/powershell-kill-process-command-tutorial/

Google it; how to kill process in Windows command line

Note: I didn't see stale OpenOCD process on my windows tests. Maybe there is a bug on MacOS only.

erhankur, 2024-05-08:
@riscy00 Check this [esp-vscode-debug-template](https://github.com/erhankur/esp-vscode-debug-template)