espressif/openocd-esp32 #324: Unable to halt ESP32S3 using FTDI debug probe

sach-edna, 2024-04-30:
### Development Kit

ESP32S3-DEVKITC-1

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

ftdi-c232hm

### OpenOCD version

 v0.12.0-esp32-20240318

### Operating System

Windows 11

### Using an IDE ?

No

### OpenOCD command line

openocd -f board/esp32s3-ftdi.cfg -c \"init;reset halt;\"

### JTAG Clock Speed

Any (20000 - 1 kHz)

### ESP-IDF version

ESP-IDF v5.2.1-443-g2ba5320112

### Problem Description

1. Build and flash blink example using uart. (or any other project, just bring ESP32S3 in known state
2. Try to open openocd using `openocd -f board/esp32s3-ftdi.cfg`
3. When trying to attatch with GDB, or when adding `-c "init; reset halt"` to the openocd command (so restart openocd with `openocd -f board/esp32s3-ftdi.cfg -c "init; reset halt"`, ESP32S3 does not halt.

### Expected behavior

- ESP32S3-DEVKITC-1 does not stop with error when executing "init; reset halt;" command
- ESP32S3-DEVKITC-1 allows gdb debugging using C232HM ftdi probe

erhankur, 2024-05-02:
@sach-edna I don't have that cable but I have ESP-PROG with the same FTDI chip. Nothing wrong with it. 
Do you see the same behavior with any ESP32 devkit or with ESP-PROG. (I hope you have)
Also can you run OpenOCD without `init, reset halt` commands and connect from telnet. Then what is happening when you send the same commands? (reset, halt, reset halt) 
Then please share the log file with -d4 option this time.

sach-edna, 2024-05-03:
@erhankur Thank you very much for picking up this issue. The chips used in both debug probes are very similar, c232hm cable uses [`DS_FT2232H`](https://ftdichip.com/wp-content/uploads/2024/03/DS_FT2232H.pdf), ESP-PROG uses the [`FT2232HL`](https://ftdichip.com/wp-content/uploads/2024/03/DS_FT2232H.pdf). They seem similar except for the 2 channels on the ESP-PROG chip.

I will also add a log file with -d4 option and the commands executed via telnet. I had to split the logs because of github size limits.
[openocd_log_d4_000.txt](https://github.com/espressif/openocd-esp32/files/15199299/openocd_log_d4_000.txt)
[openocd_log_d4_001.txt](https://github.com/espressif/openocd-esp32/files/15199304/openocd_log_d4_001.txt)
[openocd_log_d4_002.txt](https://github.com/espressif/openocd-esp32/files/15199308/openocd_log_d4_002.txt)
[openocd_log_d4_003.txt](https://github.com/espressif/openocd-esp32/files/15199318/openocd_log_d4_003.txt)

If there is anything I can do to help, please let me know.

sach-edna, 2024-05-03:
Update:

I got it to work, using the `board/esp32s3-ftdi.cfg`. Looking more into detail it is possible to use the `interface/ftdi/c232hm.cfg`, but it is required to add an override in your config file: `ftdi layout_init 0x0008 0xF00B` since the default c232hm layout is different. 

@erhankur thank you for the help!

erhankur, 2024-05-03:
Thanks for sharing the solution. It could be helpful in the future. So we can close this issue.

closed 2024-05-03 by erhankur