espressif/openocd-esp32 #325: ESP32-C6 - gdb requested a non-existing register (reg_num=73) when debugging from VS Code

QubaB, 2024-05-03:
### Development Kit

ESP32-C6 Development Board ESP32C6 Minimum System Board

### Module or chip used

ESP32-C6-WROOM-1

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.12.0-esp32-20230921

### Operating System

Windows

### Using an IDE ?

VS Code

### OpenOCD command line

openocd.exe -f board/esp32c6-builtin.cfg

### JTAG Clock Speed

24000 kHz

### ESP-IDF version

v5.2.1

### Problem Description

1. compiled ESP-ZIGBEE-SDK sample HA_on_off_light
2. trying to debug using JTAG
3. running debugger from command line is OK
4. running debugger from VS code (see launch.json for configuration) cause openocd error "Error: gdb requested a non-existing register (reg_num=73)"

[openocd_log.txt](https://github.com/espressif/openocd-esp32/files/15204990/openocd_log.txt)
[VSCode-gdb_log.txt](https://github.com/espressif/openocd-esp32/files/15204999/VSCode-gdb_log.txt)
[launch.json](https://github.com/espressif/openocd-esp32/files/15205002/launch.json)

5. If I uncomment in launch.json lines 
 ```
                { "text": "thb app_main"},
                 { "text": "c"},
```
then VS code provides error message 
> Unable to start debugging. Unexpected GDB output from command "-interpreter-exec console "c"". Unexpected ResultClass from MI Debugger. Expected 'done' but received 'running'.
6. If I use launch.json with
```
                { "text": "thb app_main"},
                { "text": "c"},
                  "ignoreFailures": true
                },
```
then VS code debuger stops to respond

[openocd-ignoreFailures_log.txt](https://github.com/espressif/openocd-esp32/files/15205067/openocd-ignoreFailures_log.txt)
[ignoreFailures_log.txt](https://github.com/espressif/openocd-esp32/files/15205069/ignoreFailures_log.txt)

### Expected behavior

Expected behavior is to be able to debug using JTAG

igrr, 2024-05-04:
It seems like it is trying to use xtensa-esp32-elf-gdb.exe while debugging a RISC-V target...

QubaB, 2024-05-04:
> It seems like it is trying to use xtensa-esp32-elf-gdb.exe while debugging a RISC-V target...
Dear @igrr, thank you for poiniting to problem.

I changed debugger to:
```
"miDebuggerPath": "C:/Espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb.exe",
```
and it started to work. I can debug in VS code now.

Attached is my configuration

[launch.json](https://github.com/espressif/openocd-esp32/files/15209791/launch.json)

closed 2024-05-04 by QubaB