espressif/openocd-esp32 #327: Error: esp_usb_jtag: could not find or open device!

aygh4266, 2024-06-06:
### OpenOCD version

v0.12.0-esp32-20240318

### Operating System

Windows 11

### Using an IDE ?

VSCode

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### ESP-IDF version

v5.2.2

### Problem Description

Hello everyone,

I am trying to debug my code on my ESP32-S3-DevKitM-1 using esp-prog. 
I think it is PID VID issue, since the PID VID on my PORT are follwing USB\VID_0403&PID_6010&MI_00 (libwdi autogenerated).

I have tried to reconfigure the esp32s3-builtin.cfg using these commands:

adapter driver ftdi
ftdi vid_pid 0x0403 0x6010
ftdi layout_init 0x0008 0x000b
ftdi layout_signal nSRST -data 0x0020 -oe 0x0020
adapter speed 20000
transport select jtag

set ESP32_ONLYCPU 1
source [find target/esp32s3.cfg]

but I still got another error: 

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32s3.cpu0: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Warn : [esp32s3.cpu0] Unexpected OCD_ID = ffffffff
Error: [esp32s3.cpu0] Examination failed
Warn : target esp32s3.cpu0 examination failed
Warn : [esp32s3.cpu1] Unexpected OCD_ID = ffffffff
Error: [esp32s3.cpu1] Examination failed
Warn : target esp32s3.cpu1 examination failed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

The Pin Connection is okay. Any help would be appreicated. 

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
C:\Users\Labor-Admin\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:\Users\Labor-Admin\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
```

### Expected behavior

OpenOCD connected successfully to the board

erhankur, 2024-06-06:
@aygh4266 Did you burn any efuse to use JTAG pins instead of USB-SERIAL-JTAG? By default usb-serial-jtag is enabled.

https://docs.espressif.com/projects/esp-idf/en/v5.2.2/esp32s3/api-guides/jtag-debugging/configure-other-jtag.html#configure-efuses

And, for the ESP-PROG you can use `-f board/esp32s3-ftdi.cfg`

aygh4266, 2024-06-06:
> @aygh4266 Did you burn any efuse to use JTAG pins instead of USB-SERIAL-JTAG? By default usb-serial-jtag is enabled. [...]

OK I understand it better right now. So basically there are two option to debug the esp idf. Using USB-Serial-JTAG by default or using JTAG with external debugger like ESP PROG. So before I burn the efuse, I am trying to use the USB SERIAL JTAG but I still having issues. I tried this command openocd -f board/esp32s3-builtin.cfg for the serial builtin.

And got this output: 

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: esp_usb_jtag: could not find or open device!
C:\Users\Labor-Admin\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "C:\Users\Labor-Admin\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
PS C:\Users\Labor-Admin\esp\v5.2.2\esp-idf\GfA_Projects\update_ueber_usb_pc> openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (F4:12:FA:87:C4:D8)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40375D4B, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu1] Target halted, PC=0x403789D2, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)

aygh4266, 2024-06-06:
Update: I can flash the esp32 using the JTAG but can't debug. Maybe it is a VSCode issue, since i am using esp idf extension

erhankur, 2024-06-06:
We are about to release the new idf extension which will be faster than the current one. You can download it from here and give it a try. https://github.com/espressif/vscode-esp-idf-extension/pull/1207#issuecomment-2141767744 

Then you can report your debugging issue there or here with the verbose logs. 

As I see from your logs, you don't have an issue with the usb-serial-jtag.

erhankur, 2024-06-07:
@aygh4266 closing this. Feel free to re-open if you have any other issues.

PragatiGarg-eaton, 2024-12-26:
> Update: I can flash the esp32 using the JTAG but can't debug. Maybe it is a VSCode issue, since i am using esp idf extension

@erhankur @aygh4266 I’m also facing the same issue, and it’s not related to VSCode since I’ve tried running it through the command line as well.

PragatiGarg-eaton, 2024-12-27:
@erhankur Following are the debug logs: 
1. For cmd: >openocd -d3 -f board/esp32c3-builtin.cfg 
```
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 15 options.c:52 configuration_output_handler(): debug_level: 3User : 4 15 options.c:52 configuration_output_handler():
Debug: 5 15 options.c:346 parse_cmdline_args(): ARGV[0] = "openocd"
Debug: 6 30 options.c:346 parse_cmdline_args(): ARGV[1] = "-d3"
Debug: 7 34 options.c:346 parse_cmdline_args(): ARGV[2] = "-f"
Debug: 8 36 options.c:346 parse_cmdline_args(): ARGV[3] = "board/esp32c3-builtin.cfg"
Debug: 9 39 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 10 39 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 11 48 options.c:235 add_default_dirs(): exepath=C:/OpenOCD-esp32/821/openocd-esp32/bin
Debug: 12 51 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 14 59 configuration.c:33 add_script_search_dir(): adding C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/site
Debug: 15 59 configuration.c:33 add_script_search_dir(): adding C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts
Debug: 16 71 command.c:152 script_debug(): command - ocd_find board/esp32c3-builtin.cfg
Debug: 17 75 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/board/esp32c3-builtin.cfg
Debug: 18 83 command.c:152 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg
Debug: 19 89 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/interface/esp_usb_jtag.cfg
Debug: 20 96 command.c:152 script_debug(): command - adapter driver esp_usb_jtag
Info : 21 99 transport.c:106 allow_transports(): only one transport option; autoselecting 'jtag'
Debug: 22 104 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 23 108 esp_usb_jtag.c:891 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 24 110 command.c:152 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 25 118 esp_usb_jtag.c:902 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 26 120 command.c:152 script_debug(): command - adapter speed 40000
Debug: 27 126 adapter.c:262 adapter_config_khz(): handle adapter khz
Debug: 28 129 adapter.c:226 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 29 134 adapter.c:226 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 30 139 command.c:152 script_debug(): command - ocd_find target/esp32c3.cfg
Debug: 31 144 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/target/esp32c3.cfg  
Debug: 32 151 command.c:152 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 33 155 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/target/esp_common.cfg
Debug: 34 161 command.c:152 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 35 164 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/bitsbytes.tcl
Debug: 36 171 command.c:152 script_debug(): command - ocd_find memory.tcl
Debug: 37 176 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/memory.tcl
Debug: 38 182 command.c:152 script_debug(): command - ocd_find mmr_helpers.tcl
Debug: 39 186 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/mmr_helpers.tcl
Debug: 40 188 command.c:152 script_debug(): command - ocd_find target/esp_version.cfg
Debug: 41 197 configuration.c:88 find_file(): found C:/OpenOCD-esp32/821/openocd-esp32/bin/../share/openocd/scripts/target/esp_version.cfg
Debug: 42 204 command.c:152 script_debug(): command - version
Debug: 43 206 command.c:152 script_debug(): command - add_help_text program_esp write an image to flash, address is only required for binary images. verify, reset, exit, compress, restore_clock and encrypt are optional
Debug: 44 214 command.c:152 script_debug(): command - add_usage_text program_esp <filename> [address] [verify] [reset] [exit] [compress] [no_clock_boost] [restore_clock] [encrypt]
Debug: 45 225 command.c:152 script_debug(): command - add_help_text program_esp_bins write all the images at address specified in flasher_args.json generated while building idf project
Debug: 46 232 command.c:152 script_debug(): command - add_usage_text program_esp_bins <build_dir> flasher_args.json [verify] [reset] [exit] [compress] [no_clock_boost] [restore_clock]
Debug: 47 241 command.c:152 script_debug(): command - add_help_text esp_get_mac Print MAC address of the chip. Use a `format` argument to return formatted MAC value
Debug: 48 248 command.c:152 script_debug(): command - add_usage_text esp_get_mac [format]
Debug: 49 252 command.c:152 script_debug(): command - jtag newtap esp32c3 tap0 -irlen 5 -expected-id 0x00005c25
Debug: 50 258 tcl.c:405 handle_jtag_newtap_args(): Creating New Tap, Chip: esp32c3, Tap: tap0, Dotted: esp32c3.tap0, 4 params
Debug: 51 259 core.c:1476 jtag_tap_init(): Created Tap: esp32c3.tap0 @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
[... cut ...]
Debug: 119 644 target.c:2139 target_free_all_working_areas_restore(): freeing all working areas
```

2. > openocd -d3 -c 'set ESP_RTOS none' -c 'set ESP_FLASH_SIZE 0' -f target/esp32.cfg -c "init; reset halt" 
``` 
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 3 0 options.c:52 configuration_output_handler(): debug_level: 3User : 4 16 options.c:52 configuration_output_handler():
Unexpected command line argument: ESP_RTOS
```

erhankur, 2024-12-27:
@PragatiGarg-eaton Please try installing the drivers using the online installer: https://dl.espressif.com/dl/esp-idf/

For the C3, we have a customized WinUSB driver. You can install it from the link.

And for the other error, try this command

```
openocd -d3 -c 'set ESP_RTOS none' -c 'set ESP_FLASH_SIZE 0' -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "init; reset halt"
```

You need to specify both an interface file and the target.

PragatiGarg-eaton, 2024-12-27:
> @PragatiGarg-eaton Please try installing the drivers using the online installer: https://dl.espressif.com/dl/esp-idf/
>
> For the C3, we have a customized WinUSB driver. You can install it from the [...]

- I already tried by installing WinUSB driver referring: https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-usb-drivers
- and https://visualgdb.com/UsbDriverTool/
- [image]
[image]
 But same results are observed.
- In addition to this, I came accross the following hardware changes- 
- you need to rotate those resistors 90 degrees: move them from the red to the green position in this image:
c3_builtin.png
c3_builtin.png (246.05 KiB) Viewed 21760 times
You probably also want to disconnect the RxD/TxD pins from the CP2104 to the ESP32C3 if you want to use those pins. You can do that by removing R21/R22, I marked those in blue.
- https://www.esp32.com/viewtopic.php?f=12&t=21007#p76800

erhankur, 2024-12-27:
Ah, yes, I forgot that the USB connector is connected to the serial bridge, not a built-in USB device.

So, you will need a cable like this: https://www.adafruit.com/product/4448

You don't need to touch the resistors. Just connect the green and white wires to GPIO19 and GPIO18, respectively.

PragatiGarg-eaton, 2024-12-27:
> Ah, yes, I forgot that the USB connector is connected to the serial bridge, not a built-in USB device.
> 
> So, you will need a cable like this: https://www.adafruit.com/product/4448
> 
> You don't need to touch the resistors. Just connect the green and white wires to GPIO19 and GPIO18, respectively.

Yeah, I already gave it a try, it is not working even after the USB/ttl connections.

erhankur, 2024-12-27:
Please make sure to use a breakout cable instead of a USB/TTL converter.

PragatiGarg-eaton, 2024-12-27:
Okay, I'll check with this once.

PragatiGarg-eaton, 2024-12-30:
> Ah, yes, I forgot that the USB connector is connected to the serial bridge, not a built-in USB device.
> 
> So, you will need a cable like this: https://www.adafruit.com/product/4448
> 
> You don't need to touch the resistors. Just connect the green and white wires to GPIO19 and GPIO18, respectively.

Hi @erhankur I tried by connecting breakout cable to GPIO18/19 as you mentioned and as per https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html 

- But I am getting unknown USB device issue now, 

[image]

- I also tried to install the driver through zadig, but it is failing too, since no USB ID is present and it is unable to identify the hardware here- 

[image]

[image]

- Output from the 
[image]

cc: @igrr

erhankur, 2025-01-02:
Please check  [here](https://github.com/espressif/openocd-esp32/issues/230#issuecomment-1134393392) to make sure driver is installed correctly. And the other comments may be useful for you.

erhankur, 2025-01-12:
@PragatiGarg-eaton any update on your side?

PragatiGarg-eaton, 2025-01-13:
@erhankur I checked, and everything seems fine with the drivers. However, I found a note in the ESP32 documentation here - https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/index.html 
[image]

That's why I am getting the unknown USB issue. I'm trying out the external USB Prog option rn. I'll keep you posted.

PragatiGarg-eaton, 2025-01-21:
Hi @erhankur I retried debugging it using esp-prog in Eclipse IDE as well as VSCode, and I am getting below error now- 

[image]

PragatiGarg-eaton, 2025-01-28:
> hi [@PragatiGarg-eaton](https://github.com/PragatiGarg-eaton), could you please collect verbose logs from openocd (-d3)?

@sobuch Please find the attached logs- [esp_debugger_logs.txt](https://github.com/user-attachments/files/18577734/esp_debugger_logs.txt)

erhankur, 2025-01-29:
```
Debug: 4374 45027 gdb_server.c:401 gdb_log_outgoing_packet(): [esp32c3] {1} sending packet: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02
Error: 4375 45034 gdb_server.c:509 gdb_put_packet_inner(): GDB missing ack(2) - assumed good
```

These errors indicate a timing issue and usually happen when the target reads from flash and cannot respond to GDB within the timeout period. You can try:

1 - Increase the adapter speed from 5000 to 20000. 
It is in the `openocd-esp32/share/openocd/scripts/interface/ftdi/esp32_devkitj_v1.cfg`

2- increase the GDB remote timeout to 10 or more to be on the safe side.
Add `set remotetimeout 10` under the `Initialization Commands`
https://docs.espressif.com/projects/espressif-ide/en/latest/openocddebugging.html#startup-tab

erhankur, 2025-02-25:
Closed due to inactivity. @PragatiGarg-eaton  feel free to re-open if you have an issue after trying above suggestions.

closed 2025-02-25 by erhankur