espressif/openocd-esp32 #328: Error LIBUSB_ERROR_NOT_FOUND and OPENOCD path does not exist

sfeutrier, 2024-06-06:
### Development Kit

ESP32C3-DevkitC

### Module or chip used

ESP32C3

### Debug Adapter

builtin

### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20230921

### Operating System

Windows

### Using an IDE ?

VS Code

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.2.1

### Problem Description

Hello,

I try for days to make Debugging through VS Code works. But it is a real pain here. WTF.
I have used this https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html and this https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/index.html and even this https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/get-started/linux-macos-setup.html#get-started-set-up-tools
I have an USB cable wired correctly to my ESP32-C3 board. It is well detected by Windows.
But nothing else work when I try to launch the Debug session. If I try to launch, I have an issue with libusb that does not find my device.
Then I tried directly with openocd uisng "openocd -f board/esp32c3-builtin.cfg". Exactly the same story.
I used this link because the issue is mentionned. But it did not resolve anything: https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html
Looking back in the links, I found that I need to check the path of OPENOCD through "echo %OPENOCD_SCRIPTS%". And indeed, it does not work. So I went through the folder of esp-idf and did the install.bat and export.bat things: STILL THE SAME.

Can you tell me how I can simply finally debug my board?

Your welcome

### Debug Logs

```shell
PS C:\Users\simon\esp\v5.2.1\esp-idf> openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: esp_usb_jtag: could not find or open device!
C:\Users\simon\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error:
at file "C:\Users\simon\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
```

### Expected behavior

Debugging through builtin USB-JTAG interface with an ESP32C3-devkitc with VSCODE

sfeutrier, 2024-06-06:
Edit:
I found this comment in this issue: https://github.com/espressif/idf-installer/issues/133#issuecomment-1263204902
It says to use Zadig and to kind of "downgrade" the WinUSB driver... And it works...
I still don't now which of all steps I have done were necessary or not. But this last step was a key. But it is a shame that it is so complicate

erhankur, 2024-06-07:
@sfeutrier Sorry for the inconvenience. We try to keep our troubleshooting page up to date with the correct links here. I'll check them again. 

Do you have any other issues now?

sfeutrier, 2024-06-10:
No I do not have any other issue. I just notice how slow it is to step-by-step.
For futur viewer of this ticket, I suggest to go with the Eclipse ESP-IDF solution (https://github.com/espressif/idf-eclipse-plugin?tab=readme-ov-file#Installation). It is far better, easier, faster and very well-documented

closed 2024-06-10 by sfeutrier