espressif/openocd-esp32 #330: OpenOCD halts CPU when connecting to gdb-server

Stefasaurus, 2024-06-29:
### OpenOCD version

v0.12.0-esp32-20240318

### Operating System

Windows 10

### Using an IDE ?

VScode esp-idf extension

### OpenOCD command line

openocd -f esp32s3-builtin.cfg

### ESP-IDF version

v4.4

### Problem Description

During the development of an open source project, we wanted to use the gdb-server to extract data from esp32 RAM. Then we fell into an issue, where the application successfully extracts the data, but when we connect to the gdb socket the target halts. This happens before sending any gdb commands over the tcp/ip gdb protocol. We wish to not make the target halt when the attach happens, so we would be happy if you could help us configure OpenOCD for esp32, so that the attachment does not halt the target. Thanks!

### Expected behavior

Custom OpenOCD 'cfg' file as to not halt the target when the attach happens.

erhankur, 2024-07-01:
To avoid halt, you can comment out https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp_common.cfg#L195 and https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp_common.cfg#L214

As the explanation says, halt is needed for the flash probe. That means, you need to disable the flash support. 

`openocd -c "set ESP_FLASH_SIZE 0" -f board/esp32s3-builtin.cfg`

erhankur, 2024-07-08:
Sorry no idea what is happening. If you can share the verbose OpenOCD logs, I can suggest something more.

```
openocd -c "set ESP_FLASH_SIZE 0" -f board/esp32s3-builtin.cfg -d3 -l log.txt
```

Stefasaurus, 2024-07-13:
Sorry for the wait @erhankur , I am attaching the log file.
[logfile.log](https://github.com/user-attachments/files/16202832/logfile.log)

erhankur, 2024-07-15:
Thanks for the log. We can comment out  configure_event_gdb_attach call inside configure_openocd_events.
```
proc configure_openocd_events { ARCH } {
	if { $ARCH == "riscv" } {
		configure_event_halted
		configure_shutdown_command
	}
	configure_event_examine_end
	configure_event_reset_assert_post
	# configure_event_gdb_attach
}
```
We disabled flash support so we don't need mem prot check anymore. After that, there is nothing to do in the gdb attach event.

erhankur, 2024-08-01:
@Stefasaurus Can we close this issue now? Do you have any other question?

erhankur, 2024-08-07:
Closed. Feel free to re-open if you have any other question.

closed 2024-08-07 by erhankur