espressif/openocd-esp32 #331: Flashing over JTAG: OpenOcd Target not halted!

ahammouri, 2024-08-01:
### Development Kit

ESP32

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

ESP-Prog

### OpenOCD version

v0.12.0-esp32-20240318

### Operating System

Windows

### Using an IDE ?

VS code

### OpenOCD command line

openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "adapter speed 26000; init; reset halt; program_esp_bins . flasher_args.json compress verify; reset; exit"  

### JTAG Clock Speed

26000

### ESP-IDF version

5.2.1

### Problem Description

Whenever I run the command openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "adapter speed 26000; init; reset halt; program_esp_bins . flasher_args.json compress verify; reset; exit" , the device fails stay in halt state I guess ? Which prevents programming over JTAG. I see the issue has been reported before but didn't found a fix to it really ? Happened on multiple units now. 

Logs : 

Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:40)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

Flash has finished. You can monitor with ESP-IDF: Monitor your device command
[/Build]
[OpenOCD]
Info : accepting 'tcl' connection on tcp/6666
Info : dropped 'tcl' connection

Info : accepting 't
cl' connection on tcp/6666

Flashing "build/bootloader/bootloader.bin" at 0x1000

** program_esp input args <0x1000 verify> **

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part
: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Target halted, PC=0x40080200, debug_reason=00000000

Info : [esp32.cpu0] Halt cause (4) - (Level-1 interrupt)

Info : [esp32.cpu0] Reset cause (1) - (Power on reset)

Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)

Info : [esp32.cpu1] requesting target halt and executing a soft reset

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Debug controller was reset.
Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu1] T
arget halted, PC=0x40000400, debug_reason=00000000

Info : [esp32.cpu1] Reset cause (14) - (CPU1 reset by CPU0)

Info : [esp32.cpu0] Reset
 cause (1) - (Power on reset)

❌ Error: timed out while waiting for target halted

Info : [esp32.cpu0] Target halted, PC=0
❌ x4000C2F6, debug_reason=00000000
Error: xtensa_wait_algorithm: not halted 0, pc 0x4000c2f6, ps 0x60025
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run fl
asher stub (-302)!
Warn : Failed to get flash mappings (-302)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorit
❌ hm (-4)!
Error: Failed to run flasher stub (-4)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!

❌ Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32.cpu0.flash'!

** Flashing Failed **
-1

[/OpenOCD]
[Flash]
Failed to flash the device (JTag), please try again [got response: '-1', expecting: '0']

### Expected behavior

Flashing over JTAG to work.

erhankur, 2024-08-02:
Are you able to flash the application to broken units using esptool? Or erasing flash from esptool fixes the issue?

ahammouri, 2024-08-02:
Thank you for your reply. Yes, I can erase units using esptool. After I did that, I was able to flash them again using JTAG. Curious to know why it happened in the first place 🤔

erhankur, 2024-08-02:
So, can you flash over and over again without waiting some time? Is there anything connected to JTAG pins? Which IDF components do you have in your application? Can you also share the OpenOCD logs with -d3 option.

ahammouri, 2024-08-02:
I can flash over and over yes. Although a new unit just stopped working now and I cant flash to it anymore. Logs for that unit is attached here. Nothing connected to the JTAG pins. 

[esp32flasherror 1.txt](https://github.com/user-attachments/files/16468569/esp32flasherror.1.txt)

erhankur, 2024-08-02:
What do you see in IDF console when the unit stops working? And can you share your component list?

Ahmad-myenergi, 2024-08-02:
Monitoring works fine, but I get the below error :

❌ Error: timed out while waiting for target halted

Info : [esp32.cpu0] Target halted, PC=0
❌ x4000C2F6, debug_reason=00000000
Error: xtensa_wait_algorithm: not halted 0, pc 0x4000c2f6, ps 0x60025
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run fl
asher stub (-302)!
Warn : Failed to get flash mappings (-302)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorit
❌ hm (-4)!
Error: Failed to run flasher stub (-4)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!

❌ Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32.cpu0.flash'!

Sorry I am new to this, where can I find my component list.

ahammouri, 2024-08-02:
Monitoring works fine, but I get the below error :

❌ Error: timed out while waiting for target halted

Info : [esp32.cpu0] Target halted, PC=0
❌ x4000C2F6, debug_reason=00000000
Error: xtensa_wait_algorithm: not halted 0, pc 0x4000c2f6, ps 0x60025
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run fl
asher stub (-302)!
Warn : Failed to get flash mappings (-302)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorit
❌ hm (-4)!
Error: Failed to run flasher stub (-4)!

❌ Error: Target is already running an algorithm
Error: Failed to start algorithm (-4)!

❌ Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32.cpu0.flash'!

Sorry I am new to this, where can I find my component list.

erhankur, 2024-08-02:
Can you try the below command adding the `no_clock_boost` option?
```
openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "adapter speed 26000; init; reset halt; program_esp_bins . flasher_args.json compress verify no_clock_boost; reset; exit"
```

ahammouri, 2024-08-02:
That did not work I am afraid. Same error 

[noclockboost.txt](https://github.com/user-attachments/files/16470392/noclockboost.txt)

erhankur, 2024-08-02:
Can you prepare a test application so that I can reproduce the issue here? Alternatively, you can send your app as is to my email. This would be the fastest way to understand the reason. Some other users have also complained about this, but I still couldn't reproduce it.

ahammouri, 2024-08-26:
Hi 
To close the loop. We figured it was a change to the partition table that caused the issue. Jumping from an old firmware (with old partition tables) to a new ones (with updated partition tables) caused the issue. We haven't seen the issue again since we moved to the updated partition table.

closed 2024-08-26 by ahammouri