espressif/openocd-esp32 #332: openocd currently not disables watchdog?

denizzzka, 2024-08-17:
### Development Kit

LuatOS ESP32-C3

### Module or chip used

ESP32c3

### Debug Adapter

directly connected USB

### OpenOCD version

v0.12.0-esp32-20240726 (2024-07-26-15:19)

### Operating System

Linux

### Using an IDE ?

CLI

### OpenOCD command line

openocd -c "telnet_port 5449" -c "gdb_port 3117" -f interface/esp_usb_jtag.cfg -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.3

### Problem Description

It seems, openocd currently not disables watchdog timer?

Iit is leads to panic condition after debug seesion stop on the breakpoint if user tries to execute next step manually

Temporary fix: just add `CONFIG_INT_WDT=n` into the project config 

### Expected behavior

Step by step debugging should work without `CONFIG_INT_WDT=n`

erhankur, 2024-08-22:
@denizzzka we do disable the watchdogs in every halt with this function. https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32c3.cfg#L26

From OpenOCD's perspective, reason of the halt doesn't matter. WDTs will be disabled after step, breakpoint or manual halt command.

Perhaps you could share your application log output to see what happens during the step. You should have seen the reset reason during the next boot. Additionally, you can monitor the OpenOCD output by running it with the -d3 option.

`openocd -c "telnet_port 5449" -c "gdb_port 3117" -f board/esp32c3-builtin.cfg -d3`

erhankur, 2024-09-14:
Closing due to inactivity.

erhankur, 2024-09-30:
I encountered the same issue with our new chip, ESP32-P4. It is probably related to the common layer and needs to be investigated for all RISCV chips.

denizzzka, 2024-09-30:
@erhankur now issue labels are misleading

erhankur, 2024-10-01:
@denizzzka can you replace `esp32c3_wdt_disable` [function](https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32c3.cfg#L26) with below and test again. 

```
proc esp32c3_wdt_disable { } {
	# Halt event can occur during config phase (before "init" is done).
	# Ignore it since mww commands don't work at that time.
	if { [string compare [command mode] config] == 0 } {
		return
	}

	# Timer Group 0 & 1 WDTs
	mww 0x6001f064 0x50D83AA1
	mww 0x6001F048 0
	# Clear TG0 wdt pending interrupt
	mww 0x6001F07C 0x2
	mww 0x60020064 0x50D83AA1
	mww 0x60020048 0
	# Clear TG1 wdt pending interrupt
	mww 0x6002007C 0x2
	# RTC WDT
	mww 0x600080a8 0x50D83AA1
	mww 0x60008090 0
	# SWD
	mww 0x600080b0 0x8F1D312A
	mww 0x600080ac 0x84B00000
```

denizzzka, 2024-10-01:
@erhankur no, I've already taken my test bench apart

erhankur, 2024-10-01:
OK.  if you have a chance to test please let us know the result.  I will add the changes to the master branch. Clearing the pending interrupts helped to the ESP32-P4.  I can not reproduce the issue on ESP32-C3. So closing this again until you come up with a bad news.

closed 2024-10-01 by erhankur