espressif/openocd-esp32 #334: ESP32S3-WROOM-2-32R8 and Arduino IDE Debugger

microfoundry, 2024-09-07:
### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)

### Operating System

Windows 11

### Using an IDE ?

Arduino 2.3.2

### OpenOCD command line

...local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.12.0-esp32-20240318/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\...\\sketchname" -f "C:/Program Files/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32s3-builtin.cfg

### ESP-IDF version

v5.x

### Problem Description

Launching OpenOCD with ESP32S3-WROOM-2-32R8 fails to identify memory

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E4:45:E0)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : starting gdb server for esp32s3.cpu0 on 50000
Info : Listening on port 50000 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037BBCE, debug_reason=00000001
Info : [esp32s3.cpu0] Reset cause (21) - (USB UART reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x4205F8AC, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (21) - (USB UART reset)
Info : accepting 'gdb' connection on tcp/50000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Warn : Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Error: Failed to get flash size!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

### Expected behavior

Additional Details in [original issue submitted](https://github.com/espressif/arduino-esp32/issues/10283) to the Espressif Arduino Core repository

erhankur, 2024-09-09:
Can you try 
```
openocd -f interface/ftdi/esp32_devkitj_v1.cfg -f target/esp32.cfg -c "adapter speed 26000; init; reset halt; program_esp_bins . flasher_args.json compress verify no_clock_boost; reset; exit"

erhankur, 2024-09-09:
Can you try 

```
 program_esp_bins . flasher_args.json compress verify no_clock_boost; reset; exit

erhankur, 2024-09-09:
We did a fix regarding ESP32-S3 flash read failure in the latest OpenOCD release. Please try with https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20240821

microfoundry, 2024-09-09:
Same - but a little different...

```
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
CDRTOSConfigure
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E4:45:E0)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : [esp32s3.cpu0] starting gdb server on 50000
Info : Listening on port 50000 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037BDEA, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (21) - (USB UART reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x420154F4, debug_reason=00000001
Info : [esp32s3.cpu1] Reset cause (21) - (USB UART reset)
Info : accepting 'gdb' connection on tcp/50000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

erhankur, 2024-09-09:
Can you add additional command to the end of OpenOCD start script? 

`openocd ..... -f board/esp32s3-builtin.cfg -c "esp stub_log on"`

You should see a logs something like;

```
STUB_D: cpu_freq:240 Mhz
STUB_D: DATA 0x3ffe4000..0x3ffe536a
STUB_D: BSS 0x3ffe536c..0x3ffe63ad
STUB_D: cmd 4:FLASH_MAP_GET
STUB_D: stub_flash_handler arg1 ffffffff, arg2 1073637296
STUB_I: Cache disable CPU1: 0x2e 0
STUB_D: stub_flash_get_size: ENTER
STUB_D: flash 204018, cs 200000, bs 10000, ss 1000, ps 100, sm ffff
STUB_D: Flash ID read 184020
STUB_D: Flash ID 18, size 16384 KB
STUB_D: stub_flash_get_map: 0xffffffff 0x3ffe67b0
STUB_D: start_page: 1 map_src: 0 map_size: 8020 page_cnt: 1 flash_page: 0 map_ptr: 3f418000
STUB_D: Found partition 0, m 0x50aa, t 0x1, st 0x2, l 'nvs'
STUB_D: start_page: 1 map_src: 0 map_size: 8040 page_cnt: 1 flash_page: 0 map_ptr: 3f418020
STUB_D: Found partition 1, m 0x50aa, t 0x1, st 0x1, l 'phy_init'
STUB_D: start_page: 1 map_src: 0 map_size: 8060 page_cnt: 1 flash_page: 0 map_ptr: 3f418040
STUB_D: Found partition 2, m 0x50aa, t 0x0, st 0x0, l 'factory'
STUB_I: Found app partition: 'factory' 1024 KB @ 0x10000
STUB_D: start_page: 1 map_src: 10000 map_size: 18 page_cnt: 1 flash_page: 1 map_ptr: 3f410000
STUB_I: Found app image: magic 0xe9, 5 segments, entry @ 0x40081264
STUB_D: start_page: 1 map_src: 10000 map_size: 20 page_cnt: 1 flash_page: 1 map_ptr: 3f410018
STUB_I: App segment 0: 39288 bytes @ 0x3f400020
STUB_I: Mapped segment 0: 39288 bytes @ 0x10020 -> 0x3f400020
STUB_D: start_page: 1 map_src: 10000 map_size: 99a0 page_cnt: 1 flash_page: 1 map_ptr: 3f419998
STUB_I: App segment 1: 10576 bytes @ 0x3ffb0000
STUB_D: start_page: 1 map_src: 10000 map_size: c2f8 page_cnt: 1 flash_page: 1 map_ptr: 3f41c2f0
STUB_I: App segment 2: 15648 bytes @ 0x40080000
STUB_D: start_page: 1 map_src: 20000 map_size: 20 page_cnt: 1 flash_page: 2 map_ptr: 3f410018
STUB_I: App segment 3: 93652 bytes @ 0x400d0020
STUB_I: Mapped segment 1: 93652 bytes @ 0x20020 -> 0x400d0020
STUB_D: start_page: 1 map_src: 30000 map_size: 6dfc page_cnt: 1 flash_page: 3 map_ptr: 3f416df4
STUB_I: App segment 4: 40400 bytes @ 0x40083d20
STUB_I: Cache restored CPU1: 0x2e 1
STUB_D: exit 0
Info : Flash mapping 0: 0x10020 -> 0x3f400020, 38 KB
Info : Flash mapping 1: 0x20020 -> 0x400d0020, 91 KB
```

microfoundry, 2024-09-09:
As per your request:

```
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
CDRTOSConfigure
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : [esp32s3.cpu0] Stub logs enabled!
Info : [esp32s3.cpu1] Stub logs enabled!
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : esp_usb_jtag: serial (7C:DF:A1:E4:45:E0)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : [esp32s3.cpu0] starting gdb server on 50000
Info : Listening on port 50000 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x4037BDEA, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (21) - (USB UART reset)
Info : [esp32s3.cpu1] Target halted, PC=0x400559E3, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (21) - (USB UART reset)
Info : accepting 'gdb' connection on tcp/50000
Info : [esp32s3.cpu0] Target halted, PC=0x4037BDEA, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x42076815, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038F906, debug_reason=00000001
STUB_D: cpu_freq:160 Mhz
STUB_D: DATA 0x3fca0000..0x3fca11da
STUB_D: BSS 0x3fca11dc..0x3fca22f1
STUB_D: cmd 4:FLASH_MAP_GET
STUB_D: stub_flash_handler arg1 ffffffff, arg2 1070213364
STUB_I: Flash state prepared...
STUB_D: stub_flash_get_size: ENTER
STUB_D: flash c28039, cs 2000000, bs 10000, ss 1000, ps 100, sm ffff
STUB_D: Flash ID read 857300
STUB_E: Unsupported octal flash manufacturer
STUB_D: Flash ID 0, size 0 KB
STUB_E: Failed to get flash size!
STUB_D: exit 0
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

erhankur, 2024-09-09:
Thanks.

Error is `Unsupported octal flash manufacturer` I will check this internally and come back.

In the meantime you can skip flash support by adding `-c "set ESP_FLASH_SIZE 0"` command just before the board config file.

erhankur, 2024-09-09:
One more thing; Can you enable PSRAM support from the menuconfig and try again?

microfoundry, 2024-09-09:
I have PSRAM enabled as required for my app. But I believe the simple memory/debug tests I was performing against the various ESP modules was without.

erhankur, 2024-09-09:
I found a board and tried. Worked as expected. 
```
STUB_D: flash c28039, cs 200000, bs 10000, ss 1000, ps 100, sm ffff
STUB_D: Flash ID read 3980c2
STUB_D: Flash ID 39, size 32768 KB
STUB_D: stub_flash_get_map: 0xffffffff 0x3fca28f4
```
Please enable PSRAM in your simple tests as well and share the results. If it works, I’m afraid we will need to revisit that fix.

microfoundry, 2024-09-09:
This all started with the issue being raised in the Arduino Core project as I'm working within the Arduino IDE and the debugger failure to attach to the 32MB module. But a point was made in that thread to bring it to y'alls attention. Now with regards to the "enable PSRAM" request, it would seem that within the current ver of Arduino Core, selecting MCU ESP32S3-WROOM-2 will set the memory to Octal and enable Octal PSRAM. And If the flash is left on Octal, changing the PSRAM setting to anything other than Octal produces an error when compiling. Additionally, Picking a std ESP32S3 Dev Module will (not surprisingly) produce the same results. 

Regardless of all that, my project at hand does have PSRAM enabled and the app currently utilizes ~30%. So I can say with it on, the attempt to launch OpenOCD fails the same way.

Chatio502, 2024-10-01:
hola a todos necesito ayuda estoy utilizando Arduino IDE y un ESP32 pero al ejecutar cualquier codigo me da el siguiente error.   me pueden ayudar estoy iniciando y me gustaria que dieran por favor que pasos tengo que hacer para solucionarlo.

Waiting for gdb server to start...[2024-10-01T01:28:09.857Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\workstation2\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\openocd-esp32\\v0.12.0-esp32-20240821/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\workstation2\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024830-2920-149u6np.r90q\\sketch_sep30a" -f "C:/Program Files/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
CDRTOSConfigure
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: unable to open ftdi device with description '*', serial '*' at bus location '*'
C:/Users/workstation2/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240821/bin/../share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "C:/Users/workstation2/AppData/Local/Arduino15/packages/esp32/tools/openocd-esp32/v0.12.0-esp32-20240821/bin/../share/openocd/scripts/target/esp_common.cfg", line 9
[2024-10-01T01:28:10.761Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

picode98, 2026-03-31:
Makes sense. Let me know if there's any testing I can do to help.

picode98, 2026-06-08:
@erhankur Sorry for the delay; I can confirm that the issue appears to be fixed as of the latest commit to the master branch (f10ecef): the stub reads the 32MB memory size and sets up the rest of the memory map properly now, and the Arduino IDE debugger functions properly when the IDE's bundled version of openocd-esp32 is replaced with the build from this commit.

Thanks again for your assistance in resolving this issue.