espressif/openocd-esp32 #335: OpenOCD/JTAG Error: timed out while waiting for target halted

FrancoisB-HEX, 2024-09-09:
### Development Kit

Custom

### Module or chip used

ESP32-S3-WROOM-1-N16

### Debug Adapter

USB-JTAG (built-in)

### OpenOCD version

v0.12.0-esp32-20240318 and v0.12.0-esp32-20240726

### Operating System

Windows

### Using an IDE ?

Command Line

### OpenOCD command line

openocd -f esp32s3-builtin.cfg -c "program_esp build/X.bin 0x10000 verify"

### JTAG Clock Speed

default

### ESP-IDF version

v5.3-369-g466a392a76

### Problem Description

I see this issue with a larger binary that is using WiFi and Bluetooth (Nimble) on an ESP32-S3 with ESP-IDF v5.3. OpenOCD flash over JTAG times out "waiting for target halted", directly after which "Target halted" is printed:
```
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0d0020, 234 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 829 KB
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Error: timed out while waiting for target halted
Info : [esp32s3.cpu0] Target halted, PC=0x40383E40, debug_reason=00000000
Error: [esp32s3.cpu0] not halted 0, pc 0x40383e40, ps 0x60e23
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
embedded:startup.tcl:1813: Error: ** Clock configuration set failed **
```

If I first flash the board with a small basic "blinky" app (or boot in bootloader), and then flash my full project using JTAG it works fine:
```
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
** Programming Started **
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 111 KB
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038D3FE, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x4038C87E, debug_reason=00000001
Info : PROF: Erased 1220608 bytes in 2891.61 ms
Info : PROF: Data transferred in 10215.3 ms @ 116.688 KB/s
Info : [esp32s3.cpu0] Target halted, PC=0x4038D3EE, debug_reason=00000001
Info : PROF: Wrote 1220608 bytes in 10589.6 ms (data transfer time included)
** Programming Finished in 14827 ms **
```

This is using the following versions:
- ESP-IDF v5.3-369-g466a392a76
- openocd-esp32\v0.12.0-esp32-20240726 with board/esp32s3-builtin.cfg
or 
- openocd-esp32\v0.12.0-esp32-20240318 with board/esp32s3-builtin.cfg

Changing to a an even older OpenOCD `v0.12.0-esp32-20230921` build has the same error, with slightly different output and times out after 30s:
```
Info : [esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c0d0020, 234 KB
Info : Flash mapping 1: 0x50020 -> 0x42000020, 829 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038E3B6, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu1.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu1.flash' size 16384 KB
** Programming Started **
... 30s wait here ...
Error: timed out while waiting for target halted
Info : [esp32s3.cpu0] Target halted, PC=0x4038065C, debug_reason=00000000
Error: xtensa_wait_algorithm: not halted 0, pc 0x4038065c, ps 0x60e23
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
embedded:startup.tcl:1524: Error: ** Clock configuration set failed **
```

### Debug Logs

```shell
openocd_log.txt: https://drive.google.com/file/d/1PBlV5hhmDxpNd9LsKIpCT7wsn9C53u72/view?usp=sharing
```

### Expected behavior

Repeatedly flashing my project using OpenOCD JTAG should work fine with the target running

FrancoisB-HEX, 2024-09-09:
Adding the `no_clock_boost` parameter to `program_esp` solves the issue as suggested by https://github.com/espressif/openocd-esp32/issues/289#issuecomment-2337384453
But I have no way of adding that to the OpenOCD Download & Run (ESP32) plugin I use in CLion IDE

erhankur, 2024-09-14:
I think we can close this issue. Feel free to re-open if you have any other questions.

closed 2024-09-14 by erhankur