espressif/openocd-esp32 #336: ESP32_ONLYCPU doesn't work

KaeLL, 2024-09-12:
### Development Kit

ESP-WROVER-KIT V4.1

### Module or chip used

ESP32-WROVER-E

### Debug Adapter

FTDI

### OpenOCD version

v0.12.0-esp32-20240821

### Operating System

Linux

### Using an IDE ?

No

### OpenOCD command line

openocd -c 'set ESP32_ONLYCPU 1' -f board/esp32-wrover-kit-3.3v.cfg

### JTAG Clock Speed

26MHz

### ESP-IDF version

v5.4-dev-2744-g59e1838270

### Problem Description

Follow the debug example on the docs, but change the project to single-core (`CONFIG_FREERTOS_UNICORE=y`), flash it, and then start openocd.

### Debug Logs

```shell
$ openocd -c 'set ESP32_ONLYCPU 1' -f board/esp32-wrover-kit-3.3v.cfg
Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:42)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
1
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 26000 kHz
Info : JTAG tap: esp32.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32.cpu0] Examination succeed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : target esp32.cpu1 examination failed
Info : [esp32.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32.cpu0] Target halted, PC=0x40090716, debug_reason=00000000
Info : [esp32.cpu0] Reset cause (12) - (Software CPU0 reset)
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
Warn : [esp32.cpu1] Unexpected OCD_ID = 00000000
Error: [esp32.cpu1] Examination failed
```

### Expected behavior

No error

erhankur, 2024-09-13:
`ESP32_ONLYCPU` deprecated and removed. Please try `ESP_ONLYCPU`.  We will update the idf docs soon.

erhankur, 2024-09-14:
Closing the issue. Feel free to re-open if you have any other questions.

closed 2024-09-14 by erhankur