espressif/openocd-esp32 #337: Unable to change the clock frequency of the JTAG adapter.

yuzehang, 2024-09-19:
### Development Kit

Customer Board

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

ESP32C3 Build-in JTAG

### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)

### Operating System

Windows

### Using an IDE ?

VSCode

### OpenOCD command line

c:\WinSoftware\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe

### JTAG Clock Speed

40000KHz

### ESP-IDF version

V5.3.1

### Problem Description

I modified the 'adapter speed 40000' field in the '.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts\interface\esp_usb_jtag.cfg' file, changing it to 'adapter speed 24000', in an attempt to lower the JTAG Clock Speed, but it didn't actually work.

### Debug Logs

```shell
c:\WinSoftware\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capa
bilities descriptor set to 0x2000

Info : Listeni
ng on port 6666 for tcl connections
Info : Listenin
g on port 4444 for telnet connections

Info : esp_us
b_jtag: serial (EC:DA:3B:19:79:90)
Info : esp_usb_jtag: Device f
ound. Base speed 40000KHz, div range 1 to 255

Info : clock speed 40000 kHz

Info : JTAG 
tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3
] datacount=2 progbufsize=16

Info : [esp32c3] Examined RISC-V co
re; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed

Info : starting gdb server for esp32c3 on 3333
Info : Lis
tening on port 3333 for gdb connections
```

### Expected behavior

Hope to reduce the clock speed of JTAG.

### Screenshots

[image]

erhankur, 2024-09-19:
@yuzehang ESP32-C3 runs the USB-JTAG state machine from the (APB clock / 2) which is 40 MHz. OpenOCD calculates the divisor for lower speeds based on this 40 MHz clock. Therefore, if you set the adapter speed to 20000, you will observe a reduced clock. Divisor will be 2.  In other words, If you select exact divisors of 40 MHz, you will see the expected result.

yuzehang, 2024-09-20:
Oh, I see. It worked. Thank you very much!
[image]

closed 2024-09-20 by yuzehang