espressif/openocd-esp32 #338: ❌ [esp32c3] not halted (gdb fileio) Warn: Target [esp32c3] not HALTED!

yuzehang, 2024-09-20:
### Development Kit

Customer Board

### Module or chip used

ESP32-C3-MINI-1

### Debug Adapter

ESP32C3 Build-in JTAG

### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)

### Operating System

Windows

### Using an IDE ?

VSCode

### OpenOCD command line

c:\WinSoftware\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe

### JTAG Clock Speed

20000KHz

### ESP-IDF version

V5.3.1

### Problem Description

Use 'Eclipse CDT GDB Adapter' to launch debugging in VSCode. The program fails to run correctly after clicking the run button. When pausing, the program halts at 'semihosting_call_noerrno' and output an error message '[esp32c3] not halted (gdb fileio) Warn: Target [esp32c3] not HALTED!'
[image]

### Debug Logs

```shell
c:\WinSoftware\esp\.espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin\riscv32-esp-elf-gdb.exe

[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html

debug_level: 2

c:\WinSoftware\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin\openocd.exe
Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

I
nfo : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

Info : esp_usb_jta
g: serial (EC:DA:3B:19:79:90)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255

Info : clock speed 20000 kHz

Info : JTAG t
ap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3] dat
acount=2 progbufsize=16

Info : [esp32c3] Ex
amined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : starting gdb server for esp32c3 on 3333

Info : Listening on port 3333 for gdb connections

Info : accepting 'g
db' connection on tcp/3333

Memory protection is enabled. Reset target to disab
le it...

Info : JTA
G tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3] Reset cause (3) - (S
oftware core reset)

Warn : No symbols for FreeRTOS!

Info : [esp32c3] Found 8 triggers

Info : Flash mapping 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x420
00020, 90 KB

Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB

Info : Flash mapp
ing 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 90 KB
Info : Using flash bank 'esp32c3.irom' size 92 KB

Info : Flash mapping 0: 0x10020 -> 0x3c020020, 35 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 90 KB
Info : Using flash bank 'esp32c3.drom' size 36 KB

Info : Detected FreeRTOS version: (10.5.1)

Info : JTAG tap: esp32
c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3] Reset
 cause (3) - (Software core reset)

Info : Detected FreeRTOS version: (10.5.
1)

❌ Error: [esp32c3] not halted (gdb fileio)
Warn
 : Target [esp32c3] not HALTED!

Info : Halt timed
 out, wake up GDB.

❌ Error: [esp32c3] Not halted.

Info : target esp32c3 was not halted when resume was requested

❌ Error: [esp32c3] not halted (gdb fileio)
Warn : Target [esp32
c3] not HALTED!

Info : Halt timed out, wake up GDB.

[Stopped] : OpenOCD Server
For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
```

### Expected behavior

Clicking 'Run' executes the program normally, while clicking 'Pause' stops at the last executed code position.

### Screenshots

[image]

erhankur, 2024-09-20:
@yuzehang When panic occurs during debugging, application sends the reason to the OpenOCD. What do you see in the application console? Anything related to panic reason? Can you enable verbose OpenOCD logs and attach here? 

[image]

yuzehang, 2024-09-20:
@erhankur Hi, Thank you for your response. 
I open 'ESP-IDF Terminal' in VSCode and start openocd with the command 'openocd -l openocd_log.txt -d3 -f board/esp32c3-builtin.cfg'. Then I launch debugging through the debugging interface in VSCode.
[image]

When running the program, I reproduced this issue. Here is the log file of openocd.
[openocd_log.txt](https://github.com/user-attachments/files/17071943/openocd_log.txt)

Here is the content output from the debugging console.
```
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
set remotetimeout 20
0x40000000 in ?? ()
connected to target using provided connectCommands
set remote hardware-watchpoint-limit 8
mon reset halt
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[esp32c3] Reset cause (3) - (Software core reset)
maintenance flush register-cache
Register cache flushed.
thb app_main
Hardware assisted breakpoint 1 at 0x420064e6: file C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c, line 25.
GDB unhandled notify: breakpoint-created: {"bkpt":{"number":"1","type":"hw breakpoint","disp":"del","enabled":"y","addr":"0x420064e6","func":"app_main","file":"C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c","fullname":"C:\\Carrier\\Low_Cost_WiFi\\sample_project\\main\\main.c","line":"25","thread-groups":["i1"],"times":"0","original-location":"app_main"}}

In the Debug Console view you can interact directly with GDB.
To display the value of an expression, type that expression which can reference
variables that are in scope. For example type '2 + 3' or the name of a variable.
Arbitrary commands can be sent to GDB by prefixing the input with a '>',
for example type '>show version' or '>help'.

[New Thread 1070132712]
[Remote target exited]
[New Thread 1070134592]
[Switching to Thread 1070132712]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:25
25	    gpio_init();

Thread 2 "main" received signal SIGINT, Interrupt.
0x42013344 in semihosting_call_noerrno (data=0x3fc8ec64, id=259) at C:/WinSoftware/esp/Espressif/frameworks/esp-idf-v5.3.1/components/riscv/include/riscv/semihosting.h:77
77	    __asm__ __volatile__ (
[esp32c3] Not halted.
target esp32c3 was not halted when resume was requested

Thread 2 "main" received signal SIGINT, Interrupt.
0x42000c3a in semihosting_call_noerrno (id=id@entry=278, data=data@entry=0x3fc8ec78) at C:/WinSoftware/esp/Espressif/frameworks/esp-idf-v5.3.1/components/riscv/include/riscv/semihosting.h:77
77	    __asm__ __volatile__ (
gdbserver stopped
```

yuzehang, 2024-09-20:
The debugging process was started in the same way, and this time it worked properly. The program can run and pause normally. However, the probability of it working correctly is very low, and I don't know where the problem lies.

Start openocd:
[image]

Here is the log file of openocd:
[openocd_log.txt](https://github.com/user-attachments/files/17072290/openocd_log.txt)

Here is the content output from the debugging console:
```
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=riscv32-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
warning: could not convert 'main' from the host encoding (CP1252) to UTF-32.
This normally should not happen, please file a bug report.
set remotetimeout 20
0x40000000 in ?? ()
connected to target using provided connectCommands
set remote hardware-watchpoint-limit 8
mon reset halt
JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
[esp32c3] Reset cause (3) - (Software core reset)
maintenance flush register-cache
Register cache flushed.
thb app_main
Hardware assisted breakpoint 1 at 0x420064e6: file C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c, line 25.
GDB unhandled notify: breakpoint-created: {"bkpt":{"number":"1","type":"hw breakpoint","disp":"del","enabled":"y","addr":"0x420064e6","func":"app_main","file":"C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c","fullname":"C:\\Carrier\\Low_Cost_WiFi\\sample_project\\main\\main.c","line":"25","thread-groups":["i1"],"times":"0","original-location":"app_main"}}

In the Debug Console view you can interact directly with GDB.
To display the value of an expression, type that expression which can reference
variables that are in scope. For example type '2 + 3' or the name of a variable.
Arbitrary commands can be sent to GDB by prefixing the input with a '>',
for example type '>show version' or '>help'.
[New Thread 1070132712]
[Remote target exited]
[New Thread 1070134592]
[Switching to Thread 1070132712]

Thread 2 "main" hit Temporary breakpoint 1, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:25
25	    gpio_init();
Note: automatically using hardware breakpoints for read-only addresses.

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 2 "main" hit Breakpoint 2, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);

Thread 3 "IDLE" received signal SIGINT, Interrupt.
[Switching to Thread 1070134592]
0x4038424c in vPortEnterCritical () at C:/WinSoftware/esp/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:618
618	}

Thread 3 "IDLE" received signal SIGINT, Interrupt.
0x42008b60 in esp_vApplicationIdleHook () at C:/WinSoftware/esp/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/freertos_hooks.c:46
46	        if (idle_cb[core][n] != NULL && !idle_cb[core][n]()) {

Thread 3 "IDLE" received signal SIGINT, Interrupt.
task_wdt_timer_feed () at C:/WinSoftware/esp/Espressif/frameworks/esp-idf-v5.3.1/components/esp_system/task_wdt/task_wdt.c:108
108	        entry->has_reset = false;
[Switching to Thread 1070132712]

Thread 2 "main" hit Breakpoint 3, app_main () at C:/Carrier/Low_Cost_WiFi/sample_project/main/main.c:31
31	        vTaskDelay(50);
```

erhankur, 2024-09-20:
One more thing, please try using the latest OpenOCD release. We have implemented some fixes related to the flash and debug processes. Might be helpful for a better experience. https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20240821 

Regarding the error I see in the log;  The message  `timed out while waiting for target halted` usually indicates a hardware issue such as problems with the cable, JTAG pin connections or incorrect setup of bootstrap pins.

Can you try to connect from telnet instead of gdb and send `reset halt` command? It has to be succesfull each time. 

And might be also related, Why did you need to reduce the jtag clock speed on your board? Have you tried using another board?

yuzehang, 2024-09-22:
Hi, thank you for providing the solution to the problem. **It is indeed possible that the issue was caused by a cable malfunction**. After testing it on my own computer, all the debugging methods seem to work well. The previous connection method during testing may have been too unreliable: [PC] -> [Dock] -> [USB Hub] -> [ESP32].

Tomorrow, I will test it again using my company's computer and try to improve the cable connection to see how it affects the results."

The reason why I need to lower the JTAG clock speed on your board is because I saw this prompt message.
[Optimize JTAG Speed](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/jtag-debugging/tips-and-quirks.html#optimize-jtag-speed).
[image]

Once again, thank you for your helpful response.

yuzehang, 2024-09-23:
The problem has been greatly improved by directly connecting the ESP32 to the computer using a USB cable without ferrite beads. Although occasionally this issue still occurs, it no longer affects usage.

closed 2024-09-23 by yuzehang