espressif/openocd-esp32 #340: ESP32-P4: Algorithm timed out

arturo182, 2024-10-14:
### Development Kit

ESP32-P4-Function-EV-Board

### Module or chip used

ESP32-P4

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.12.0-esp32-20240821

### Operating System

Linux

### Using an IDE ?

-

### OpenOCD command line

openocd -f board/esp32p4-ftdi.cfg -c "init ; halt"

### JTAG Clock Speed

default

### ESP-IDF version

v5.3

### Problem Description

I don't know how well ESP32-P4 is supported, but I see that it does have script files and there are commits related to the chip actively being commited.

I'm using ESP-PROG connected to JTAG pins on the ESP32-P4.

Steps to reproduce:

1. I start openocd with `openocd -d -f board/esp32p4-ftdi.cfg -c "init ; halt"`
2. Then I run gdb with `riscv32-esp-elf-gdb build/hello_world.elf` and I try to connect to openocd with `tar rem :3333` (and `tar ext :3333`).
3. I get `Error: Failed to wait algorithm (-302)!` from openocd and `Remote communication error.  Target disconnected: Connection reset by peer.` from gdb

I have also tried with a manually built openocd from this repository.

### Expected behavior

I would like to connect gdb to openocd and be able to debug the chip.

erhankur, 2024-10-14:
@arturo182 Yes we still continue to fix issues regarding P4.  So I highly recommend to get the built artifact from the latest master.  https://github.com/espressif/openocd-esp32/actions/runs/11291578866 

And your issue is probably related to memory protection. To overcome this, you can add reset command to your script;
```
openocd -d -f board/esp32p4-ftdi.cfg -c "init ; reset halt"
```

When target has been reset while debugger is attached, memory protection will not be enabled.

arturo182, 2024-10-14:
I actually did check out the latest master, built it, and tried with that binary, but had the same issue. 

Your advice regarding resetting worked like a charm, thanks for the quick reply :)

Shall I close the issue?

erhankur, 2024-10-14:
Thanks for verifying. Closed

closed 2024-10-14 by erhankur