espressif/openocd-esp32 #341: OpenOCD with LLDB crashes with "Segmentation Fault"

HeiderSati, 2024-10-15:
### Development Kit

ESP32-S3-DevKitC v1.6

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

ESP-IDF

### OpenOCD version

v0.12.0-esp32-20240821

### Operating System

MacOSX Sequoia 15.0.1 (Latest as of 15-OCT-2024)

### Using an IDE ?

YES: Crashes with "clion".....   and also Command Line crashes

### OpenOCD command line

openocd -d3 -f board/esp32s3-builtin.cfg  

### JTAG Clock Speed

40000 Khz (Default)

### ESP-IDF version

v5.4-dev-3489-g6e5a178b31-dirty

### Problem Description

On the latest Mac OSX (Apple Silicon), GDB is not available as it's only compiled for Intel so far, therefore, GDB (client) cannot be used and rather using LLDB, when connecting LLDB into OPENOCD the server crashes with Segmentation Fault and LLDB disconnects. 

On SHELL, Terminal-1, I type in:
lldb
   (lldb) gdb-remote localhost:3333

On Terminal-2, I use the command:
 openocd -f board/esp32s3-builtin.cfg 

The server starts as follows:

Open On-Chip Debugger v0.12.0-esp32-20240821 (2024-08-21-14:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (F4:12:FA:52:02:1C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.tap0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.tap1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] Examination succeed
Info : [esp32s3.cpu1] Examination succeed
Info : [esp32s3.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)

>>>>>> When I start LLDB connection/session, it crashes with: 

Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 108 KB
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 8192 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 108 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 112 KB
Info : [esp32s3.cpu0] Target halted, PC=0x4038CB86, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 48 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 108 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 52 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
zsh: segmentation fault  openocd -f board/esp32s3-builtin.cfg

### Debug Logs

```shell
openocd -d3 -f board/esp32s3-builtin.cfg

Sorry could not upload here, got an error that the page exceed 65535 characters, logs are available on request if needed.
```

### Expected behavior

No Crashes, and OpenOCD should communicate with LLDB 
Also on CLION it crashes with a different error of: 

com.jetbrains.cidr.execution.debugger.backend.gdb.GDBDriver$GDBCommandException: Remote 'g' packet reply is too long (expected 452 bytes, got 512 bytes): 0004004000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000000fefff0c21f0f092325020600000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Debugger disconnected

It should work on both SHELL and CLION, but it is not at the moment.

HeiderSati, 2024-10-15:
I also tried with BASH as well, same results, the error this time is:

Segmentation fault: 11

HeiderSati, 2024-10-16:
Ok, I dag a lot deeper now and found the lines that causing the crash, I did a quick fix to resolve this issue, I am still testing not sure if you agree with my solution: 

1) Filename: src/rtos/FreeRTOS.c
2) Function: freertos_read_esp_symbol_table     
3) Line: 379 (Well; at the moment) contains the following statement:

>>>     if (!rtos->symbols[FREERTOS_VAL_ESP_OPENOCD_PARAMS].address)

4) For some reason the caller is passing NULL into ( rots->symbols ), this causes the whole like to crash due to the array being NULL

5) I solved this one for now by adding the following above this function:

// Added by Heider:
		if (!rtos->symbols)
		{
			return ERROR_FAIL;
		}

7) Doing so, it no longer crashes, which is good news. 

So, I believe I solved it, however, I will leave it to your expertise to either confirm or come back with a better fix. 

Kind Regards
Heider

erhankur, 2024-10-16:
@HeiderSati Can you send verbose OpenOCD logs as a file (-d3 -l log.txt) to see the command requests from LLDB.

HeiderSati, 2024-10-16:
Sure, I tried yesterday but the site could not take long text, I will see where can I upload the test as a file/attachment. 

[OpenOCD Problem 2024-10-15.txt](https://github.com/user-attachments/files/17394609/OpenOCD.Problem.2024-10-15.txt)

I have uploaded it now, please note that it had a couple of iterations one with ZSH and the other is with BASH,... I resolved the issue and re-compiled it now which works fine (please see my previous post), let me know if you're happy with my solution, or perhaps you like to do a different suggestion, 

Kind Regards
Heider

erhankur, 2024-10-17:
Oo sorry, I missed your description here. 

> On the latest Mac OSX (Apple Silicon), GDB is not available as it's only compiled for Intel so far, therefore, GDB (client) cannot be used and rather using LLDB, when connecting LLDB into OPENOCD the server crashes with Segmentation Fault and LLDB disconnects.

So you can use our xtensa-esp32s3-elf-gdb from the installed tools path. Let us know if there is any issue with gdb.

closed 2024-10-22 by erhankur