espressif/openocd-esp32 #342: ESP32-C3 JTAG Connection Fails with Watchdog Reset and USB Device Errors

Jimmy2027, 2024-10-17:
### Development Kit

openocd-esp32

### Module or chip used

Beetle ESP32-C3

### Debug Adapter

usb-jtag

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

linux

### Using an IDE ?

no

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.3.1

### Problem Description

I would like to get a minimal bare-metal example working with openocd on my Beetle ESP32-C3 board.
As a first step, I'm following these instructions:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/index.html
However after executing the following steps, I get errors.

1. Attach board via usb
2. install latest version of openocd-esp32 from source
3. run `openocd -f board/esp32c3-builtin.cfg`

Logs extract:
```
Debug: 284 321 riscv-013.c:1710 register_write_direct(): [esp32c3] dcsr <- 0x90c3
Debug: 285 321 riscv.c:5222 riscv_set_or_write_register(): [esp32c3] Wrote 0x90c3 to dcsr (cacheable=false, valid=false, dirty=false)
Debug: 286 321 riscv-013.c:5281 riscv013_step_or_resume_current_hart(): [esp32c3] resuming (for step?=0)
Debug: 287 417 riscv-013.c:3432 log_mem_access_result(): [esp32c3] Succeeded to read memory via system bus.
Debug: 288 417 target.c:2572 target_read_u32(): address: 0x60008038, value: 0x0000f1c7
Info : 289 417 esp32c3.c:114 esp32c3_print_reset_reason(): [esp32c3] Reset cause (7) - (Main WDT0 core reset)
Debug: 290 417 FreeRTOS.c:1381 freertos_post_reset_cleanup(): freertos_post_reset_cleanup
Error: 291 515 libusb_helper.c:357 jtag_libusb_bulk_write(): libusb_bulk_write error: LIBUSB_ERROR_PIPE
Debug: 292 515 esp_usb_jtag.c:435 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 17 bytes.
Debug: 293 515 libusb_helper.c:515 jtag_libusb_find_device(): USB dev found 303a:1001 @ 3:1-75
Error: 294 516 libusb_helper.c:104 string_descriptor_equal(): libusb_get_string_descriptor_ascii() failed with 'LIBUSB_ERROR_PIPE'
Debug: 295 1025 libusb_helper.c:515 jtag_libusb_find_device(): USB dev found 303a:1001 @ 3:1-76
Info : 296 1139 esp_usb_jtag.c:651 esp_usb_jtag_init(): esp_usb_jtag: serial (60:55:F9:B0:63:3C)
Debug: 297 1139 libusb_helper.c:440 jtag_libusb_choose_interface(): usb ep out 02
Debug: 298 1139 libusb_helper.c:440 jtag_libusb_choose_interface(): usb ep in 83
Debug: 299 1139 libusb_helper.c:449 jtag_libusb_choose_interface(): Claiming interface 2
Info : 300 1139 esp_usb_jtag.c:727 esp_usb_jtag_init(): esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Error: 301 1139 bitq.c:268 bitq_execute_queue(): missing data from bitq interface
Error: 302 1140 riscv-013.c:535 dmi_scan(): dmi_scan failed jtag scan
Debug: 303 1140 riscv-013.c:453 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1071
Error: 304 1140 riscv-013.c:626 dmi_op_timeout(): [esp32c3] Failed DMI read at 0x11; status=2
Error: 305 1140 riscv.c:3219 riscv_openocd_poll(): [esp32c3] polling failed!
Info : 306 1140 riscv-013.c:2723 riscv013_get_hart_state(): [esp32c3] Hart unexpectedly reset!

```

### Debug Logs

```shell
[openocd_log.txt](https://github.com/user-attachments/files/17413883/openocd_log.txt)
```

### Expected behavior

I'm expecting the command to execute without errors.

erhankur, 2024-10-18:
@Jimmy2027 I see that the application resets periodically and the JTAG connection fails. Does this also happen when the debugger is not attached? Have you tried starting a debugging session with GDB? 
What happens when you `reset-halt` the board during the openocd connection?

`openocd -f board/esp32c3-builtin.cfg -c 'init; reset halt'`

Jimmy2027, 2024-10-19:
right, adding `-c 'init; reset halt'` fixed the issue, thanks @erhankur !

closed 2024-10-19 by erhankur