espressif/openocd-esp32 #343: Can't debug on esp32c3

jsmith173, 2024-10-20:
I'd like to run C and assembly level debugging in Eclipse using this product
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c3/esp32-c3-devkitm-1/index.html

I played a lot but only the code upload is working (I tried in Arduino)

I have installed the latest ESP-IDF tool

OpenOCD shows the following error:

------
C:\Espressif\frameworks\esp-idf-v5.3.1>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: esp_usb_jtag: could not find or open device!
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error:
at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
Error: [esp32c3] Unsupported DTM version: -1
Error: [esp32c3] Could not identify target type.
------

I tried it in both on Linux and Windows 11

Will help when we change the eFuse and using external JTAG?

Please help.

igrr, 2024-10-20:
Hi @jsmith173,

Unfortunately this devkit has an external USB-serial converter (CP2102N) connected to the USB port, not the built-in USB_SERIAL_JTAG peripheral of the C3.

You have the following options for JTAG debugging:

- connect a USB breakout cable (such as [this one](https://www.adafruit.com/product/4448)) to GPIO18 and GPIO19. 
- move two resistors (R1 and R4 -> R2 and R3, see screenshot of the schematic below) on the PCB to route the USB lines to USB_SERIAL_JTAG
- [burn the efuse](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/jtag-debugging/configure-other-jtag.html) to make JTAG pins available directly on GPIOs, then attach a JTAG adapter such as ESP-Prog.

I would recommend the first option as you don't need to make any permanent changes to the devkit.

[image]

jsmith173, 2024-10-21:
We have also this development boards and we'd like c, asm sourve level debug in Eclipse
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32c6/esp32-c6-devkitc-1/user_guide.html
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/index.html

These boards has the same problems also?

igrr, 2024-10-21:
No, ESP32-C6 and ESP32-S3 devkits both have two USB ports — one for CP2102 (labelled "UART") and one for USB_SERIAL_JTAG (labelled "USB"). If you plug the cable into the port labelled "USB", debugging should work.

closed 2024-10-29 by jsmith173