espressif/openocd-esp32 #344: ESP32-S3-DevKitC-1: debug is not working

jsmith173, 2024-10-29:
I'm using the latest Espressif-IDE. When I try to create a new openocd debug configuration and press the Debug button the console shows the following log and the Debug is not started.

> GNU gdb (esp-gdb) 14.2_20240403
http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452, debug_reason=00000001
Warn : Failed to get flash maps (-1)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x4038E
452, debug_reason=00000001
Error: Failed to get flash size!
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Info : [esp32s3.cpu0] Target halted, PC=0x4038E452,
 debug_reason=00000001
Error: Failed to get 
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
flash size!
Error: Failed to probe flash, s
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
ize 0 KB
Error: auto_probe failed
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: attempted 'gdb' connection rejected
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
shutdown command invoked

I tried to start separately openocd and connect gdb. Openocd starts but gdb can't connect.

jsmith173, 2024-10-30:
I'm sending the log file

[log.txt](https://github.com/user-attachments/files/17566973/log.txt)

erhankur, 2024-10-30:
```
Debug: 2244 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+
Debug: 2245 67528 xtensa.c:1420 xtensa_get_gdb_reg_list(): reg_class=0, num_regs=228
Debug: 2246 67528 gdb_server.c:397 gdb_log_outgoing_packet(): [esp32s3.cpu0] {1} sending packet: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02
Error: 2247 67528 gdb_server.c:505 gdb_put_packet_inner(): GDB missing ack(2) - assumed good
Debug: 2248 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;fork-events+;vfork-events+;exec-events+;vContSupported+;QThreadEvents+;no-resumed+;memory-tagging+
Debug: 2249 67528 xtensa.c:1420 xtensa_get_gdb_reg_list(): reg_class=0, num_regs=228
Debug: 2250 67528 gdb_server.c:397 gdb_log_outgoing_packet(): [esp32s3.cpu0] {1} sending packet: $PacketSize=4000;qXfer:memory-map:read+;qXfer:features:read+;qXfer:threads:read+;QStartNoAckMode+;vContSupported+#02
Debug: 2251 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: +
Debug: 2252 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: vCont?
Debug: 2253 67528 gdb_server.c:397 gdb_log_outgoing_packet(): [esp32s3.cpu0] {1} sending packet: $vCont;c;C;s;S#62
Debug: 2254 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: +
Debug: 2255 67528 gdb_server.c:379 gdb_log_incoming_packet(): [esp32s3.cpu0] {1} received packet: vMustReplyEmpty
Debug: 2256 67528 gdb_server.c:397 gdb_log_outgoing_packet(): [esp32s3.cpu0] {1} sending packet: $#00
Warn : 2257 67528 gdb_server.c:345 gdb_write(): Error writing to GDB socket. Dropping the connection.
```

Looks like a gdb timeout issue for me.  Can you check the `gdb remotetimeout` value set from the IDE or from the console you run? According to the timestamp in the log file, 5 seconds will be enough when all debugs are enabled.

From the GDB console, before connection to the remote target (OpenOCD) you can set it like;

`set remotetimeout 5`

In the IDE, it is under GDB client setup. 

[image]

Other than this, I don't see an issue regarding reading flash mapping from the target.

jsmith173, 2024-10-30:
[log2.txt](https://github.com/user-attachments/files/17569195/log2.txt)
[image]
[image]

It looks like the problem similar to #334 
Does this even work?

jsmith173, 2024-10-30:
It works with the fix you provided.
Thanks

closed 2024-10-30 by jsmith173