espressif/openocd-esp32 #346: Cannot start a GDB session in VS code for Zephyr build

StefanHri, 2024-11-07:
### Development Kit

nanoESP32-C6 v1.0 https://github.com/wuxx/nanoESP32-C6/blob/master/README_en.md

### Module or chip used

ESP32-C6-WROOM-1

### Debug Adapter

On-board JTAG (no additional addapter)

### OpenOCD version

Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)

### Operating System

Ubuntu 24.10

### Using an IDE ?

vs code

### OpenOCD command line

/opt/openocd-esp32/bin/openocd -l openocd_log.txt -d3 -f /opt/openocd-esp32/share/openocd/scripts/board/esp32c6-builtin.cfg

### JTAG Clock Speed

clock speed 24000 kHz

### ESP-IDF version

No ESP-IDF used. Instead Zephyr OS v3.7.0

### Problem Description

1. Install Zephyr OS v 3.7.0.
2. Build the hello world sample with `west build -b esp32c6_devkitc  -- -DOPENOCD=/opt/openocd-esp32/bin/openocd -DOPENOCD_DEFAULT_PATH=/opt/openocd-esp32/share/openocd/scripts`
3. In VS Code set up a launch.json with 
```json
{
        // Use IntelliSense to learn about possible attributes.
        // Hover to view descriptions of existing attributes.
        // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
        "version": "0.2.0",
        "configurations": [
                {
                        "preLaunchTask": "openocd",
                        "name": "esp32c6",
                        "type": "cppdbg",
                        "request": "launch",
                        "cwd": "${workspaceFolder}",
                        "program": "${workspaceFolder}/build/zephyr/zephyr.elf",
                        "miDebuggerPath": "/home/stefan/Programs/esp/esp-idf/tools/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb",
                        "setupCommands": [
                                {
                                        "text": "target remote 127.0.0.1:3333"
                                },
                                {
                                        "text": "set remote hardware-watchpoint-limit 4"
                                },
                                {
                                        "text": "mon reset halt"
                                },
                                {
                                        "text": "thb main"
                                },
                                {
                                        "text": "flushregs"
                                },
                                {
                                        "text": "mon program_esp ${workspaceFolder}/build/zephyr/zephyr.bin 0x0 verify"
                                },
                                {
                                        "text": "mon reset halt"
                                },
                                {
                                        "text": "maintenance flush register-cache"
                                },
                                {
                                        "text": "thb main"
                                },
                        ],
                },
        ]
}
```
and task.json
```json
{
        "version": "2.0.0",
        "tasks": [
                { //more information: https://higaski.at/vscode-esp32-debugging/
                        "label": "openocd",
                        "type": "shell",
                        "isBackground": true,
                        "options": {
                                "cwd": "/opt/openocd-esp32"
                        },
                        "command": "bin/openocd -f share/openocd/scripts/board/esp32c6-builtin.cfg -c \"set ESP_RTOS none\"",
                        
                }
        ]
}
```
start the debug session.

Logs:
[openocd_log.txt](https://github.com/user-attachments/files/17668949/openocd_log.txt)
[gdb_log.txt](https://github.com/user-attachments/files/17668962/gdb_log.txt)

### Debug Logs

```shell
I tried to reproduce the behavior on the command line with the flowing commands:
1. start openocd: `stefan@p14s:~$ /opt/openocd-esp32/bin/openocd -l openocd_log.txt -d3 -f /opt/openocd-esp32/share/openocd/scripts/board/esp32c6-builtin.cfg
`
2. start GDB: `/home/stefan/Programs/esp/esp-idf/tools/tools/riscv32-esp-elf-gdb/14.2_20240403/riscv32-esp-elf-gdb/bin/riscv32-esp-elf-gdb -ex "set remotelogfile gdb_log.txt" -x gdbinit build/zephyr/zephyr.elf `

Logs are attached to the section "Problem Description"
```

### Expected behavior

The problem is that openOCD starts normally, the program execution starts as well but it is not stopped at break point at main.

sobuch, 2024-11-08:
Hi @StefanHri, could you please try replacing "setupCommands" with "postRemoteConnectCommands"?

erhankur, 2024-11-18:
@StefanHri  Is there any update on your side?

StefanHri, 2024-11-18:
Works for me. Thanks a lot!

closed 2024-11-18 by erhankur