espressif/openocd-esp32 #349: OpenOCD Server will not start in VS Code with ESP-IDF Extension

zathlute, 2024-12-02:
### Development Kit

ESP32-S3 Devkit C-1

### Module or chip used

ESP32-S3-N16R8

### Debug Adapter

ESP-32-S3 Chip (Via Builtin JTAG)

### OpenOCD version

v0.12.0-esp32-20240318

### Operating System

Mac

### Using an IDE ?

Visual Studio Code

### OpenOCD command line

/Users/zathlute/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin/openocd

### JTAG Clock Speed

default

### ESP-IDF version

V5.3.1

### Problem Description

1. in VS Studio Code start OpenOCD
2. Error

The board builds flashes and monitors fine, when attempting to enter debugging mode the error below is shown.

Previously the debugger was working fine until I made a project using multiple LEDS on pins G11 G9 G3 G19 and G18. Could I have burnt one of the efuses or is the issue something much more straightforward .

Apologies if this is a simple fix, I've only just started.

### Debug Logs

```shell
/Users/zathlute/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/bin/openocd

Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:24)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

❌ Error: esp_usb_jtag: could not find or open device!

❌ /Users/zathlute/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/zathlute/.espressif/tools/openocd-esp32/v0.12.0-esp32-20240318/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
```

### Expected behavior

Its expected that OpenOCD starts and the debugger is started using the builtin JTAG. 

### Screenshots

after the error VS Studio shows this prompt

[image]

sobuch, 2024-12-02:
Hi @zathlute to help us narrow this down, please try:

- newer openocd version, ideally from master branch in this repository;
- confirm efuses are in expected state, they can be read using something like `espefuse.py summary`. I suppose JTAG and USB fuses could be of interest here.

zathlute, 2024-12-10:
FIXED! Took a while... I'm not entirely sure what the main cause was although initially i think brew installed a version of openOCD in a different directory and VScode was trying to use that instead of the files in user/.espressif . To fix this I uninstalled and deleted whatever I could find that was related to the openOCD installation and used brew uninstall openocd. I then started the installation process from scratch following closely the official install documentation. The available youtube tutorials could lead you astray. 

The second issue was with the DEV board where there are two usb c connections. 1 connection is directly connected to the UART and the other is connected to the built in jtag. The aftermarket board that i purchased had these ports switched so i was using the wrong connection most times. 

And finally after rigourously following the debug setup documentation i was under the impression that you were to copy and use the examples in the getting started folder just like the setup documentation prior tells you to. I found that after doing that and trying to the use run and debug option in VS code it gave me broken default options... Turns out you need to create a new project first through the esp-idf extension in vs code and then choose the example when prompted to. This means that the launch.json settings.json and tasks.json are created properly in the .vscode folder of your project. Then you can select the correct debugger option- eclipse CDT GDB Adapter!

This was a very steep learning curve for me and i think if there was more effort put into the documentation to account for idiots like myself you would introduce many more people to embedded systems using the esp32 and not relying on arduino  :)