espressif/openocd-esp32 #350: Debugging a Zephyr OS application with openocd-esp32 on esp32c6

StefanHri, 2025-01-04:
### Development Kit

nanoESP32-C6 v1.0 https://github.com/wuxx/nanoESP32-C6/blob/master/README_en.md

### Module or chip used

ESP32-C6-WROOM-1

### Debug Adapter

On-board JTAG (no additional addapter)

### OpenOCD version

v0.12.0-esp32-20241016 (2024-10-16-14:17)

### Operating System

Ubuntu 24.04

### Using an IDE ?

vs code

### OpenOCD command line

bin/openocd -f share/openocd/scripts/board/esp32c6-builtin.cfg -c "set ESP_RTOS none" 

### JTAG Clock Speed

default

### ESP-IDF version

No ESP-IDF used. Instead Zephyr OS v4.0.99

### Problem Description

I am trying to debug a hello world Zephyr OS project on esp32c6.

1) I am building the application following the instructions from [here](https://docs.zephyrproject.org/latest/boards/espressif/esp32c6_devkitc/doc/index.html#debugging)
with the following comand:
```
rm -rf build/; west build -b esp32c6_devkitc  -- -DOPENOCD="/opt/OpenOCD-ESP/bin/openocd" -DOPENOCD_DEFAULT_PATH="/opt/OpenOCD-ESP/share/openocd/scripts"
```

2) I configured my VS code with the two files in the .vscode folder -- launch.json and tasks.json
launch.json:
```
{
        // Use IntelliSense to learn about possible attributes.
        // Hover to view descriptions of existing attributes.
        // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
        "version": "0.2.0",
        "configurations": [
                {
                        "preLaunchTask": "openocd",
                        "name": "hello_world-esp32c6",
                        "type": "cppdbg",
                        "request": "launch",
                        "cwd": "${workspaceFolder}/hello_world",
                        "program": "${workspaceFolder}/hello_world/build/zephyr/zephyr.elf",
                        "miDebuggerPath": "/home/ubuntu/zephyr-sdk-0.17.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gdb",
                        "postRemoteConnectCommands": [
                                {
                                        "text": "target remote 127.0.0.1:3333"
                                },
                                {
                                        "text": "set remote hardware-watchpoint-limit 4"
                                },
                                {
                                        "text": "mon reset halt"
                                },
                                {
                                        "text": "thb main"
                                },
                                {
                                        "text": "flushregs"
                                },
                                {
                                        "text": "mon program_esp ${workspaceFolder}/hello_world/build/zephyr/zephyr.bin 0x0 verify"
                                },
                                {
                                        "text": "mon reset halt"
                                },
                                {
                                        "text": "maintenance flush register-cache"
                                },
                                {
                                        "text": "thb main"
                                },
                        ],
                },
        ]
}
```

tasks.json:
```
{
        "version": "2.0.0",
        "tasks": [
                { //more information: https://higaski.at/vscode-esp32-debugging/
                        "label": "openocd",
                        "type": "shell",
                        "isBackground": true,
                        "options": {
                                "cwd": "/opt/OpenOCD-ESP"
                        },
                        "command": "bin/openocd -f share/openocd/scripts/board/esp32c6-builtin.cfg -c \"set ESP_RTOS none\"",
                }
        ]
}
```

After starting the debug session vscode shows an error regarding "problemMatcher", see screenshot
[image]

The debug sessions starts but it is not possible to step through the code. When I try that the program executions starts but it is not stopping on breakpoints. In addition the openocd terminal shows a lot of warnings regarding flash mappings and an error regarding FreeRTOS.
``` 
*  Executing task: bin/openocd -f share/openocd/scripts/board/esp32c6-builtin.cfg -c "set ESP_RTOS none" 

Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
none
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (54:32:04:0C:FB:E0)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] datacount=2 progbufsize=16
Info : [esp32c6] Examined RISC-V core; found 2 harts
Info : [esp32c6]  XLEN=32, misa=0x40903105
Info : [esp32c6] Examination succeed
Info : [esp32c6] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c6] Target halted, PC=0x42000002, debug_reason=00000008
Warn : No symbols for FreeRTOS!
Info : [esp32c6] Found 4 triggers
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32c6.flash' size 4096 KB
Info : Using flash bank 'esp32c6.flash' size 4096 KB
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32c6.irom' size 0 KB
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32c6.drom' size 0 KB
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Warn : RTOS FreeRTOS not detected. (GDB could not find symbol 'pxReadyTasksLists')
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
** program_esp input args <0x0 verify> **
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : PROF: Erased 147456 bytes in 619.575 ms
Info : PROF: Data transferred in 818.164 ms @ 176.004 KB/s
Info : PROF: Wrote 147456 bytes in 1151.57 ms (data transfer time included)
** Programming Finished in 1933 ms **
** Verify Started **
Info : PROF: Flash verified in 180.399 ms 
** Verify OK **
Info : JTAG tap: esp32c6.tap0 tap/device found: 0x0000dc25 (mfg: 0x612 (Espressif Systems), part: 0x000d, ver: 0x0)
Info : [esp32c6] Reset cause (24) - (JTAG CPU reset)
Error: Don't have the number of threads in FreeRTOS!
Info : [esp32c6] Target halted, PC=0x42000002, debug_reason=00000001
```
How to configure openocd-esp32 to work correctly with a zephyr OS image?

### Expected behavior

I can debug the application by stepping through the code.

erhankur, 2025-01-04:
Zephyr doesn't use the IDF bootloader, so OpenOCD cannot retrieve the flash mappings by reading the target's flash. Therefore, you must disable flash support to avoid the mappings error. Additionally, you need to specify the RTOS type before the configuration file; otherwise, OpenOCD will assume your application runs on top of FreeRTOS. 

```
bin/openocd -s share/openocd/scripts/ -c 'set ESP_RTOS none; set ESP_FLASH_SIZE 0` -f board/esp32c6-builtin.cfg
```

After disabling flash support, you cannot flash your application using OpenOCD. Instead, you can flash it using `west flash --esp-device <port>`

StefanHri, 2025-01-04:
Thanks! It looks like with Zephyr we are not using some (or may me all) additional features of the Espressif specific OpenOSD-esp32. Instead, is it possible to use the vanilla OpenOCD or JLINK GDB server? Will it be possible to flash and debug at ones with some of them?

StefanHri, 2025-01-04:
I try it out and doesn't work. First I am still getting: 
[image]
and after clicking on debug anyway I am getting:
[image]

erhankur, 2025-01-04:
As I said, you can not run `program_esp` command since you have to disable flash support.  
Please try to get it run from the command line first. I am not familiar enough with VS Code settings.

- First, flash your application with `west flash` or `esptool` directly.
- Run OpenOCD in one terminal 
- Run gdb in another terminal

As an alternative, looks like Jlink has ESP32-C6 support. https://www.segger.com/supported-devices/espressif

erhankur, 2025-01-12:
@StefanHri any update on your side?

erhankur, 2025-01-16:
Closing due to inactivity. Feel free to re-open if you have something new.

closed 2025-01-16 by erhankur