espressif/openocd-esp32 #351: OPEN OCD TIMEOUT ERROR

Manasabalakrishna, 2025-01-07:
### Development Kit

ESP32S2

### Module or chip used

CUSTOM BOARD

### Debug Adapter

custom board

### OpenOCD version

C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin/openocd.exe

### Operating System

windows

### Using an IDE ?

espressif ide

### OpenOCD command line

im just ruuning by selecting debug configuartoion and there gdb ocd configuration

### JTAG Clock Speed

default

### ESP-IDF version

v5.3.1

### Problem Description

[image]
[image]
  
  
  Every time when i select debug configuration and gave debug, its showing as open ocd error and the above output , & sometime automatically it will work and again later it will show the same error .how to resolve this

### Debug Logs

```shell
NU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
Type "apropos word" to search for commands related to "word".
Info : accepting 'gdb' connection on tcp/3333
Error: Target not examined yet
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error executing event gdb-attach on target esp32s2:
embedded:startup.tcl:1561: Error: 
in procedure 'program_esp' called at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 562
in procedure 'program_error' called at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 395
at file "embedded:startup.tcl", line 1561
Warn : No symbols for FreeRTOS!
Error: Target not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: auto_probe failed
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: attempted 'gdb' connection rejected
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
shutdown command invoked
```

### Expected behavior

Im using esp32s2 custom board with debugger for esp32s2 and when i press debug it should start debugging but now im not able to debug

### Screenshots

[image]
[image]

kolipakakondal, 2025-01-08:
> Error: Target not examined yet

- Verify the connection - This issue can occur due to incorrect connections or wiring.
- Verify the ESP32-S2 configuration file in the debug configuration window - Ensure you select the appropriate ESP32-S2 configuration file based on the board you've.

[image]

Manasabalakrishna, 2025-01-08:
GNU gdb (esp-gdb) 14.2_20240403
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=xtensa-esp-elf".
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Type "apropos word" to search for commands related to "word".
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : [esp32s2] Target halted, PC=0x40029EA6, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 197 KB
Info : Flash mapping 1: 0x50020 -> 0x40080020, 678 KB
Info : Using flash bank 'esp32s2.irom' size 680 KB
Info : [esp32s2] Target halted, PC=0x40029EA6, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3f000020, 197 KB
Info : Flash mapping 1: 0x50020 -> 0x40080020, 678 KB
Info : Using flash bank 'esp32s2.drom' size 200 KB
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40012B13, debug_reason=00000000
Info : [esp32s2] Reset cause (3) - (Software core reset)
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s2] Reset cause (3)
 - (Software core reset)
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 
(mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target h
alted, PC=0x4001275C, debug_reason=00000000
Info : [esp32s2] Reset cause (3) - (Software core reset)
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s2] Reset cause (3) - (Software core reset)
Info : JTAG tap: esp32s2.cpu tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s2] requesting target halt and executing a soft reset
Info : [esp32s2] Debug controller was reset.
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x4000F974, debug_reason=00000000
Info : [esp32s2] Reset cause (3) - (Software core reset)
Info : [esp32s2] Core was reset.
Info : [esp32s2] Target halted, PC=0x4000
0400, debug_reason=00000000
Info : [esp32s2] Reset cause (3) - (Software core reset)
Info : The target is not running when halt was requested, stopping GDB.
Error: [esp32s2] not halted (gdb fileio)
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32s2] not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32s2] not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ

Program
 received signal SIGINT, Interrupt.
0x40000400 in ?? ()
Error: [esp32s2] not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32s2] not halted
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32s2] Failed to prepare for resume!
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Info : target esp32s2 was not halted when resume was requested

[image]
[image]
[image]

erhankur, 2025-01-16:
@Manasabalakrishna Could you try debugging the hello-world example first to verify your setup is completely correct? And If you stil have an issue please share verbose OpenOCD logs. 

Check here to enable it. https://docs.espressif.com/projects/espressif-ide/en/latest/openocddebugging.html#startup-tab

erhankur, 2025-02-01:
@Manasabalakrishna any update on this?

Manasabalakrishna, 2025-02-03:
@Erhan Kurubas , I dont know the exact reason but if I clean project
completely and build it the "open OCD error " will not be there.
1.The issue could be with hardware that is due to loose wire connections.
2.Clean the project before building it.

closed 2025-02-03 by erhankur