espressif/openocd-esp32 #353: Unable to debug ESP32-C3-DevKitC-02v1.1 on M1 MacOS using VS Code + ESP-IDF (But Flashing Code Works)

Shu244, 2025-01-12:
### Development Kit

ESP32-C3-DevKitC-02v1.1

### Module or chip used

ESP32-C3-WROOM-02

### Debug Adapter

Built-in JTAG

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

M1 Mac, MacOS 14.6.1 

### Using an IDE ?

VS Code with ESP-IDF extension

### OpenOCD command line

Using VS Code built in debug button (settings below)

### JTAG Clock Speed

N/A

### ESP-IDF version

v5.4.0

### Problem Description

I am unable to debug my ESP32 dev kit using the settings 

```
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.espIdfPath": "/Users/shu/esp/esp-idf",
    "idf.openOcdConfigs": [
        "board/esp32c3-builtin.cfg"
    ],
    "idf.port": "/dev/tty.usbserial-110",
    "idf.toolsPath": "/Users/shu/.espressif",
    "files.associations": {
        "led_strip.h": "c"
    },
    "idf.flashType": "UART"
}
```

I get the error: 

```
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

❌ Error: esp_usb_jtag: could not find or open device!

❌ /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
at file "/Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9
Error: [esp32c3] Unsupported DTM version: -1
Error: [esp32c3] Could not identify target type.

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
```

I am able to flash the MCU, but I am unable to debug. I have been trying to get the debugging to work for many days and read a lot of the Espressif documentation, but none seem to help. Any help is greatly appreciated. Thank you! 

### Expected behavior

Enter debug mode.

erhankur, 2025-01-12:
@Shu244 can you explain your jtag connection? 

https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-esp32-c3-built-in-jtag-interface

You need a breakout cable like this: https://www.adafruit.com/product/4448

Shu244, 2025-01-12:
Hi @erhankur thank you for taking a look. I was reading [this](https://docs.espressif.com/projects/esp-idf/en/stable/esp32c3/api-guides/jtag-debugging/index.html#jtag-debugging-selecting-jtag-adapter) doc

```
The quickest and most convenient way to start with JTAG debugging is through a USB cable connected to the D+/D- USB pins of ESP32-C3. No need for an external JTAG adapter and extra wiring/cable to connect JTAG to ESP32-C3.
```

I assumed that this means I can use the existing onboard micro USB on the dev board for debugging. Is this not the case? I wish the document was more clear about this.

erhankur, 2025-01-16:
@Shu244 do you still need some help from our side?

Shu244, 2025-01-19:
Sorry for the delay @erhankur . I am still having a lot of trouble trying to get the debugger to work.

The following is the output when I clear the flash and try to start the debugger:

```
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : Listening on port 6666 for tcl connections

Info : Listening on port 4444 for telnet connections

Info : esp_usb_jtag: serial (24:58:7C:AC:B8:8C)

Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255

Info : clock speed 40000 kHz

Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)

Info : [esp32c3] datacount=2 progbufsize=16

Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104

Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)

Info : [esp32c3] Hart unexpectedly reset!

Info : [esp32c3] Reset cause (7) - (Main WDT0 core reset)
```

Thank you for helping, I have been stuck here and been unable to make progress.

erhankur, 2025-01-20:
Before debugging, you first need to load the application. How do you do that?
I suggest starting with the hello-world example from the command line within the ESP-IDF environment. Then, once you’re sure your hw and sw setup is ready, switch to the IDE and try debugging there. After that, it’s mainly a matter of playing with the IDE settings.

From one terminal
```
cd hello_world
idf.py set-target esp32c3
idf.py flash
idf.py openocd
```

From the other terminal
```
idf.py gdb
# idf.py gdbgui or idf.py gdbtui depends on your choice
```

Shu244, 2025-01-21:
Hi @erhankur I am using VS Code with ESP IDF extension. I am starting with the Blinky example and was able to flash the MCU and see the LED blinking. I feel confident the hardware and software is now set up properly. To start debugging, I am simply using the debug button in the IDE. 

[image]

erhankur, 2025-01-21:
Can you share your settings.json? 

In the meantime you can add `"idf.openOcdDebugLevel": 3,` and send us OpenOCD logs. You should see the logs at the output tab.

Shu244, 2025-01-22:
@erhankur  here is my `settings.json`: 

```
{
    "C_Cpp.intelliSenseEngine": "default",
    "idf.espIdfPath": "/Users/shuhao/esp/esp-idf",
    "idf.openOcdConfigs": [
        "board/esp32c3-builtin.cfg"
    ],
    "idf.port": "/dev/tty.usbmodem11201",
    "idf.toolsPath": "/Users/shuhao/.espressif",
    "files.associations": {
        "led_strip.h": "c"
    },
    "idf.flashType": "UART"
}

```

Here are the logs:

```
pen On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 3 1 options.c:52 configuration_output_handler(): debug_level: 3User : 4 1 options.c:52 configuration_output_handler(): 
Debug: 5 1 options.c:346 parse_cmdline_args(): ARGV[0] = "openocd"
Debug: 6 1 options.c:346 parse_cmdline_args(): ARGV[1] = "-d3"
Debug: 7 1 options.c:346 parse_cmdline_args(): ARGV[2] = "-f"
Debug: 8 1 options.c:346 parse_cmdline_args(): ARGV[3] = "board/esp32c3-builtin.cfg"
Debug: 9 1 options.c:233 add_default_dirs(): bindir=/builds/idf/openocd-esp32/_build/../openocd-esp32/bin
Debug: 10 1 options.c:234 add_default_dirs(): pkgdatadir=/builds/idf/openocd-esp32/_build/../openocd-esp32/share/openocd
Debug: 11 1 options.c:235 add_default_dirs(): exepath=/Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin
Debug: 12 1 options.c:236 add_default_dirs(): bin2data=../share/openocd
Debug: 13 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts
Debug: 14 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/Library/Preferences/org.openocd
Debug: 15 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/.config/openocd
Debug: 16 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/.openocd
Debug: 17 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin/../share/openocd/site

Debug: 18 1 configuration.c:33 add_script_search_dir(): adding /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin/../share/openocd/scripts
Debug: 19 1 command.c:152 script_debug(): command - ocd_find board/esp32c3-builtin.cfg
Debug: 20 1 configuration.c:88 find_file(): found /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/board/esp32c3-builtin.cfg

Debug: 21 1 command.c:152 script_debug(): command - ocd_find interface/esp_usb_jtag.cfg

Debug: 22 1 configuration.c:88 find_file(): found /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/interface/esp_usb_jtag.cfg

Debug: 23 2 command.c:152 script_debug(): command - adapter driver esp_usb_jtag
Info : 24 2 transport.c:106 allow_transports(): only one transport option; autoselecting 'jtag'

Debug: 25 2 command.c:152 script_debug(): command - espusbjtag vid_pid 0x303a 0x1001
Info : 26 2 esp_usb_jtag.c:891 esp_usb_jtag_vid_pid(): esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Debug: 27 2 command.c:152 script_debug(): command - espusbjtag caps_descriptor 0x2000
Info : 28 2 esp_usb_jtag.c:902 esp_usb_jtag_caps_descriptor(): esp_usb_jtag: capabilities descriptor set to 0x2000
Debug: 29 2 command.c:152 script_debug(): command - adapter speed 40000
Debug: 30 2 adapter.c:262 adapter_config_khz(): handle adapter khz
Debug: 31 2 adapter.c:226 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 32 2 adapter.c:226 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 33 2 command.c:152 script_debug(): command - ocd_find target/esp32c3.cfg
Debug: 34 2 configuration.c:88 find_file(): found /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp32c3.cfg

Debug: 35 2 command.c:152 script_debug(): command - ocd_find target/esp_common.cfg
Debug: 36 2 configuration.c:88 find_file(): found /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/target/esp_common.cfg

Debug: 37 3 command.c:152 script_debug(): command - ocd_find bitsbytes.tcl
Debug: 38 3 configuration.c:88 find_file(): found /Users/shuhao/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/share/openocd/scripts/bitsbytes.tcl

Debug: 39 3 command.c:152 script_debug(): command - ocd_find memory.tcl
[... 310 lines cut ...]
	
Debug: 244 51 command.c:152 script_debug(): command - arm semihosting enable

Debug: 245 51 riscv_semihosting.c:169 riscv_semihosting_setup(): [esp32c3] enable=1
Debug: 246 51 command.c:152 script_debug(): command - arm semihosting_resexit enable

Debug: 247 51 command.c:152 script_debug(): command - arm semihosting_basedir .
Info : 248 51 target.c:699 target_examine_one(): [esp32c3] Examination succeed
Debug: 249 51 command.c:152 script_debug(): command - flash init
Debug: 250 51 tcl.c:1365 handle_flash_init_command(): Initializing flash devices...

Debug: 251 51 command.c:152 script_debug(): command - nand init
Debug: 252 51 tcl.c:484 handle_nand_init_command(): Initializing NAND devices...
Debug: 253 51 command.c:152 script_debug(): command - pld init
Debug: 254 51 pld.c:337 handle_pld_init_command(): Initializing PLDs...
Debug: 255 51 command.c:152 script_debug(): command - tpiu init
Info : 256 51 gdb_server.c:3854 gdb_target_start(): [esp32c3] starting gdb server on 3333

Info : 257 51 server.c:299 add_service(): Listening on port 3333 for gdb connections
Debug: 258 51 command.c:152 script_debug(): command - target names

Debug: 259 52 command.c:152 script_debug(): command - target names
Debug: 260 52 command.c:152 script_debug(): command - esp32c3 cget -type

```

erhankur, 2025-01-22:
@Shu244 Thanks for the log but I don't see any error. What happens when you start debugging? What is the error you get? 
Did you add a breakpoint to the app_main and try to step from there? Please share more details so that we can help you.

Shu244, 2025-01-22:
@erhankur yes I have a break point set, the debugger simply never starts: 

[image]

erhankur, 2025-01-23:
And any logs in the termninal/ debug console?

@brianignacio5 could you please help?

erhankur, 2025-01-31:
@Shu244 I am closing this since there are no issues related to OpenOCD for now.

closed 2025-01-31 by erhankur

erhankur, 2025-01-31:
And regarding the gdb issue; please try to run `riscv32-esp-elf-gdb`  and `riscv32-esp-elf-gdb-no-python`  from the command line and see if there is an error for both.