espressif/openocd-esp32 #355: esp_usb_jtag: could not find or open device

Vineetha1705, 2025-01-16:
### Development Kit

Esp32P4-Function-EV-Board

### Module or chip used

Esp32P4

### Debug Adapter

Builtin USB-JTAG

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

Windows

### Using an IDE ?

Espressif ide

### OpenOCD command line

-s ${openocd_path}/share/openocd/scripts -f board/esp32p4-builtin.cfg

### JTAG Clock Speed

Default

### ESP-IDF version

 v5.3.2

### Problem Description

I am connecting the one USB cable to the USB-to-UART port for flashing, and the code is successfully flashed and also another usb cable connecting to the USB Power-in Port using adapter. However, when I start debugging, it does not enter debug mode and instead shows errors.

### Debug Logs

```shell
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'gdb port', not 'gdb_port'
DEPRECATED! use 'tcl port' not 'tcl_port'
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Flashing "D:/Vineetha/Risc-V/Esp32p4test/Test/build/bootloader/bootloader.bin" at 0x2000
** program_esp input args <0x2000 verify> **
Error: esp_usb_jtag: could not find or open device!
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
** OpenOCD init failed **
shutdown command invoked
Error: [esp32p4.hp.cpu0] dmi_write is not implemented for this target.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error: 
in procedure 'program_esp_bins' 
in procedure 'program_esp' 
in procedure 'program_error' called at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 443
at file "C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: [esp32p4.hp.cpu0] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu0] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
```

### Expected behavior

Should enter into debug mode

### Screenshots

[image]

erhankur, 2025-01-16:
@Vineetha1705 What is your board version? If you have v1.4, try to connect a breakout cable to the pins GPIOs 24 and 25. They are D+ and D- of the USB/JTAG interface. 

Cable: https://www.adafruit.com/product/4448

Vineetha1705, 2025-01-16:
What is your board version? If you have v1.4-Yes.

Actually, until yesterday, the code was successfully flashing into the ESP32-P4. However, today, when I started flashing, it failed and displayed errors like the ones mentioned below:

********************************************************************************************************************************************************************
Executing action: flash
Running ninja in directory C:\Users\Vineetha\Desktop\Test1\hello_world\build
Executing "ninja flash"...
[1/5] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\Vineetha\Desktop\Test1\hello_world\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe D:/Vineetha/esp-idf-v5.3.2/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/Vineetha/Desktop/Test1/hello_world/build/partition_table/partition-table.bin C:/Users/Vineetha/Desktop/Test1/hello_world/build/hello_world.bin"

hello_world.bin binary size 0x30890 bytes. Smallest app partition is 0x100000 bytes. 0xcf770 bytes (81%) free.

[2/5] Performing build step for 'bootloader'

[1/1] C:\WINDOWS\system32\cmd.exe /C "cd /D C:\Users\Vineetha\Desktop\Test1\hello_world\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe D:/Vineetha/esp-idf-v5.3.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x2000 C:/Users/Vineetha/Desktop/Test1/hello_world/build/bootloader/bootloader.bin"

Bootloader binary size 0x58e0 bytes. 0x720 bytes (7%) free.

[3/5] No install step for 'bootloader'

[4/5] Completed 'bootloader'

[4/5] C:\WINDOWS\system32\cmd.exe /C "cd /D D:\Vineetha\esp-idf-v5.3.2\components\esptool_py && C:\Espressif\tools\cmake\3.30.2\bin\cmake.exe -D IDF_PATH=D:/Vineetha/esp-idf-v5.3.2 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;D:/Vineetha/esp-idf-v5.3.2/components/esptool_py/esptool/[esptool.py](https://esptool.py/);--chip;esp32p4 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/Vineetha/Desktop/Test1/hello_world/build -P D:/Vineetha/esp-idf-v5.3.2/components/esptool_py/run_serial_tool.cmake"

[esptool.py](https://esptool.py/) --chip esp32p4 -p COM25 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x2000 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin

[esptool.py](https://esptool.py/) v4.8.1

Serial port COM25

A fatal error occurred: Could not open COM25, the port is busy or doesn't exist.

(could not open port 'COM25': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2))

Hint: Check if the port is correct and ESP connected

CMake Error at run_serial_tool.cmake:66 (message):
C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;D:/Vineetha/esp-idf-v5.3.2/components/esptool_py/esptool/[esptool.py](https://esptool.py/);--chip;esp32p4
failed.

FAILED: CMakeFiles/flash C:/Users/Vineetha/Desktop/Test1/hello_world/build/CMakeFiles/flash

C:\WINDOWS\system32\cmd.exe /C "cd /D D:\Vineetha\esp-idf-v5.3.2\components\esptool_py && C:\Espressif\tools\cmake\3.30.2\bin\cmake.exe -D IDF_PATH=D:/Vineetha/esp-idf-v5.3.2 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.3_py3.11_env/Scripts/python.exe;;D:/Vineetha/esp-idf-v5.3.2/components/esptool_py/esptool/[esptool.py](https://esptool.py/);--chip;esp32p4 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/Vineetha/Desktop/Test1/hello_world/build -P D:/Vineetha/esp-idf-v5.3.2/components/esptool_py/run_serial_tool.cmake"

ninja: build stopped: subcommand failed.

ninja failed with exit code 1, output of the command is in the C:\Users\Vineetha\Desktop\Test1\hello_world\build\log\idf_py_stderr_output_13964 and C:\Users\Vineetha\Desktop\Test1\hello_world\build\log\idf_py_stdout_output_13964
**************************************************************************************************************************************************************************

I don’t understand why this problem is occurring. Please help me resolve it.
In the meantime, I tried restarting the system and updating the drivers using the Zadig tool.

But it is working in another system.

Regards,
D. Vineetha.

erhankur, 2025-01-16:
If you want to flash application from UART, you can use USB-UART connector (j20). It should't require Zadig things..  You may need to install CP210x USB to UART Bridge driver. 

If you want to debug application from JTAG through GPIO24-25, then use IDF installer for  the necessary drivers.  https://docs.espressif.com/projects/esp-idf/en/latest/esp32p4/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-usb-drivers

And check here to verify https://github.com/espressif/openocd-esp32/issues/230#issuecomment-1134393392

Vineetha1705, 2025-01-16:
Yes, I am using the USB-UART connect(j20) for flashing but it is not working properly now. It is showing the above errors.

Vineetha1705, 2025-01-16:
I already tried all these suggestions but still same problem is getting.

erhankur, 2025-01-16:
I am sorry, I don't have any other suggestions. This is not related to OpenOCD. Maybe you might try your luck in the esptool repo. Someone might have encountered the same issue before.

Vineetha1705, 2025-01-16:
Okay, Thank you so much.

Vineetha1705, 2025-01-17:
Hello ,

Now the problem got solved by upgrading the driver.
I am facing one more issue with Esp32p4 debugging. When I start debugging it is not entering into debug mode instead of it is showing errors which are mentioned below:

*******************************************************************************************************************************************************************
Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
WARNING!!! OpenOCD binary version (Open On-Chip Debugger v0.12.0-esp32-20240318 (2024-03-18-18:26)) doesn't match the .cfg file version (v0.12.0-esp32-20241016)
Flashing "C:/Users/Vineetha/Desktop/Test/hello_world/build/bootloader/bootloader.bin" at 0x2000
** program_esp input args <0x2000 verify> **
Error: esp_usb_jtag: could not find or open device!
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
** OpenOCD init failed **
shutdown command invoked
esp
esp appimage_offset offset
esp apptrace (start <destination> [poll_period [trace_size [stop_tmo
[wait4halt [skip_size]]]]) | (stop) | (status) | (dump
<destination>)
esp compression ['on'|'off']
esp encrypt_binary ['yes'|'no']
esp flash_stub_clock_boost ['on'|'off']
esp gcov [dump] [<prefix> [<prefix_strip>]]
esp gdb_detach_handler
esp halted_event_handler
esp stub_log ['on'|'off']
esp sysview (start file://<outfile1> [file://<outfile2>] [poll_period
[trace_size [stop_tmo [wait4halt [skip_size]]]]) | (stop) |
(status)
esp sysview_mcore (start file://<outfile> [poll_period [trace_size
[stop_tmo [wait4halt [skip_size]]]]) | (stop) | (status)
esp verify_bank_hash bank_id filename [offset]
esp32p4.hp.cpu0
esp32p4.hp.cpu0 arm
esp32p4.hp.cpu0 arm semihosting ['enable'|'disable']
esp32p4.hp.cpu0 arm semihosting_basedir [dir]
esp32p4.hp.cpu0 arm semihosting_cmdline arguments
esp32p4.hp.cpu0 arm semihosting_fileio ['enable'|'disable']
esp32p4.hp.cpu0 arm semihosting_read_user_param
esp32p4.hp.cpu0 arm semihosting_redirect (disable | tcp <port>
['debug'|'stdio'|'all'])
esp32p4.hp.cpu0 arm semihosting_resexit ['enable'|'disable']
esp32p4.hp.cpu0 arp_examine ['allow-defer']
esp32p4.hp.cpu0 arp_halt
esp32p4.hp.cpu0 arp_halt_gdb
esp32p4.hp.cpu0 arp_poll
esp32p4.hp.cpu0 arp_reset 'assert'|'deassert' halt
esp32p4.hp.cpu0 arp_waitstate statename timeoutmsecs
esp32p4.hp.cpu0 cget target_attribute
esp32p4.hp.cpu0 configure [target_attribute ...]
esp32p4.hp.cpu0 curstate
esp32p4.hp.cpu0 debug_reason
esp32p4.hp.cpu0 esp
esp32p4.hp.cpu0 esp apptrace (start <destination> [poll_period
[trace_size [stop_tmo [wait4halt [skip_size]]]]) | (stop) |
(status) | (dump <destination>)
esp32p4.hp.cpu0 esp gcov [dump] [<prefix> [<prefix_strip>]]
esp32p4.hp.cpu0 esp gdb_detach_handler
esp32p4.hp.cpu0 esp halted_event_handler
esp32p4.hp.cpu0 esp sysview (start file://<outfile1>
[file://<outfile2>] [poll_period [trace_size [stop_tmo
[wait4halt [skip_size]]]]) | (stop) | (status)
esp32p4.hp.cpu0 esp sysview_mcore (start file://<outfile> [poll_period
[... 167 lines cut ...]
** Flashing Failed **
-1
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: [esp32p4.hp.cpu0] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu0] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Unsupported DTM version: -1
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
Error: [esp32p4.hp.cpu1] Could not identify target type.
Please refer to the troubleshooting guide below to identify the problem.
https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ

*****************************************************************************************************************************************
Please help me to solve this problem.

Regards,
D. Vineetha.

Closed #355<https://github.com/espressif/openocd-esp32/issues/355> as completed.

erhankur, 2025-01-17:
> Error: esp_usb_jtag: could not find or open device!

Did you check the driver link?  https://github.com/espressif/openocd-esp32/issues/355#issuecomment-2595187799
And do you have a cable like I explained here?  https://github.com/espressif/openocd-esp32/issues/355#issuecomment-2594933494

Vineetha1705, 2025-01-22:
_do you have a cable like I explained here? https://github.com/espressif/openocd-esp32/issues/355#issuecomment-2594933494_-Yes, Now it is entering into debug mode.

Thanks & Regards.,
D. Vineetha.

closed 2025-01-28 by erhankur