espressif/openocd-esp32 #356: ESP32: profiling samples are inappropriately

bryghtlabs-richard, 2025-01-24:
### Development Kit

ESP32-S3-WROOM-1

### Module or chip used

ESP32-S3-WROOM-1

### Debug Adapter

Internal USB JTAG

### OpenOCD version

v0.12.0-esp32-20240821

### Operating System

Windows 11

### Using an IDE ?

No

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

40MHz

### ESP-IDF version

v5.1.5

### Problem Description

1. In OpenOCD, run `profile 10 gmon.out`
2. While profiling, make sure to run a mix ROM, IRAM, and QSPI Flash instructions
3. Run `gprof gmon.out`
4. GProf's program-counter samples are all messed up because the histogram doesn't have enough bins to handle the spare address space used by ESP32 chips.

### Debug Logs

```shell

```

### Expected behavior

I would like instruction-accurate profiling, so that I may know which functions and which instructions we're spending time on during a long random-sampling run. The following series of patches addresses this with upstream OpenOCD:

- https://review.openocd.org/c/openocd/+/8603/
- https://review.openocd.org/c/openocd/+/8604/
- https://review.openocd.org/c/openocd/+/8605/
- https://review.openocd.org/c/openocd/+/8277/

bryghtlabs-richard, 2025-01-24:
Other requests for this here: https://www.esp32.com/viewtopic.php?f=13&t=39619

erhankur, 2025-01-31:
@bryghtlabs-richard Sorry, just to understand. Are your upstream patches fixing the problem you described, or are you waiting for a solution on the target side ?

We have an internal task to keep all histograms on the target and sends them (via apptrace JTAG interface) to the host when user invokes special OpenOCD command. We did POC but it is not optimal in terms of RAM usage.  So we will revisit it for optimization.

bryghtlabs-richard, 2025-01-31:
@erhankur, my upstream patches fix the problem, and I was just hoping that as an impacted CPU vendor, Espressif might be interested in reviewing.

erhankur, 2025-02-03:
@bryghtlabs-richard thanks for the info. We will help with merging your patches.

closed 2025-02-03 by erhankur