espressif/openocd-esp32 #357: Fast JTAG profile sampler for ESP32/ESP32-S2

bryghtlabs-richard, 2025-01-31:
I implemented a non-intrusive program-counter sampler for ESP32-S3 [in this patch](https://review.openocd.org/c/openocd/+/8431), and it was a bit over 1000x faster than the current approach. [In this thread](https://esp32.com/viewtopic.php?f=13&t=39619), ESP-Sprite mentioned that this approach should work for the other xtensa chips.

Please consider this a low priority request if anyone is interested to switch ESP32 and ESP32-S2 to the new sampler. For `esp32_target` and `esp32s2_target`, each just needs `.profiling = esp_xtensa_profiling,` assigned to switch over to the esp-xtensa non-intrusive sampler already in use by ESP32-S3, then a profile collected and analyzed.

erhankur, 2025-02-03:
@bryghtlabs-richard, thank you for your contributions. We’ll certainly consider adding the new sampler to the ESP32 and ESP32-S2 targets.

rsaxvc, 2025-05-17:
Verified original ESP32 works with non-intrusive sampler: https://review.openocd.org/c/openocd/+/8910

sobuch, 2025-07-31:
Hi @rsaxvc thanks again for your efforts. We also checked internally, and the new sampler works with esp32s2 as well, so we added it for both targets. Please feel free to add for esp32s2 in your change request to upstream.

closed 2025-07-31 by sobuch