espressif/openocd-esp32 #358: OpenOCD Error: attempted 'gdb' connection rejected when using ESP-IDF command-line and ESP32-C3

mnigbor, 2025-02-17:
### Development Kit

Seeed Studio XIAO ESP32C3

### Module or chip used

ESP32-C3

### Debug Adapter

On-Chip JTAG via USB

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

Windows 11

### Using an IDE ?

ESP-IDF command-line

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

40000 kHz

### ESP-IDF version

v5.4

### Problem Description

I am unable to establish a connection between riscv32-esp-elf-gdb and a Xiao ESP32-C3.  The OpenOCD server starts normally. During startup, it is able to see the ESP32-C3 and retrieve information about it.  I am using the hello_world example application from the ESP-IDF. 

```
C:\Espressif\frameworks\esp-idf-v5.4>openocd -f board/esp32c3-builtin.cfg
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (54:32:04:87:F8:34)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections

```
When I attempt to connect the debugger, the connection fails.  Here is what the OpenOCD server output says:

```
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Target halted, PC=0x40000000, debug_reason=00000008
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```
[This seems very similar to an issue reported for the ESP32-S3](https://github.com/espressif/openocd-esp32/issues/315#issuecomment-1956404126).  The two main things that fixed this issue were: 

1. Changing the flash memory size to 4MB, 
2. Setting some environment variables that I didn't understand. 

On the assumption that the C3 and S3 are similar, I changed the flash memory size from 2M to 4M and got my first successful connection.  The next day, after installing the VS Code Extension for ESP-IDF, it stopped working (with the same errors). I gradually retraced my steps, returning to 2M, then back to 4M, and it started working again. I was even able to single step and set breakpoints in VS Code.  Then, it stopped working again and I haven't been able to get it working since.  

I am not experiencing the timeout issues reported in this issue.

Yesterday, I was pretty sure the C3 had the same issue as the S3.  Today, I don't know what to think.

Here is the OpenOCD log:
[openocd_log.txt](https://github.com/user-attachments/files/18833110/openocd_log.txt)

Here is the gdb long: 
[gdb_log.txt](https://github.com/user-attachments/files/18833124/gdb_log.txt)

### Debug Logs

```shell

```

### Expected behavior

I expect gdb to connect and that I'm able to set breakpoints, view variables.

erhankur, 2025-02-18:
@mnigbor Thanks for reporting the issue. Some questions,

1. When you say S3, do you mean C6? Because the link provided was for the C6 issue.
2. I don't have that board, so I can't test it now (I will order one). Could you double check if the bootstrap pins are all in the correct state during startup? Some users had same issue due to incorrect pin states. They fixed by setting pin D8 state to high during boot. 
3. Which application did you upload for debugging? Did you try with the hello_world example first? What is the result?
4. Could you also enable the stub logs and share the log file? We can understand why it fails while reading the flash size.
```
openocd -f board/esp32c3-builtin.cfg -c "esp stub_log on" -d3 -l log.txt
``

mnigbor, 2025-02-18:
@erhankur, Thanks for the quick response. I meant C6. The text for the link was incorrect. I edited the original post to correct the error. (I managed to confuse this issue with another post on esp32.com)

I'll need a little time to research the bootstrap pins and run with stub logs enabled.