espressif/openocd-esp32 #359: Compiling on macOS Sequoia 15.3.1 will fail

zetaerre, 2025-03-09:
Compiling on macOs will fail with following error:

-----
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in jimtcl
make[2]: Nothing to be done for `all'.
restore=: && backupdir=".am$$" && \
	am__cwd=`pwd` && CDPATH="${ZSH_VERSION+.}:" && cd . && \
	rm -rf $backupdir && mkdir $backupdir && \
	if (echo makeinfo missing; true --version) >/dev/null 2>&1; then \
	  for f in doc/openocd.info doc/openocd.info-[0-9] doc/openocd.info-[0-9][0-9] doc/openocd.i[0-9] doc/openocd.i[0-9][0-9]; do \
	    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
	  done; \
	else :; fi && \
	cd "$am__cwd"; \
	if echo makeinfo missing; true   -I doc -I ./doc \
	 -o doc/openocd.info ./doc/openocd.texi; \
	then \
	  rc=0; \
	  CDPATH="${ZSH_VERSION+.}:" && cd .; \
	else \
	  rc=$?; \
	  CDPATH="${ZSH_VERSION+.}:" && cd . && \
	  $restore $backupdir/* `echo "./doc/openocd.info" | sed 's|[^/]*$||'`; \
	fi; \
	rm -rf $backupdir; exit $rc
makeinfo missing
depbase=`echo src/target/espressif/esp32_apptrace.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
	/bin/sh ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.   -I./src -I./src -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl   -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Werror  -g -O2 -MT src/target/espressif/esp32_apptrace.lo -MD -MP -MF $depbase.Tpo -c -o src/target/espressif/esp32_apptrace.lo src/target/espressif/esp32_apptrace.c &&\
	mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Werror -g -O2 -MT src/target/espressif/esp32_apptrace.lo -MD -MP -MF src/target/espressif/.deps/esp32_apptrace.Tpo -c src/target/espressif/esp32_apptrace.c -o src/target/espressif/esp32_apptrace.o
In file included from src/target/espressif/esp32_apptrace.c:25:
./src/helper/list.h:49:9: error: 'LIST_HEAD' macro redefined [-Werror,-Wmacro-redefined]
   49 | #define LIST_HEAD(name) \
      |         ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/queue.h:465:9: note: previous definition is here
  465 | #define LIST_HEAD(name, type)                                           \
      |         ^
1 error generated.
make[2]: *** [src/target/espressif/esp32_apptrace.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2

erhankur, 2025-03-10:
@zetaerre Thank you for reporting the issue. For the same error, there is a fix in the mainline OpenOCD.  We will apply that fix to our repo and let you know.

erhankur, 2025-03-12:
@zetaerre Fixed in the master branch. Please try.

closed 2025-03-13 by erhankur