espressif/openocd-esp32 #360: Unable to dump Gcov data on ESP32-P4

RCapolungo, 2025-03-12:
### Development Kit

ESP32-P4-Function-EV-Board

### Module or chip used

ESP32-P4

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.12.0-esp32-20241016 (2024-10-16-14:17)

### Operating System

Ubuntu 24.04

### Using an IDE ?

VSCode

### OpenOCD command line

openocd -f board/esp32p4-ftdi.cfg

### JTAG Clock Speed

Clock Speed 10000 kHz

### ESP-IDF version

5.4

### Problem Description

1. First of all i burned the `DIS_USB_JTAG` efuse.
2. I connected the relative pins of the ESP-PROG to the board (GPIO2 to GPIO5).
3. I run the openocd command
4. run on another terminal `telnet localhost 4444`
5. Sent the `reset` command to synchronize the MCU with openocd
6. Sent `gcov dump`
7. On the telent terminal i got 
```
Total trace memory: 16384 bytes
Connect targets...
[esp32p4.hp.cpu0] Target halted, PC=0x4FC18DCE, debug_reason=00000000
[esp32p4.hp.cpu1] Target halted, PC=0x4FF06142, debug_reason=00000000
Targets connected.
Connection closed by foreign host.
```

And this is what was printed out in the openocd terminal

```
openocd: ../src/target/xtensa/xtensa.h:294: target_to_xtensa: Assertion `xtensa->common_magic == 0x54E4E555U' failed.
Aborted (core dumped)
```

### Debug Logs

```shell

```

### Expected behavior

openocd doen't crash and receive the gcov data

sobuch, 2025-03-12:
Hi @RCapolungo I was able to reproduce with the same version, but not with a newer one.

Could you please try whether updating openocd (to master in this repo or at least latest release) solves the issue for you?

RCapolungo, 2025-03-12:
Hi @sobuch, I tried with latest release and worked, thanks :)

closed 2025-03-12 by RCapolungo