espressif/openocd-esp32 #361: Algorithm timed out after 3012 ms

BeiyueZhou, 2025-04-02:
### Development Kit

ESP32C3/ESP8685

### Module or chip used

ESP8685/ESP32C3

### Debug Adapter

ESP-PROG

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

Windows

### Using an IDE ?

ESP-IDF V5.4

### OpenOCD command line

openocd -f board/esp32c3-ftdi.cfg -c "program_esp D:\wenjian1\arduinosth\arduinomessage\combine/target.bin 0x10000 verify exit"

### JTAG Clock Speed

20000 kHz

### ESP-IDF version

V5.4

### Problem Description

All chip connections are normal and are not burned, but some devices can be burned successfully, while others will have errors.
Use a multimeter to test all pins and they are all good.

### Debug Logs

```shell
PS D:\Espressif\frameworks\esp-idf-v5.4> openocd -f board/esp32c3-ftdi.cfg -c "program_esp D:\wenjian1\arduinosth\arduinomessage\combine/target.bin 0x0000 verify exit"
Open On-Chip Debugger v0.12.0-esp32-20241016 (2024-10-16-14:17)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
** program_esp input args <0x0000 verify exit> **
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Reset cause (3) - (Software core reset)
** Programming Started **
Info : [esp32c3] Found 8 triggers
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Error: [esp32c3] Algorithm timed out after 3011 ms.
Info : [esp32c3] Target halted, PC=0x00000000, debug_reason=00000000
Info : [esp32c3] Halt cause (2) - (Illegal Instruction)
Error: [esp32c3] ra = 0x40380570
Error: [esp32c3] sp = 0x3fc843d0
Error: [esp32c3] gp = 0x0
Error: [esp32c3] tp = 0x0
Error: [esp32c3] t0 = 0x0
Error: [esp32c3] t1 = 0x0
Error: [esp32c3] t2 = 0x0
Error: [esp32c3] fp = 0x60008000
Error: [esp32c3] s1 = 0x14
Error: [esp32c3] a0 = 0x78
Error: [esp32c3] a1 = 0x0
Error: [esp32c3] a2 = 0x0
Error: [esp32c3] a3 = 0x11583218
Error: [esp32c3] a4 = 0x8
Error: [esp32c3] a5 = 0xa
Error: [esp32c3] a6 = 0xa0000000
Error: [esp32c3] a7 = 0x14
Error: [esp32c3] s2 = 0x0
Error: [esp32c3] s3 = 0xffe7ffff
Error: [esp32c3] s4 = 0x3
Error: [esp32c3] s5 = 0x10c80218
Error: [esp32c3] s6 = 0x1fff00
Error: [esp32c3] s7 = 0xa000
Error: [esp32c3] s8 = 0x0
Error: [esp32c3] s9 = 0x0
Error: [esp32c3] s10 = 0xa0
Error: [esp32c3] s11 = 0x14
Error: [esp32c3] t3 = 0x0
Error: [esp32c3] t4 = 0x0
Error: [esp32c3] t5 = 0x0
Error: [esp32c3] t6 = 0x0
Error: [esp32c3] pc = 0x0
Error: [esp32c3] mstatus = 0x201800
Error: [esp32c3] mepc = 0x0
Error: [esp32c3] mcause = 0x2
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
** Clock configuration set failed **
shutdown command invoked
D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg:9: Error:
at file "D:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20241016\openocd-esp32\share\openocd\scripts/target/esp_common.cfg", line 9
```

### Expected behavior

How to solve the error to burn correctly？

### Screenshots

[image]

erhankur, 2025-04-06:
@BeiyueZhou As I understand, this is a custom board, not an Espressif kit. Could you check the strapping pins?https://github.com/espressif/openocd-esp32/issues/358#issuecomment-2666088923

erhankur, 2025-04-10:
@BeiyueZhou any update on your side?

BeiyueZhou, 2025-04-10:
> [@BeiyueZhou](https://github.com/BeiyueZhou) any update on your side?

The chip is Lexin's module esp8685-wroom-01. Among hundreds of modules, only three modules were found to have this situation. It was confirmed through the multimeter that all the pins were normal. I tested it many times, and the chip was also in download mode, but it kept showing Algorithm timeout. I can't determine the problem point now.

closed 2025-04-21 by erhankur