espressif/openocd-esp32 #363: [ERROR] Cannot “Debugging With JTAG” as the ESP-IDF Programming Guide says

Hanning-Liu, 2025-05-06:
### Development Kit

Custom Board

### Module or chip used

ESP32-S3-R8 with 16Mb external flash

### Debug Adapter

ESP32-S3 Builtin

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

Windows

### Using an IDE ?

VSCode with ESP-IDF extension

### OpenOCD command line

openocd -f board/esp32s3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.4.1

### Problem Description

1. Copy the blink example project to a new directory.
2. Select the Flash Method to UART.
3. Select Port to Use as COM8.
4. Set Espressif Device as esp32s3 with builtin USB-JTAG config
5. Configure the project
6. Build the Project with no error
7. Flash the Project with no error
8. Hit the debug button at the bottom of VSCode
9. Get ERROR messages.

### Debug Logs

```shell

```

### Expected behavior

The connection from PC to ESP32-S3 is done effectively with a single USB cable. 

This is made possible by the ESP32-S3 chip itself, which provides two USB channels, one for JTAG and the other for the USB terminal connection. 

The USB cable should be connected to the D+/D- USB pins of ESP32-S3 and not to the serial RxD/TxD through a USB-to-UART chip.

### Screenshots

[image]

Hanning-Liu, 2025-05-06:
Here is the openocd debug log:

[openocd.log](https://github.com/user-attachments/files/20055370/openocd.log)

Hanning-Liu, 2025-05-06:
The usb-c port of my custom development board is connect to the GPIO19 and GPIO20 correctly.

[image]

[image]

Hanning-Liu, 2025-05-06:
I met the same issue in Linux. But it is solved after I followed the driver configuration instruction in the link below:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/configure-builtin-jtag.html#configure-usb-drivers

[image]

I have done the same thing in Windows according to the instruction in the link above. And the powershell says I have the driver installed already. But the JTAG debug still doesn't work.

[image]

erhankur, 2025-05-08:
@Hanning-Liu You can verify your driver installation by checking the registry (see https://github.com/espressif/openocd-esp32/issues/230#issuecomment-1134393392).

If everything looks correct, make sure you run OpenOCD with admin rights.

If the issue persists, try using Espressif’s official dev-kit to verify your hardware.

By the way, the log in the description shows you are using `xtensa-esp32-elf-gdb` instead of `xtensa-esp32s3-elf-gdb`. You’ll need to fix this once you have a proper OpenOCD connection.

Hanning-Liu, 2025-05-08:
@erhankur 

Thank you for your response! I've donwload and installed the driver manually from this [link](https://dl.espressif.com/dl/idf-driver/idf-driver-esp32-usb-jtag-2021-07-15.zip). And I've checked my registry and it seems lack of the data named "DeviceInterfaceGUIDs" compared with your screenshot. I am not sure if my installation is done correctly or not.

[image]

erhankur, 2025-05-08:
Easy way to fix it, you can add missing `DeviceInterfaceGUIDs` key with this value `{80869ad8-c37f-476a-a6b4-ae241c30a473}`

Hanning-Liu, 2025-05-08:
@erhankur 
Yes, after the operations above. I can see the same number you said in registry.

[image]

erhankur, 2025-05-09:
Probably because of missing windows specific descriptors on esp-usb-bridge device.  We will have a look at it. Could you please create an issue in https://github.com/espressif/esp-usb-bridge? 

We can close this for now.

closed 2025-05-09 by erhankur

Hanning-Liu, 2025-05-09:
OK, I will create an issue there.