espressif/openocd-esp32 #364: Openocd don't work when changing freertos tickrate

dhugoexe, 2025-06-02:
Hi everyone,
OpenOCD can't find breakpoints when changing configTICK_RATE_HZ to 1000 (instead of 100). I suspect there's an argument to pass in the openocd command but I can't find it. 

Is there exact documentation for ESP-specific arguments for OpenOCD (e.g., esp appimage_offset 0x20000)?

### More about my environnement
- ESP32-P4 DevKit
- FreeRTOS at 1000Hz for running  LVGL correctly
- ESP-IDF V5.4.1

Thanks

sobuch, 2025-06-03:
Hi @dhugoexe , based on the `WARNING: Memory protection support is not implemented yet!` message, this is an older version (memory protection support was added last year), multiple other fixes were certainly added since.

Could you please try with latest version?

dhugoexe, 2025-06-19:
Sorry for waiting.
So I've tried with the latest release and I observe two things:
- Can't run debug when I have the debug flags `-d3 -l log.txt`
- If I use appimage_offset (needed in my case, it can't find flash maps)

### Command used
```
 .\bin\openocd.exe -s .\share\openocd\scripts -f board/esp32p4-builtin.cfg -c "init; halt; esp appimage_offset 0x20000"
````

### Result

```
Open On-Chip Debugger v0.12.0-esp32-20250422 (2025-04-22-13:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : esp_usb_jtag: serial (60:55:F9:F9:89:13)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu0] Core 0 made part of halt group 1.
Info : [esp32p4.hp.cpu0] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu0]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu0] Examination succeed
Info : [esp32p4.hp.cpu1] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu1] Core 1 made part of halt group 1.
Info : [esp32p4.hp.cpu1] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu1]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu1] Examination succeed
Info : [esp32p4.hp.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FF0504E, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FF0BD28, debug_reason=00000000
Info : [esp32p4.hp.cpu0] Found 3 triggers
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32p4.hp.cpu0.irom' size 0 KB
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Info : Using flash bank 'esp32p4.hp.cpu0.drom' size 0 KB
Warn : Failed to get flash maps (-9)!
Warn : Failed to read flash size!
Warn : Failed to get flash mappings (-4)!
Error: Failed to probe flash, size 0 KB
.\share\openocd\scripts/target/esp_common.cfg:9: Error:
Traceback (most recent call last):
  File ".\share\openocd\scripts/target/esp_common.cfg", line 9, in script
```

### Current partition table

```csv
# Name,   Type, SubType,  Offset,  Size, Flags
# Note: if you have increased the bootloader size, make sure to update the offsets to avoid overlap
nvs,      data, nvs,      0x11000, 0x6000,
phy_init, data, phy,      0x18000, 0x1000,
factory,  app,  factory,  0x20000, 4M,
flash,    data, littlefs, ,        0xF0000, 
```

dhugoexe, 2025-06-24:
With the main branch, got the following error:

```
Info : Using flash bank 'esp32p4.hp.cpu1.flash' size 16384 KB
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FC00B10, debug_reason=00000001
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC027EE, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : New GDB Connection: 1, Target esp32p4.hp.cpu0, state: halted
Info : [esp32p4.hp.cpu0] Disabling abstract command reads from FPRs.
Info : [esp32p4.hp.cpu1] Disabling abstract command reads from FPRs.
Memory Map (10 regions):
----------------------------------------
Type    Start Addr    Length    BlockSize
----------------------------------------
FLASH   0x00000000  0x01000000  0x00001000
R/W     0x01000000  0x3f000000  0x00000000
FLASH   0x40000000  0x0015f000  0x00001000
R/W     0x4015f000  0x00001000  0x00000000
FLASH   0x40160000  0x00112000  0x00001000
R/W     0x40272000  0x0f98e000  0x00000000
ROM     0x4fc00000  0x00020000  0x00000000
R/W     0x4fc20000  0x004e0000  0x00000000
ROM     0x50100000  0x00004000  0x00000000
R/W     0x50104000  0xafefc000  0x00000000
----------------------------------------
Info : Detected FreeRTOS version: (10.5.1)
Warn : [esp32p4.hp.cpu0] Failed to read memory via system bus.
Warn : [esp32p4.hp.cpu0] Failed to read memory via program buffer.
Warn : [esp32p4.hp.cpu0] Failed to read memory via abstract access.
Error: [esp32p4.hp.cpu0] Failed to read memory (addr=0xde408118)
Error: [esp32p4.hp.cpu0]   progbuf=failed, sysbus=failed, abstract=failed
Warn : Error reading FreeRTOS thread 0xde4080e7 name!
Error: [esp32p4.hp.cpu0] Failed to read memory (addr=0x850a8434)
Error: [esp32p4.hp.cpu0]   progbuf=failed, sysbus=failed, abstract=failed
Warn : Error reading thread list item object in FreeRTOS thread list!
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FC00B10, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC027EE, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Reset cause (24) - (JTAG CPU reset)
Info : [esp32p4.hp.cpu1] Found 3 triggers
Info : Detected FreeRTOS version: (10.5.1)
Error: [esp32p4.hp.cpu0] Unexpected read of fp via program buffer.
Error: [esp32p4.hp.cpu0] Unexpected read of fp via program buffer.
Error: Failed to read mcause register. Unknown exception reason!
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FF050F8, debug_reason=00000000
Info : [esp32p4.hp.cpu0] Hart unexpectedly reset!
Info : [esp32p4.hp.cpu1] Hart unexpectedly reset!
Info : [esp32p4.hp.cpu0] Reset cause (7) - (HP WDT resets system)
Info : [esp32p4.hp.cpu1] Reset cause (7) - (HP WDT resets system)
Error: [esp32p4.hp.cpu0] could not remove breakpoint #59168800 on this target
Error: [esp32p4.hp.cpu0] failed to remove breakpoint at address 0x4001e00a
Error: [esp32p4.hp.cpu1] could not remove breakpoint #59166104 on this target
Error: [esp32p4.hp.cpu1] failed to remove breakpoint at address 0x4001e00a
Error: [esp32p4.hp.cpu0] could not remove breakpoint #59163408 on this target
Error: [esp32p4.hp.cpu0] failed to remove breakpoint at address 0x4005d732
Error: [esp32p4.hp.cpu1] could not remove breakpoint #59168800 on this target
Error: [esp32p4.hp.cpu1] failed to remove breakpoint at address 0x4005d732
```

So VSCode starts a debug session, but nothing happens on the ESP32P4 - it starts normally without being halted. Unfortunately, I have only this board and can't try with other MCU. I can recreate the bug every time I set `FREERTOS_HZ=1000` on esp32p4 devkit.

erhankur, 2025-06-30:
We still couldn't reproduce the issue. What is your P4 revision?  Could you try with the hello-world example by changing app offset  and freertos tick? And please run OpenOCD and GDB from the command line and share the verbose -d3 logs.

dhugoexe, 2025-07-01:
Thanks for your feedback. It works fine with the latest version (and only the latest version of OpenOCD) on the hello world sample project. On my project, it doesn't work as expected. I'm trying to investigate if other config issues are troubling OpenOCD before closing this issue.

dhugoexe, 2025-07-01:
Just found something interesting. The problem is not directly due to FreeRTOS tick rate (because it works fine on the hello_world sample). But it occurs when used with espressif/esp_hosted and espressif/esp_wifi_remote components, with errors like failed to remove breakpoint or duplicated breakpoints.

sobuch, 2025-07-09:
Hi @dhugoexe we included some p4 fixes in the recent release, could you please confirm whether you are still able to reproduce the issue with your project and the new OpenOCD version? https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20250707 (or current master branch)

dhugoexe, 2025-07-09:
Still the same issue. So I'm providing the full logs from OpenOCD (sorry, I'm not exactly sure what information you're looking for).

[log.txt](https://github.com/user-attachments/files/21142177/log.txt)

erhankur, 2025-07-10:
@dhugoexe Could you please try with this build.

[openocd-esp32-win64-0.12.0-esp32-20250707-10-g8701624ce.zip](https://github.com/user-attachments/files/21161982/openocd-esp32-win64-0.12.0-esp32-20250707-10-g8701624ce.zip)

If it doesn't work, please get full log with 

```
-c 'init; reset halt; esp stub_log on; esp appimage_offset 0x20000' -d3 -l log.txt
```

dhugoexe, 2025-07-14:
Debugging works partially but fails when stepping through breakpoints. After setting a breakpoint and using "step into" to enter the target function, the debugger becomes unresponsive and fails to halt execution at any further breakpoints in the session.
```
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Info : [esp32p4.hp.cpu0] Target halted, PC=0x40061996, debug_reason=00000001
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FF0AE10, debug_reason=00000000
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4006199A, debug_reason=00000004
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4006199E, debug_reason=00000004
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Info : [esp32p4.hp.cpu0] Target halted, PC=0x40061616, debug_reason=00000004
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
Error: [esp32p4.hp.cpu0] Breakpoints couldn't be processed
Error: esp32p4.hp.cpu0: Failed to set bp (-1)!
[...]
```
Here is the full debug log:

[log.zip](https://github.com/user-attachments/files/21210704/log.zip)