espressif/openocd-esp32 #366: Openocd failed to get flash maps in micropython firmware

XIAO-Q7, 2025-06-12:
### Development Kit

ESP32-S3-DevKitC-1 N32R16V

### Module or chip used

ESP32-S3-WROOM-2 N32R16V

### Debug Adapter

USB-JTAG

### OpenOCD version

v0.12.0-esp32-20241016

### Operating System

WSL-Ubuntu24.04 / WSL-Ubuntu22.04

### Using an IDE ?

VSCode

### OpenOCD command line

openocd -f interface/esp_usb_jtag.cfg -f target/esp32s3.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.4.1

### Problem Description

Openocd failed to get flash maps in micropython firmware

**micropython version 1.25.0**

1. compiled the ESP32S3 MicroPython firmware based on the MicroPython project. I made the following changes to fit the ESP32 dev board.
    1. select partitions-32MiB.csv
    2. I set MICROPY_HW_ENABLE_USBDEVD, MICROPY_HW_ESP_USB_SERIAL_JTAG, and MICROPY_HW_USB_CDC to 0 to disable MicroPython operations on the USB port.
2. I flashed the firmware to the ESP32S3 board, and MicroPython runs as expected.
3. However, when I try to start debugging, OpenOCD throws the following errors:

```
Warn : 1435 12843 esp_flash.c:383 esp_algo_flash_get_mappings(): Failed to get flash maps (-9)!
Warn : 1436 12843 esp_flash.c:392 esp_algo_flash_get_mappings(): Failed to read flash size!
Warn : 1437 12843 esp_flash.c:986 esp_algo_flash_probe(): Failed to get flash mappings (-4)!
Error: 1438 12843 esp_flash.c:1022 esp_algo_flash_probe(): Failed to probe flash, size 0 KB
Error: 1439 12843 core.c:291 get_flash_bank_by_num(): auto_probe failed
Error: 1440 12843 gdb_server.c:1056 gdb_new_connection(): Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
```
4. I later tried debugging the "hello world" example, and everything worked fine.
5. To avoid this error, I started OpenOCD with -c 'set ESP_FLASH_SIZE 0', and debugging worked normally.

I'm not sure what caused this problem. I checked the partition table offset in my project and it looks fine (although I'm not absolutely sure, as I'm not very familiar with this part). Is it possible that MicroPython changes something related to this?
I would appreciate any help or advice to locate the root cause.
If you are also using MicroPython, you may be able to reproduce this problem easily.

Failed debug with MicroPyhton FW
[fail_debug_openocd.log](https://github.com/user-attachments/files/20706317/fail_debug_openocd.log)
Succeed debug with ”hello world“ example
[succeed_debug_openocd.log](https://github.com/user-attachments/files/20706318/succeed_debug_openocd.log)

### Debug Logs

```shell

```

### Expected behavior

openocd throws no errors and debug normally

XIAO-Q7, 2025-06-12:
@erhankur  Sure. It's the binary file

[mpy.bin.txt](https://github.com/user-attachments/files/20707475/mpy.bin.txt)

GitHub does not allow uploading binaries, and you may need to modify the suffix to use it.

erhankur, 2025-06-12:
Is this merged binary? Which address should I load it? Could you send the others? (bootloader and partition table or merged all)

XIAO-Q7, 2025-06-13:
This is not a merged binary, just an app binary.
Here are the other binaries:

[bootloader.bin.txt](https://github.com/user-attachments/files/20718245/bootloader.bin.txt)

[partition-table.bin.txt](https://github.com/user-attachments/files/20718229/partition-table.bin.txt)

The correspondence between the different bin files and their offsets is as follows:

```
        "0x0" : "bootloader.bin",
        "0x10000" : "mpy.bin",
        "0x8000" : "partition-table.bin"
```

I used the following command to flash the binaries:
```
/home/xq7/.espressif/python_env/idf5.4_py3.12_env/bin/python /home/xq7/esp/v5.4.1/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyACM0 -b 460800 --before default_reset --after hard_reset --chip esp32s3 write_flash --flash_mode dout --flash_freq 80m --flash_size detect 0x0 bootloader/bootloader.bin 0x10000 mpy.bin 0x8000 partition_table/partition-table.bin
```

erhankur, 2026-08-31:
@XIAO-Q7 any chance to test with the [latest](https://github.com/espressif/openocd-esp32/releases/tag/v0.12.0-esp32-20260703) OpenOCD release?