espressif/openocd-esp32 #368: Issue while debugging the ESP32C5 with builtin debugger

mgandler-vahle, 2025-07-04:
### Development Kit

Custom Board

### Module or chip used

ESP32-C5NF4

### Debug Adapter

Builtin Debugger

### OpenOCD version

v0.12.0-esp32-20250422

### Operating System

Docker on Windows with Ubuntu-22.04 Distro

### Using an IDE ?

VS Code

### OpenOCD command line

openocd -d${config:idf.openOcdDebugLevel} ${config:idf.openOcdConfigs,-f} -c init -c reset halt

### JTAG Clock Speed

24000 kHz

### ESP-IDF version

ESP-IDF v5.4.1-844-g8497af9b77

### Problem Description

1. Starting a demo project with nothing but the app_main function and a while loop.
2. Setting a breakpoint on the while loop
3. Starting the debugger

4. Expecting the debugger to stop at the breakpoint

Debugger is running and after klicking the pause button, the debugger stops somewhere but nothing is visible.

[gdb_log.txt](https://github.com/user-attachments/files/21054483/gdb_log.txt)
[openocd_log.txt](https://github.com/user-attachments/files/21054484/openocd_log.txt)
[demo_app.zip](https://github.com/user-attachments/files/21054482/demo_app.zip)
[demo-binaries.zip](https://github.com/user-attachments/files/21054528/demo-binaries.zip)

### Debug Logs

```shell
[OpenOCD]
Open On-Chip Debugger v0.12.0-esp32-20250422 (2025-04-22-13:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : only one transport option; autoselecting 'jtag'

Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000

Info : esp_usb_jtag: serial (60:55:F9:FF:FE:FC:8C:B8)

Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255

Info : clock speed 24000 kHz
[... 130 lines cut ...]
GDB notify async: breakpoint-created,bkpt={number="1",type="hw breakpoint",disp="del",enabled="y",addr="0x4200ad1e",func="app_main",file="/workspaces/dev-container-demo/main/demo.c",fullname="/workspaces/dev-container-demo/main/demo.c",line="70",thread-groups=["i1"],times="0",original-location="app_main"}
GDB unhandled notify: breakpoint-created: {"bkpt":{"number":"1","type":"hw breakpoint","disp":"del","enabled":"y","addr":"0x4200ad1e","func":"app_main","file":"/workspaces/dev-container-demo/main/demo.c","fullname":"/workspaces/dev-container-demo/main/demo.c","line":"70","thread-groups":["i1"],"times":"0","original-location":"app_main"}}
GDB result: 9 done
GDB command: 10 c
To client: {"seq":0,"type":"event","event":"output","body":{"category":"log","output":"c\n"}}
c
To client: {"seq":0,"type":"event","event":"output","body":{"category":"stdout","output":"Continuing.\n"}}
Continuing.
GDB result: 10 running
GDB exec async: running,thread-id="all"
To client: {"seq":0,"type":"event","event":"initialized"}
To client: {"seq":0,"type":"response","request_seq":2,"command":"attach","success":true}
From client: setBreakpoints({"source":{"name":"demo.c","path":"/workspaces/dev-container-demo/main/demo.c"},"lines":[70],"breakpoints":[{"line":70}],"sourceModified":false})
GDB command: 11 -exec-interrupt --all
From client: setFunctionBreakpoints({"breakpoints":[]})
GDB command: 12 -exec-interrupt --all
From client: setDataBreakpoints({"breakpoints":[]})
GDB command: 13 -exec-interrupt --all
From client: disconnect({"restart":false})
GDB command: 14 -gdb-exit
```

### Expected behavior

Debugger should stop at the breakpoint or jump to the source code after hitting pause button.

sobuch, 2025-07-04:
Hi @mgandler-vahle could you please share more information:

- verbose OpenOCD logs (passing `-d3 -l log.txt` argument to OpenOCD)
- serial port (UART) output

I see similar behavior if my board is in the download mode, could you please confirm whether the issue persists after resetting the board?

sobuch, 2025-07-09:
Hi @mgandler-vahle please see

- https://docs.espressif.com/projects/esptool/en/latest/esp32c5/advanced-topics/boot-mode-selection.html
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32c5/get-started/flashing-troubleshooting.html

seems to me you should be able to debug by setting GPIO28 high and resetting/power cycling the board after programming, so that the its in "Normal execution mode" again.

erhankur, 2025-07-16:
@mgandler-vahle looks like there is nothing to do in our side. Let us know if you have any other issue.