espressif/openocd-esp32 #374: bad

lianggan13, 2025-08-08:
### Development Kit

bad

### Module or chip used

bad

### Debug Adapter

bad

### OpenOCD version

bad

### Operating System

Windows

### Using an IDE ?

bad

### OpenOCD command line

bad

### JTAG Clock Speed

bad

### ESP-IDF version

bad

### Problem Description

[image]

### Debug Logs

```shell

```

### Expected behavior

bad

lianggan13, 2025-08-09:
> Hi [@lianggan13](https://github.com/lianggan13) , sorry that you are experiencing issues.
> 
> Have you checked https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ#usb-adapter-is-not-connected ? You might just need to install drivers, or select correct configuration file, but we will not be able to help without more details.

[image] No way,man.

sobuch, 2025-08-09:
Hi @lianggan13 , so the device is detected but that doesnt mean its correctly configured. Can you share `properties->driver->driver details` for it from the device manager? We are looking for `winusb.sys`, if its missing you may need to manually use something like https://visualgdb.com/UsbDriverTool/

lianggan13, 2025-08-09:
> Hi [@lianggan13](https://github.com/lianggan13) , so the device is detected but that doesnt mean its correctly configured. Can you share `properties->driver->driver details` for it from the device [...]

Thank you, sir. Well, I have solved the issue with the help of the following links:

https://www.softool.cn/blog-282.html
https://www.bilibili.com/video/BV1cChNz7Erf/?spm_id_from=333.337.search-card.all.click&vd_source=209996515a9bd0496489c385fbc38dbd

sobuch, 2025-08-09:
Glad to hear the issue is solved!

closed 2025-08-09 by sobuch