espressif/openocd-esp32 #376: ESP32-P4: Flashing via builtin JTAG works only if board is not in boot mode (IDFGH-16474)

orsolic, 2025-09-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

HW:
```
FC00 silicon

ESP-ROM:esp32p4-eco2-20240710
Build:Jul 10 2024
```
SW:
```
IDF:
commit 129cd0d247 (origin/release/v5.5)
Author:     Michael (XIAO Xufeng) <xiaoxufeng@espressif.com>
CommitDate: Mon 2025.09.01 12:22:06 +0800 

OpenOCD:
commit b981c3be8 (HEAD -> master, origin/master, origin/HEAD)
CommitDate: Mon 2025.09.15 16:02:45 +0800 

```

Steps to reproduce
- flash examples/get-started/blink somehow (UART or JTAG)
- restart board on phy button
- board now will blink LED and show that is running app on UART
- flash it via JTAG:
```
% openocd-esp32 -f board/esp32p4-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify reset exit"
Open On-Chip Debugger v0.12.0-esp32-20250707-211-gb981c3be (2025-09-16-16:47)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
** program_esp input args <0x10000 verify reset exit> **
Info : esp_usb_jtag: serial (30:ED:A0:E0:D5:BE)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu0] Core 0 made part of halt group 1.
Info : [esp32p4.hp.cpu0] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu0]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu0] Examination succeed
Info : [esp32p4.hp.cpu1] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu1] Core 1 made part of halt group 1.
Info : [esp32p4.hp.cpu1] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu1]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu1] Examination succeed
Info : [esp32p4.hp.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FC00B10, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC0283E, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Info : Flash mapping 0: 0x10020 -> 0x40020020, 39 KB
Info : Flash mapping 1: 0x20020 -> 0x40000020, 104 KB
Info : Auto-detected flash bank 'esp32p4.hp.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32p4.hp.cpu0.flash' size 16384 KB
You can enable `skip_loaded` to increase download speed.
Info : PROF: Erased 225280 bytes in 1587.85 ms
Info : PROF: Data transferred in 2721.43 ms @ 80.8399 KB/s
Info : PROF: Wrote 225280 bytes in 3065.3 ms (data transfer time included)
** Programming Finished in 5220 ms **
** Verify Started **
Info : PROF: Flash verified in 210.645 ms
** Verify OK **
** Resetting Target **
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
Info : [esp32p4.hp.cpu1] Reset cause (24) - (JTAG CPU reset)
shutdown command invoked
```
- board flashed, restarted (via JTAG), LED will blink, UART will show app is running
- put board in boot mode: physically hold boot button while pressing and releasing phy restart button
- rerun OpenOCD command from above, now it will fail
```
openocd-esp32 -f board/esp32p4-builtin.cfg -c "program_esp build/blink.bin 0x10000 verify reset exit"
Open On-Chip Debugger v0.12.0-esp32-20250707-211-gb981c3be (2025-09-16-16:47)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
** program_esp input args <0x10000 verify reset exit> **
Info : esp_usb_jtag: serial (30:ED:A0:E0:D5:BE)
Info : esp_usb_jtag: Device found. Base speed 24000KHz, div range 1 to 255
Info : clock speed 24000 kHz
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu0] Core 0 made part of halt group 1.
Info : [esp32p4.hp.cpu0] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu0]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu0] Examination succeed
Info : [esp32p4.hp.cpu1] datacount=1 progbufsize=2
Info : [esp32p4.hp.cpu1] Hart unexpectedly reset!
Info : [esp32p4.hp.cpu1] Core 1 made part of halt group 1.
Info : [esp32p4.hp.cpu1] Examined RISC-V core; found 2 harts
Info : [esp32p4.hp.cpu1]  XLEN=32, misa=0x40901125
Info : [esp32p4.hp.cpu1] Examination succeed
Info : [esp32p4.hp.cpu0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32p4.tap0 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : JTAG tap: esp32p4.tap1 tap/device found: 0x00012c25 (mfg: 0x612 (Espressif Systems), part: 0x0012, ver: 0x0)
Info : [esp32p4.hp.cpu0] Reset cause (24) - (JTAG CPU reset)
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FC00B10, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC0283E, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Reset cause (24) - (JTAG CPU reset)
** Programming Started **
Error: [esp32p4.hp.cpu0] Algorithm timed out after 3001 ms.
Info : [esp32p4.hp.cpu0] Target halted, PC=0x00000000, debug_reason=00000000
Info : [esp32p4.hp.cpu0] Halt cause (1) - (PMP Instruction access fault)
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC0283E, debug_reason=00000000
Error: [esp32p4.hp.cpu0] ra = 0x8ff000ba
Error: [esp32p4.hp.cpu0] sp = 0x8ff043e0
Error: [esp32p4.hp.cpu0] gp = 0x0
Error: [esp32p4.hp.cpu0] tp = 0x0
Error: [esp32p4.hp.cpu0] t0 = 0x4fc11e4e
Error: [esp32p4.hp.cpu0] t1 = 0xfffffff0
Error: [esp32p4.hp.cpu0] t2 = 0x0
Error: [esp32p4.hp.cpu0] fp = 0x1
Error: [esp32p4.hp.cpu0] s1 = 0x8000
Error: [esp32p4.hp.cpu0] a0 = 0x8ff04440
Error: [esp32p4.hp.cpu0] a1 = 0x43f88004
Error: [esp32p4.hp.cpu0] a2 = 0x0
Error: [esp32p4.hp.cpu0] a3 = 0x8ff04460
Error: [esp32p4.hp.cpu0] a4 = 0x8ff04460
Error: [esp32p4.hp.cpu0] a5 = 0x8ff04444
Error: [esp32p4.hp.cpu0] a6 = 0x20
Error: [esp32p4.hp.cpu0] a7 = 0x4fc10a5c
Error: [esp32p4.hp.cpu0] s2 = 0x20
Error: [esp32p4.hp.cpu0] s3 = 0x3000
Error: [esp32p4.hp.cpu0] s4 = 0x1000
Error: [esp32p4.hp.cpu0] s5 = 0x8ff040a0
Error: [esp32p4.hp.cpu0] s6 = 0x1
Error: [esp32p4.hp.cpu0] s7 = 0x43f80000
Error: [esp32p4.hp.cpu0] s8 = 0x3ffffff
Error: [esp32p4.hp.cpu0] s9 = 0x10000
Error: [esp32p4.hp.cpu0] s10 = 0x5012d034
Error: [esp32p4.hp.cpu0] s11 = 0x5008c380
Error: [esp32p4.hp.cpu0] t3 = 0x0
Error: [esp32p4.hp.cpu0] t4 = 0x0
Error: [esp32p4.hp.cpu0] t5 = 0x0
Error: [esp32p4.hp.cpu0] t6 = 0x0
Error: [esp32p4.hp.cpu0] pc = 0x0
Error: [esp32p4.hp.cpu0] mstatus = 0x1800
Error: [esp32p4.hp.cpu0] mepc = 0x0
Error: [esp32p4.hp.cpu0] mcause = 0x30000001
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
Warn : Failed to get flash mappings (-302)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32p4.hp.cpu0.flash'!
** Clock configuration set failed **
shutdown command invoked
embedded:startup.tcl:1893: Error:
Traceback (most recent call last):
  File "/usr/home/user/.opt/bin/../share/openocd/scripts/target/esp_common.cfg", line 507, in program_esp
    program_error {** Clock configuration set failed **} 1
  File "embedded:startup.tcl", line 1893, in program_error
    shutdown error
zsh: exit 1     openocd-esp32 -f board/esp32p4-builtin.cfg -c
```

orsolic, 2025-09-18:
Tried something recommended on internet for same situations for other ESP32 boards
Board is restarted into the boot mode (press and hold boot button, restart board).
```
esptool --port=/dev/cuaU0 erase-flash
esptool v5.0.1
Connected to ESP32-P4 on /dev/cuaU0:
Chip type:          ESP32-P4 (revision v1.0)
Features:           Dual Core + LP Core, 400MHz
Crystal frequency:  40MHz
USB mode:           USB-Serial/JTAG
MAC:                30:ed:a0:e0:d5:be

Stub flasher running.

Flash memory erased successfully in 51.7 seconds.

Hard resetting via RTS pin...
```

Writing bootloader.bin after erasing flash will work:
```
openocd-esp32 -f board/esp32p4-builtin.cfg -c "program_esp build/bootloader/bootloader.bin 0x2000; exit"
Open On-Chip Debugger v0.12.0-esp32-20250707-211-gb981c3be (2025-09-16-16:47)
...
** Programming Started **
Warn : Failed to get flash maps (-6)!
Warn : Application image is invalid! Check configured binary flash offset 'appimage_offset'.
Warn : Failed to get flash mappings (-4)!
Info : Auto-detected flash bank 'esp32p4.hp.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32p4.hp.cpu0.flash' size 16384 KB
You can enable `skip_loaded` to increase download speed.
Info : PROF: Erased 24576 bytes in 286.863 ms
Info : PROF: Data transferred in 382.876 ms @ 62.6835 KB/s
Info : PROF: Wrote 24576 bytes in 718.912 ms (data transfer time included)
** Programming Finished in 2508 ms **
```
It will continue to work until board is put in boot mode

Now, with board is in the boot mode (press and hold boot button, restart board) flashing will fail
```
** Programming Started **
Error: [esp32p4.hp.cpu0] Algorithm timed out after 3001 ms.
Info : [esp32p4.hp.cpu0] Target halted, PC=0x4FC11E96, debug_reason=00000000
Info : [esp32p4.hp.cpu1] Target halted, PC=0x4FC0283E, debug_reason=00000000
Error: [esp32p4.hp.cpu0] ra = 0x4fc11e84
Error: [esp32p4.hp.cpu0] sp = 0x8ff043d0
Error: [esp32p4.hp.cpu0] gp = 0x0
Error: [esp32p4.hp.cpu0] tp = 0x0
Error: [esp32p4.hp.cpu0] t0 = 0x4fc11e74
Error: [esp32p4.hp.cpu0] t1 = 0xffffffe0
Error: [esp32p4.hp.cpu0] t2 = 0x0
Error: [esp32p4.hp.cpu0] fp = 0x4ff3ffd8
Error: [esp32p4.hp.cpu0] s1 = 0x0
Error: [esp32p4.hp.cpu0] a0 = 0x0
Error: [esp32p4.hp.cpu0] a1 = 0x8ff04440
Error: [esp32p4.hp.cpu0] a2 = 0x20
Error: [esp32p4.hp.cpu0] a3 = 0x5008d02c
Error: [esp32p4.hp.cpu0] a4 = 0x3ff10000
Error: [esp32p4.hp.cpu0] a5 = 0x0
Error: [esp32p4.hp.cpu0] a6 = 0x0
Error: [esp32p4.hp.cpu0] a7 = 0x80000000
Error: [esp32p4.hp.cpu0] s2 = 0x20
Error: [esp32p4.hp.cpu0] s3 = 0x50aa
Error: [esp32p4.hp.cpu0] s4 = 0x0
Error: [esp32p4.hp.cpu0] s5 = 0x8ff04000
Error: [esp32p4.hp.cpu0] s6 = 0x0
Error: [esp32p4.hp.cpu0] s7 = 0x0
Error: [esp32p4.hp.cpu0] s8 = 0x0
Error: [esp32p4.hp.cpu0] s9 = 0x0
Error: [esp32p4.hp.cpu0] s10 = 0x0
[... 89 lines cut ...]
```

Disable memory protection - didn't help
```
(Top) → Component config → ESP System Settings → Memory protection
[ ] Enable IRAM/DRAM split protection
```

Reduce JTAG speed from 24 MHz to 1 MHz - didn't help
```
File $INSTALL_PREFIX/share/openocd/scripts/interface/esp_usb_jtag.cfg
adapter speed 1000

openocd output:
Info : clock speed 1000 kHz
```

Attached are logs, output of commands:
> openocd-esp32 -l jtag-app-mode.log -d3 -f board/esp32p4-builtin.cfg -c "program_esp build/bootloader/bootloader.bin 0x2000; exit"
> openocd-esp32 -l jtag-boot-mode.log -d3 -f board/esp32p4-builtin.cfg -c "program_esp build/bootloader/bootloader.bin 0x2000; exit"
[jtag-boot-mode.log](https://github.com/user-attachments/files/22415076/jtag-boot-mode.log)
[jtag-app-mode.log](https://github.com/user-attachments/files/22415077/jtag-app-mode.log)

Is there something else I could try?
What is the difference in OpenOCD flashing modes when the board is in boot mode and when it is in running app?

sobuch, 2025-09-30:
Hi @orsolic , thanks for the detailed report! As I mentioned on the other issue, we were able to reproduce this on our side as well. We now also have a workaround that seems to help. Would you be able to test the problematic scenario with this version?

Please see the appended archive for prebuild binaries, in case you prefer to build yourself I also pushed the sources in https://github.com/sobuch/openocd-esp32/tree/esp32p4_dowload_mode_flasher_fix

[openocd-esp32.zip](https://github.com/user-attachments/files/22617935/openocd-esp32.zip)

orsolic, 2025-10-01:
I am on latest master of openocd-esp32 (063922b3d) and cherry picked only latest commit from your branch (09f1ac1fd2e3bd6da1620ecbaac5715ff3f3d92b) and it fixed problem.
Thanks!

erhankur, 2025-10-16:
The fix ac509d67d556e898ea7e3c140b81cf5b0d9fbf3e is merged to master branch.