espressif/openocd-esp32 #378: Click Debug button in VS Code, then print "embedded:startup.tcl:72: Error: Can't find interface/ftdi/esp32_devkitj_v1.cfg"

dnit11, 2025-09-27:
### Development Kit

esp32_devkitj_v1

### Module or chip used

ESP32-S3-N16R8

### Debug Adapter

JTAG

### OpenOCD version

v0.12.0-esp32-20250422

### Operating System

Windows

### Using an IDE ?

esp-idf

### OpenOCD command line

openocd -f

### JTAG Clock Speed

default

### ESP-IDF version

v5.5.1

### Problem Description

1. Click Debug button in VS Code, but failed 

### Debug Logs

```shell
Traceback (most recent call last):
  File "embedded:startup.tcl", line 72, in script
    find interface/ftdi/esp32_devkitj_v1.cfg

For assistance with OpenOCD errors, please refer to our Troubleshooting FAQ: https://github.com/espressif/openocd-esp32/wiki/Troubleshooting-FAQ
OpenOCD Exit with non-zero error code 1
[Stopped] : OpenOCD Server
❌ Open On-Chip Debugger v0.12.0-esp32-20250707 (2025-07-06-17:44)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
embedded:startup.tcl:72: Error: Can't find interface/ftdi/esp32_devkitj_v1.cfg
Traceback (most recent call last):
  File "embedded:startup.tcl", line 72, in script
    find interface/ftdi/esp32_devkitj_v1.cfg
```

### Expected behavior

How to resolved the issue ? Thanks.

sobuch, 2025-09-27:
Hi @dnit11 the `interface/ftdi/esp32_devkitj_v1.cfg` config file was deprecated and is now removed, `interface/ftdi/esp32_ftdi.cfg` should be used instead.

Could you please check settings.json whether incorrect config file is not set under "idf.openOcdConfigs"?

Seems we should update vscode extension, I still see the old config for example in https://github.com/espressif/vscode-esp-idf-extension/blob/master/src/espIdf/openOcd/defaultBoards.ts#L98 However, when I try to configure esp32s3 it sets the correct (esp32_ftdi.cfg) config for me in settings.json.

@brianignacio5 could you please check the above? do you know why it works for me even if the config in defaultBoards.ts is not correct?

erhankur, 2025-10-16:
@dnit11 feel free to reopen if you have another issue.