espressif/openocd-esp32 #379: ESP32C3 enters download mode instead of flash mode when using openocd + gdb + ESP Prog Debugger.

vamsikrishnaA9, 2025-09-30:
hello, 

I’m debugging an ESP32-C3 with OpenOCD + GDB.
The firmware is flashed correctly and boots fine when I power cycle the board (serial logs appear, app runs normally).
But when I start OpenOCD with:

```
openocd -s $OPENOCD_SCRIPTS \
  -f board/esp32c3-ftdi.cfg \
  -c "adapter speed 100; init; reset halt"

output : Open On-Chip Debugger v0.12.0-esp32-20250422 (2025-04-22-13:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : clock speed 100 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Reset cause (3) - (Software core reset)
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Target halted, PC=0x40000000, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Using flash bank 'esp32c3.irom' size 948 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Using flash bank 'esp32c3.drom' size 580 KB
Error: GDB missing ack(2) - assumed good
Error: GDB missing ack(2) - assumed good
Error: Error on socket 'GDB': errno==54, message: Connection reset by peer.
Info : dropped 'gdb' connection
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Target halted, PC=0x40000000, debug_reason=00000001
Warn : No symbols for FreeRTOS!
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Target halted, PC=0x40002090, debug_reason=00000004
Warn : No symbols for FreeRTOS!
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Target halted, PC=0x40002094, debug_reason=00000004
Warn : No symbols for FreeRTOS!
Warn : No symbols for FreeRTOS!

```

and then attach GDB, the chip often drops into UART download mode and prints “waiting for download” instead of booting from flash.
So the problem only happens under OpenOCD reset, not during normal boot.
It looks like OpenOCD’s reset handling might be toggling the BOOT/EN lines and confusing the BootROM.

My question:
What’s the correct OpenOCD configuration for ESP32-C3 to reset + halt the chip for debugging without accidentally forcing it into download mode? Should I be using reset_config none or soft_reset_halt?

I even tried pulling the GPIO9 high, but it didn't worked. 

My board config: 
ESP32 C3 Mini Devout. 

```
esptool.py v4.10.0
Found 6 serial ports
Serial port /dev/cu.usbserial-2120
Connecting........................
Chip is ESP32-C3 AZ (QFN32) (revision v1.1)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 18:8b:0e:b1:7e:8c
Uploading stub...
Running stub...
Stub running...
MAC: 18:8b:0e:b1:7e:8c
Hard resetting via RTS pin...
```

_NOTE : I have fused this board to Disable the USB JTAG._ 
```
Usb fuses:
**DIS_USB_JTAG (BLOCK0)                              Set this bit to disable function of usb switch to  = True R/W (0b1)**
 jtag in module of usb device                      
DIS_USB_SERIAL_JTAG (BLOCK0)                       USB-Serial-JTAG                                    = Enable R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Set this bit to exchange USB D+ and D- pins        = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             USB printing                                       = Enable R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disable UART download mode through USB-Serial-JTAG = False R/W (0b0)

Vdd fuses:
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
                                                   ock cycle   
```

Any ideas or suggestions on this issue would be greatly appreciated.

thanks
vamsi

sobuch, 2025-10-06:
Hi @vamsikrishnaA9 I was not able to reproduce this issue, but could you please explain why the `adapter speed 100` setting? does the issue happen also without it?

vamsikrishnaA9, 2025-10-06:
Hi @sobuch ,

The `adapter speed 100`, enables the OpenOCD to drive Jtag clock at defined speed. The issue still persists with out `adapter speed 100` parameter.  here are the logs 

```
Open On-Chip Debugger v0.12.0-esp32-20250422 (2025-04-22-13:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
/Users/vamsikrishna/.espressif/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
Traceback (most recent call last):
  File "/Users/vamsikrishna/.espressif/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9, in script
/Users/vamsikrishna/.espressif/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/share/openocd/scripts/target/esp_common.cfg:9: Error: 
Traceback (most recent call last):
  File "/Users/vamsikrishna/.espressif/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/share/openocd/scripts/target/esp_common.cfg", line 9, in script
Error: [esp32c3] Unsupported DTM version: -1
Error: [esp32c3] Could not identify target type.
vamsikrishna@MacBook ~ % /Users/vamsikrishna/.espressif/tools/openocd-esp32/v0.12.0-esp32-20250422/openocd-esp32/bin/openocd \
  -s $OPENOCD_SCRIPTS \
  -f board/esp32c3-ftdi.cfg \
  -c "reset_config none; init; reset halt"

Open On-Chip Debugger v0.12.0-esp32-20250422 (2025-04-22-13:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] datacount=2 progbufsize=16
Info : [esp32c3] Examined RISC-V core; found 1 harts
Info : [esp32c3]  XLEN=32, misa=0x40101104
Info : [esp32c3] Examination succeed
Info : [esp32c3] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.tap0 tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : [esp32c3] Reset cause (3) - (Software core reset)
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32c3] Target halted, PC=0x40000000, debug_reason=00000000
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Using flash bank 'esp32c3.irom' size 948 KB
Info : Flash mapping 0: 0x10020 -> 0x3c0f0020, 578 KB
Info : Flash mapping 1: 0xb0020 -> 0x42000020, 944 KB
Info : Using flash bank 'esp32c3.drom' size 580 KB
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
Warn : No symbols for FreeRTOS!
Info : [esp32c3] Target halted, PC=0x40002090, debug_reason=00000004
Warn : No symbols for FreeRTOS!
Warn : RTOS FreeRTOS not detected. (GDB could not find symbol 'pxReadyTasksLists')
Error: Don't have the number of threads in FreeRTOS!
Info : [esp32c3] Target halted, PC=0x40002094, debug_reason=00000004
Error: Don't have the number of threads in FreeRTOS!
Error: Don't have the number of threads in FreeRTOS!
Info : [esp32c3] Target halted, PC=0x40002098, debug_reason=00000004
Error: Don't have the number of threads in FreeRTOS!
[... 201 lines cut ...]
SPI_PAD_CONFIG_WP (BLOCK1)                         SPI PAD WP(D2)                                     = 0 R/W (0b000000)
SPI_PAD_CONFIG_DQS (BLOCK1)                        SPI PAD DQS                                        = 0 R/W (0b000000)
SPI_PAD_CONFIG_D4 (BLOCK1)                         SPI PAD D4                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D5 (BLOCK1)                         SPI PAD D5                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D6 (BLOCK1)                         SPI PAD D6                                         = 0 R/W (0b000000)
SPI_PAD_CONFIG_D7 (BLOCK1)                         SPI PAD D7                                         = 0 R/W (0b000000)

Usb fuses:
DIS_USB_JTAG (BLOCK0)                              Set this bit to disable function of usb switch to  = True R/W (0b1)
                                                   jtag in module of usb device                      
DIS_USB_SERIAL_JTAG (BLOCK0)                       USB-Serial-JTAG                                    = Enable R/W (0b0)
USB_EXCHG_PINS (BLOCK0)                            Set this bit to exchange USB D+ and D- pins        = False R/W (0b0)
DIS_USB_SERIAL_JTAG_ROM_PRINT (BLOCK0)             USB printing                                       = Enable R/W (0b0)
DIS_USB_SERIAL_JTAG_DOWNLOAD_MODE (BLOCK0)         Disable UART download mode through USB-Serial-JTAG = False R/W (0b0)

Vdd fuses:
VDD_SPI_AS_GPIO (BLOCK0)                           Set this bit to vdd spi pin function as gpio       = False R/W (0b0)

Wdt fuses:
WDT_DELAY_SEL (BLOCK0)                             RTC watchdog timeout threshold; in unit of slow cl = 40000 R/W (0b00)
                                                   ock cycle
```

Thanks
Vamsi

sobuch, 2025-10-07:
Thanks for the logs! Could you please also collect UART output? (you can use `idf.py monitor`) Since we are in the `ets_vprintf` function, I am curious whats printed.

Please note that theres a difference from the original output now, there is no `Error: GDB missing ack(2) - assumed good` message anymore. So its likely the slow jtag speed was causing some issues, even if its not the only problem here

erhankur, 2026-01-07:
Closed due to inactivity.